espressif/esp-adf #832: Bluetooth Intercom Between Two ESP32 LyraT Kit

shridattdudhat, 2022-05-31:
Hello, We want to try Bluetooth Intercom between Two ESP32 Audio Kits. Has anybody tried the solution ?

Please guide us how we can achieve the Intercom between two ESP32 boards?

shridattdudhat, 2022-06-01:
Any updates for this issue ?

HengYongChao, 2022-06-02:
Hi @shridattdudhat 

[pipeline_hfp_stream](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_hfp_stream) is what you need.

Also `hfp_ag`, `hfp_hf` examples in esp-idf repo, https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt

shridattdudhat, 2022-06-06:
Hi @HengYongChao 

Thank you for the suggestion. 

We tried hfp_ag and hfp_hf but that example does not have audio codec code.

Can you guide us how we can use hfp_ag and hfp_hf for Bluetooth Intercom functionality?

HengYongChao, 2022-06-08:
Hi @shridattdudhat 

Please refer to this routine, [pipeline_hfp_stream](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_hfp_stream).

shridattdudhat, 2022-07-05:
Hi @HengYongChao 
We used hfp_ag and hfp_hf example code, we are able to connect two audio kit with each other but not able to transfer the audio between them. Is there any another example code or sample code which we can try out to achieve bluetooth intercom functionality ?

shridattdudhat, 2022-07-08:
The Audio Connect is not working, I used hfp_ag and hfp_hf example code.

```
I (10453) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:2
connect audio
I (49523) BT_HF: APP HFP event: AUDIO_STATE_EVT
E (49533) BT_BTM: btm_sco_connected, handle 81
I (49533) BT_HF: --audio state connecting
E (49533) BT_BTM: btm_sco_connected, handle 81
hfp_hf> I (49543) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (49553) BT_HF: --audio state disconnected
```

shridattdudhat, 2022-11-03:
Any Sample code for this Bluetooth intercom ??