espressif/esp-adf #835: audio_thread_create use stack on ext psram, the stack size is limit to 64K

yah, 2022-06-16:
Hi,

use audio_thread_create to create task which use psram as stack,  if the stack size is equal or big than than 64k, it will cause stack overflow. 

the env is:
- board: ESP-BOX
- esp-idf:  commit f51d746a9a1b103e4dc45b5dfb48bfecc55e99d9 (HEAD -> release/v4.4)
- esp-adf: commit 8301789fa95fd40a65c90f0ed5cabf30a8d6441b (HEAD -> master, origin/master, origin/HEAD)

the log is:
```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1664
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f88
entry 0x403b6254
I (25) boot: ESP-IDF v4.4.1-247-gf51d746a9-dirty 2nd stage bootloader
I (25) boot: compile time 09:23:48
I (25) boot: chip revision: 0
I (29) boot.esp32s3: Boot SPI Speed : 80MHz
I (34) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 16MB
....
....
I (173) opi psram: vendor id : 0x0d (AP)
I (173) opi psram: dev id    : 0x02 (generation 3)
I (174) opi psram: density   : 0x03 (64 Mbit)
I (177) opi psram: good-die  : 0x01 (Pass)
I (182) opi psram: Latency   : 0x01 (Fixed)
I (187) opi psram: VCC       : 0x01 (3V)
I (191) opi psram: SRF       : 0x01 (Fast Refresh)
I (197) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (202) opi psram: BurstLen  : 0x01 (32 Byte)
I (207) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (213) opi psram: DriveStrength: 0x00 (1/1)
W (218) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (229) spiram: Found 64MBit SPI RAM device
I (234) spiram: SPI RAM mode: sram 80m
I (238) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (245) cpu_start: Pro cpu up.
I (248) cpu_start: Starting app cpu, entry point is 0x4037521c
I (0) cpu_start: App cpu up.
I (668) spiram: SPI SRAM memory test OK
I (676) cpu_start: Pro cpu start user code
I (682) cpu_start: cpu freq: 160000000
I (682) cpu_start: Application information:
I (684) cpu_start: Project name:     oid_ble
I (689) cpu_start: App version:      b0b0401-dirty
I (695) cpu_start: Compile time:     Jun 16 2022 10:17:41
I (701) cpu_start: ELF file SHA256:  5c11d6d8d33a6170...
I (707) cpu_start: ESP-IDF:          v4.4.1-247-gf51d746a9-dirty
I (714) heap_init: Initializing. RAM available for dynamic allocation:
I (721) heap_init: At 3FC95A70 len 0004A590 (297 KiB): D/IRAM
I (727) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (734) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (740) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (746) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1000) spi_flash: detected chip: gd
I (1001) spi_flash: flash io: dio
I (1002) sleep: Configure to isolate all GPIO pins in sleep state
I (1005) sleep: Enable automatic switching of GPIO sleep configuration
I (1012) coexist: coexist rom version e7ae62f
I (1017) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1027) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1037) AUDIO_THREAD: The psram_stack task allocate stack on external memory

***ERROR*** A stack overflow in task psram_stack has been detected.

Backtrace:0x40375b5a:0x3d800640 |<-CORRUPTED

ELF file SHA256: 5c11d6d8d33a6170

Rebooting...
```

the code is:
``` c
void audio_task(void *args) {
    while (1) {
        vTaskDelay(portTICK_PERIOD_MS * 22);
    }
}

void app_main(void) {
    audio_thread_t t;
    audio_thread_create(&t, "psram_stack", &audio_task, (void*)0, 64 * 1024, 1, true , 0);

    while (1) {
        vTaskDelay(portTICK_PERIOD_MS * 22);
    }
}
```

yah, 2022-06-16:
switch to idf master branch and use xTaskCreateRestricted to create task,  it works with stack size bigger than 64k

shootao, 2022-06-16:
Hi @yah ， Thank you for your feedback.
The reason for this is that the function `audio_thread_create` has a stack size of `uint16_t` type. The solution to this problem is to change the type to `uint32_t`. Reference [Link](https://github.com/espressif/esp-idf/blob/009e66e6252ab675ba126afa8919bae14a958ecd/components/freertos/include/freertos/FreeRTOS.h#L936)