espressif/esp-adf #837: VOIP example crashes ESP32-LyraTD-MSC V2.2

soloex1991eric, 2022-06-24:
**Enviroment:**
ESP-IDF Version: 4.4.1
ESP-ADF Version: Master
SIP Server: [miniSIPserver](https://www.myvoipapp.com/)
SIP Codec: G711 PCMA
Audio Hal: ESP32-LyraTD-MSC V2.2

**Issue:**
ESP32-LyraTD-MSC V2.2crashes, after answering call.

**Steps taken:**
Flashed Voip example after configuring wifi, SIP and Audio Hal. Then Calling from another attached SIP device and answering the call from ESP32-LyraTD-MSC V2.2

**Log File:**
[ESP_ADF.txt](https://github.com/espressif/esp-adf/files/8974781/ESP_ADF.txt)

soloex1991eric, 2022-06-28:
It's the original example I didn't change anything. I only configured the WiFi ssid, WiFi password, audio hal and VoIP url, as directed by the example's instructions.

ahhfzhang, 2022-08-04:
Hi @soloex1991eric 

`W (19042) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1`

Please use Module with `ESP32-WROVER-E`, then try again.

Thanks.

soloex1991eric, 2022-08-04:
Please remove any mention of VOIP support on ESP32-LyraTD-MSC V2.2, at https://github.com/espressif/esp-adf/tree/master/examples. If that is the case. Also remove all marketing materials regarding its support.

ahhfzhang, 2022-08-04:
Your board should be many years ago. 

The ESP32-LyraTD-MSC board has been using the ESP32-WROVER-E module for two years.

soloex1991eric, 2022-08-04:
That is odd I only bought it this year, in January. From Digikey. https://www.digikey.my/en/products/detail/schtoeta-engineering-limited/ESP32-LYRATD-MSC/9381705. Never mind please add disclaimer to example. I'll close the Issue.

closed 2022-08-04 by soloex1991eric