espressif/esp-adf #842: HLS with cookies.

KwangryeolPark, 2022-07-04:
I am using HLS example.As you know, HLS needs m3u8 URI and my server requires cookies for m3u8 URI. However, I couldn't find any examples or features which use cookies.

So what I want to know is that should I edit libraries for cookies? or is there any function for cookies?

I am thinking modifying libraries or related codes.

If there are solution, then please tell me..
If there are no official solution, then please add the feature.

Thank you.

HengYongChao, 2022-07-15:
Hi @KwangryeolPark 

http client is not supported cookies now, We will add this feature when IDF supports it.

KwangryeolPark, 2022-07-15:
I just modify the _http_open function
Thanks

closed 2022-07-15 by KwangryeolPark