espressif/esp-adf #843: Example for writing audio data to TCP

soloex1991eric, 2022-07-07:
I've been trying to send Audio data to a TCP server to no avail. I've configured the audio stream element, type. Following the "pipeline_tcp_client" but instead of setting it to "AUDIO_STREAM_READER" I've instead set it to "AUDIO_STREAM_WRITER". Then I created a "_tcp_stream_event_handle", event handler following the "pipeline_raw_http" example. But ran into an issue, a sthere  was no equivalent of the http_stream_event_id_t" component in the "tcp_stream_event_msg_t" structure. I've tried instead to use a simple while loop and the function, "esp_transport_write" to write data to the TCP server but encountered a watch dog issue. Honestly, I am currently at a loss on how to progress. Any help on this matter will be greatly appreciated.

HengYongChao, 2022-07-15:
Hi @soloex1991eric 

Here is the unit test code, it contains the tcp stream client write  test source.

FYI: https://github.com/espressif/esp-adf/blob/master/components/audio_stream/test/tcp_client_stream_test.c#L68

soloex1991eric, 2022-07-15:
Cool, Thanks for the info.

closed 2022-07-15 by soloex1991eric