espressif/esp-adf #844: when i add another task to the wwe example, abort happens

peng86000501, 2022-07-08:
## Environment

- Audio development kit:      [ESP32-LyraT-Mini_V1.2]
- Audio kit version (for ESP32-LyraT-Mini): [V1.2]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
 v4.4
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
master downloaded on 20220702
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
- Operating system:     [Windows]
- (Windows only) Environment type: [ESP Command Prompt]
- Using an IDE?: [No]
- Power supply:         [USB]

## Problem Description

when i try to add another task (i tried gg_translation or speech_to_mp3_baidu) to the wwe example, abort happens 

### Expected Behavior
the added task can run correctly
### Actual Behavior
abort and reboot
### Steps to Reproduce

1. flash and run the programe

### Code to Reproduce This Issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
void app_main(void)
{

#if 1
    log_clear();
	ESP_LOGI(TAG, "[ 1 ] Initialize Buttons & Connect to Wi-Fi network, ssid=%s", "tt&pengpeng");
	// Initialize peripherals management

    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    periph_cfg.extern_stack = true;
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    if (set != NULL) {
        esp_periph_set_register_callback(set, periph_callback, NULL);
    }
    
#if VOICE2FILE
    audio_board_sdcard_init(set, SD_MODE_1_LINE);
#endif
    audio_board_key_init(set);

    audio_board_init();
	
    setup_player();
    start_recorder();
	xTaskCreate(speech_to_mp3_baidu_main, "speech_to_mp3_baidu", 2 * 4096, NULL, 5, NULL);

    rec_q = xQueueCreate(3, sizeof(int));
    audio_thread_create(NULL, "read_task", voice_read_task, NULL, 4 * 1024 + 4 * 1024 , 5, false, 0);
    //xTaskCreate(voice_read_task, "read_task", 4 * 1024 + 3 * 4096, NULL, 5, NULL);
#endif
    //xTaskCreate(translate_task, "gg_translation", 3 * 4096, NULL, 5, NULL);
    //audio_thread_create(NULL, "gg_translation", translate_task, NULL, 3 * 4096, 5, true, 1);

[... 128 lines cut ...]
func: _wifi_init
expression: esp_event_loop_create_default()

abort() was called at PC 0x4009109f on core 0
0x4009109f: _esp_error_check_failed at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/esp_system/esp_err.c:43

Backtrace:0x4008247d:0x3fff8ba00x400910a9:0x3fff8bc0 0x40098bd5:0x3fff8be0 0x4009109f:0x3fff8c50 0x400dd7a8:0x3ff0x4008247d: panic_abort at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/esp_system/panic.c:402

0x400910a9: esp_system_abort at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40098bd5: abort at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/newlib/abort.c:46

0x4009109f: _esp_error_check_failed at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/esp_system/esp_err.c:43

0x400dd7a8: _wifi_init at D:/2022/20220505_esp32/20220702_adf/esp-adf-master/components/esp_peripherals/periph_wifi.c:393 (discriminator 1)

f8c70 0x400dd6fe:0x3fff8de0
0x400dd6fe: esp_periph_task at D:/2022/20220505_esp32/20220702_adf/esp-adf-master/components/esp_peripherals/esp_peripherals.c:104 (discriminator 15)
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

HengYongChao, 2022-07-15:
Hi @peng86000501 

```c
0x4009109c: _esp_error_check_failed at D:/Espressif4.4/frameworks/esp-idf-v4.4/components/esp_system/esp_err.c:42

file: "D:/2022/20220505_esp32/20220702iver task: 3ffc68c8, prio:23, stack:6656, core=0
```
Please check `iver task` in your source code. It seems wifi stack was not init correctly.

peng86000501, 2022-07-23:
Thanks a lot!

xTaskCreate(speech_to_mp3_baidu_main, "speech_to_mp3_baidu", 2 * 4096, NULL, 5, NULL);
-> this code causes the problem, and I find another way to create a new task

I can use audio_thread_create(...) to create new task.

And the wifi problem, I found in the duer_os example to use the wifi_service_connect(...) to connect via wifi.

closed 2022-07-23 by peng86000501