espressif/esp-adf #848: how to support i2s chips like INMP441 and MAX98357

hyansuper, 2022-07-17:
如题，默认的ESP-ADF支持这些板子https://github.com/espressif/esp-adf/tree/master/components/audio_board
但如果我想用ESP32-S3-DevKitC-1接市面上较常见的I2S语音芯片比如[INMP441](https://item.taobao.com/item.htm?spm=a230r.1.14.32.5a955018AnBibq&id=630204400000&ns=1&abbucket=10#detail)和[MAX98357](https://detail.tmall.com/item.htm?spm=a230r.1.14.16.ce8a426d1Tze5i&id=589950203794&ns=1&abbucket=10)，请问因该怎么做呢？

HengYongChao, 2022-07-19:
Hi @hyansuper 

你提到的这个数字功放和麦克都是有 i2s 接口的，可按照 i2s 的接口初始化后使用， 一个 i2s 做 mic 输入， 一个 i2s 做 PA 输出。

hyansuper, 2022-07-20:
谢谢
我知道esp-idf的i2s接口就可以控制这两个模块。
但要怎么使用esp-adf的i2s_stream控制从而能够方便的使用adf的各功能呢

jason-mao, 2022-08-08:
@hyansuper 
INMP441 是I2S的mic，可以直接接，不用适配ADF的接口。
MAX98357 和ES7148一样都是不需要I2C控制的，只输出I2S就行，参考https://github.com/espressif/esp-adf/tree/master/components/audio_hal/driver/es7148

这两个片子比较简单，基本只要I2S的格式对了，就可以跑ADF的demo。

hyansuper, 2022-08-09:
谢谢！

closed 2022-08-09 by hyansuper