espressif/esp-adf #849: Streaming problem, and raw streaming problem

KwangryeolPark, 2022-07-18:
Hello, I'm doing a streaming test. There are cases where the sound is cut off during streaming, but I don't know if this is a Wi-Fi problem or an internal program problem. Do you have any solution? This is a project that plays consecutive ts files in m3u8 format. And also, when I try to use raw streaming for storing .ts data, the audio is also cut off (but at different position).

The code is same with play_living_stream_example but some SDK configurations are different. (I am using ESP32-Lyrat V4.3)
```
I am using PSRAM and WiFi task is running on CPU 1

Flash SPI speed: 80 MHz
CPU frequency: 240Hz
Max number of WiFi static RX buffers: 16
Max number of WiFi dynamic RX buffers: 128
Max number of WiFi static TX buffers: 32
Max number of WiFi cache TX buffers: 128

Default send buffer size: 65534
Default receive window size: 65534
Default TCP receive mail box size: 64

```

HengYongChao, 2022-07-19:
Hi @KwangryeolPark 

Could you provide a log about the sound cut off ?  SDK configurations is not enough for debug those issue.

KwangryeolPark, 2022-07-19:
@HengYongChao 
There are no error or warning message, but here is my log

**This is the log when the music was briefly interrupted.**
~ <- This is for security

```
I (191007) HTTP_STREAM: _playlist_get_next_track
I (191012) HTTP_STREAM: URI=~~~0.ts
I (191025) HTTP_STREAM: URL: ~~~0.ts
I (191384) HTTP_STREAM: get_audio_type
I (191386) HTTP_STREAM: total_bytes 1: 433716
I (191387) HTTP_STREAM: _is_playlist
I (191400) AUDIO_ELEMENT: REPORT_FMT,[http]evt out cmd:10,
I (191400) AUDIO_ELEMENT: [http] el opened
I (191402) AUDIO_ELEMENT: REPORT_STATUS,[http]evt out cmd = 8,status:12
I (191925) HTTP_STREAM: req lengh=2048, read=2048, pos=0/433716
I (191936) HTTP_STREAM: req lengh=2048, read=2048, pos=2048/433716
I (191937) HTTP_STREAM: req lengh=2048, read=2048, pos=4096/433716
I (191940) HTTP_STREAM: req lengh=2048, read=2048, pos=6144/433716
I (191947) HTTP_STREAM: req lengh=2048, read=2048, pos=8192/433716
I (191959) HTTP_STREAM: req lengh=2048, read=2048, pos=10240/433716
I (191961) HTTP_STREAM: req lengh=2048, read=2048, pos=12288/433716
I (191967) HTTP_STREAM: req lengh=2048, read=2048, pos=14336/433716
I (192036) HTTP_STREAM: req lengh=2048, read=2048, pos=16384/433716
I (192048) HTTP_STREAM: req lengh=2048, read=2048, pos=18432/433716
I (192049) HTTP_STREAM: req lengh=2048, read=2048, pos=20480/433716
I (192056) HTTP_STREAM: req lengh=2048, read=2048, pos=22528/433716
I (192063) HTTP_STREAM: req lengh=2048, read=2048, pos=24576/433716
I (192074) HTTP_STREAM: req lengh=2048, read=2048, pos=26624/433716
I (192094) HTTP_STREAM: req lengh=2048, read=2048, pos=28672/433716
I (192142) HTTP_STREAM: req lengh=2048, read=2048, pos=30720/433716
I (192258) HTTP_STREAM: req lengh=2048, read=2048, pos=32768/433716
I (192259) HTTP_STREAM: req lengh=2048, read=2048, pos=34816/433716
I (192278) HTTP_STREAM: req lengh=2048, read=2048, pos=36864/433716
I (192325) HTTP_STREAM: req lengh=2048, read=2048, pos=38912/433716
I (192375) HTTP_STREAM: req lengh=2048, read=2048, pos=40960/433716
I (192441) HTTP_STREAM: req lengh=2048, read=2048, pos=43008/433716
I (192469) HTTP_STREAM: req lengh=2048, read=2048, pos=45056/433716
I (192516) HTTP_STREAM: req lengh=2048, read=2048, pos=47104/433716
I (192563) HTTP_STREAM: req lengh=2048, read=2048, pos=49152/433716
I (192638) HTTP_STREAM: req lengh=2048, read=2048, pos=51200/433716
I (192749) HTTP_STREAM: req lengh=2048, read=2048, pos=53248/433716
I (192751) HTTP_STREAM: req lengh=2048, read=2048, pos=55296/433716
I (192768) HTTP_STREAM: req lengh=2048, read=2048, pos=57344/433716
I (192816) HTTP_STREAM: req lengh=2048, read=2048, pos=59392/433716
I (192863) HTTP_STREAM: req lengh=2048, read=2048, pos=61440/433716
I (192911) HTTP_STREAM: req lengh=2048, read=2048, pos=63488/433716
I (192951) HTTP_STREAM: req lengh=2048, read=2048, pos=65536/433716
I (193005) HTTP_STREAM: req lengh=2048, read=2048, pos=67584/433716
I (193046) HTTP_STREAM: req lengh=2048, read=2048, pos=69632/433716
I (193121) HTTP_STREAM: req lengh=2048, read=2048, pos=71680/433716
I (193162) HTTP_STREAM: req lengh=2048, read=2048, pos=73728/433716
I (193209) HTTP_STREAM: req lengh=2048, read=2048, pos=75776/433716
I (193263) HTTP_STREAM: req lengh=2048, read=2048, pos=77824/433716
I (193278) HTTP_STREAM: req lengh=2048, read=2048, pos=79872/433716
I (193345) HTTP_STREAM: req lengh=2048, read=2048, pos=81920/433716
I (193396) HTTP_STREAM: req lengh=2048, read=2048, pos=83968/433716
I (193492) HTTP_STREAM: req lengh=2048, read=2048, pos=86016/433716
I (193494) HTTP_STREAM: req lengh=2048, read=2048, pos=88064/433716
I (193536) HTTP_STREAM: req lengh=2048, read=2048, pos=90112/433716
I (193585) HTTP_STREAM: req lengh=2048, read=2048, pos=92160/433716
I (193604) HTTP_STREAM: req lengh=2048, read=2048, pos=94208/433716
I (193651) HTTP_STREAM: req lengh=2048, read=2048, pos=96256/433716
I (193699) HTTP_STREAM: req lengh=2048, read=2048, pos=98304/433716
I (193747) HTTP_STREAM: req lengh=2048, read=2048, pos=100352/433716
I (193847) HTTP_STREAM: req lengh=2048, read=2048, pos=102400/433716
I (193848) HTTP_STREAM: req lengh=2048, read=2048, pos=104448/433716
I (193884) HTTP_STREAM: req lengh=2048, read=2048, pos=106496/433716
I (193932) HTTP_STREAM: req lengh=2048, read=2048, pos=108544/433716
[... 269 lines cut ...]
I (205563) HTTP_STREAM: req lengh=2048, read=2048, pos=192512/221464
I (205660) HTTP_STREAM: req lengh=2048, read=2048, pos=194560/221464
I (205661) HTTP_STREAM: req lengh=2048, read=2048, pos=196608/221464
I (205701) HTTP_STREAM: req lengh=2048, read=2048, pos=198656/221464
I (205747) HTTP_STREAM: req lengh=2048, read=2048, pos=200704/221464
I (205795) HTTP_STREAM: req lengh=2048, read=2048, pos=202752/221464
I (205815) HTTP_STREAM: req lengh=2048, read=2048, pos=204800/221464
I (205862) HTTP_STREAM: req lengh=2048, read=2048, pos=206848/221464
I (205910) HTTP_STREAM: req lengh=2048, read=2048, pos=208896/221464
I (206006) HTTP_STREAM: req lengh=2048, read=2048, pos=210944/221464
I (206008) HTTP_STREAM: req lengh=2048, read=2048, pos=212992/221464
I (206053) HTTP_STREAM: req lengh=2048, read=2048, pos=215040/221464
I (206121) HTTP_STREAM: req lengh=2048, read=2048, pos=217088/221464
I (206169) HTTP_STREAM: req lengh=2048, read=2048, pos=219136/221464
I (206191) HTTP_STREAM: req lengh=2048, read=280, pos=221184/221464
W (206210) HTTP_STREAM: No more data, errno:0, total_bytes:221464, rlen = 0
D (206210)                              PLAY_AND_SAVE_FILE: HTTP_STREAM_FINISH_TRACK
I (206212) HTTP_STREAM: http_stream_next_track
```

And I also saw a warning message.
```
W (141519) AUDIO_EVT: There are no space to dispatch queue
```
What is this about and how to fix the problem?

jason-mao, 2022-08-08:
@KwangryeolPark We have fixed an issue by 2834242984d4a22a9c5a269ed7cbd6d192f1fc43, Could you help to test with latest master?

KwangryeolPark, 2022-08-08:
> @KwangryeolPark We have fixed an issue by 2834242984d4a22a9c5a269ed7cbd6d192f1fc43, Could you help to test with latest master?

Okay l, let me try