espressif/esp-adf #854: CONFIG_NEWLIB_NANO_FORMAT=y Causes VoIP Example to Crash

sidwarkd, 2022-07-25:
## Environment

- Audio development kit:      ESP32 Custom Board

- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version: v4.4.1
- [Required] ADF version: v2.3-309-gca673ab
- Build system:  CMake
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating system:    Linux
- Using an IDE?: No
- Power supply: USB

## Problem Description

Enabling NEWLIB_NANO_FORMAT=y in sdkconfig causes the application to crash while connecting to the SIP URI.

### Expected Behavior

The application should either fail to build if the newlib nano format config setting is set or it should fail gracefully with a runtime error.

### Actual Behavior

A LoadProhibited crash occurs resetting the device.

### Steps to Reproduce

1. Build the [VoIP Advanced Example](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/voip) project with a valid PBX URI and notice that it successfully registers.
2. Set CONFIG_NEWLIB_NANO_FORMAT=y in the sdkconfig file
3. Rebuild and run the application. Crash will occur 100% of the time while trying to connect to the PBX.

### Code to Reproduce This Issue

Easily reproduced with existing example code. Only need to set the config option as stated above.

ahhfzhang, 2022-08-04:
Hi @sidwarkd 

Can you provide some logs?

thanks.

sidwarkd, 2022-08-15:
@ahhfzhang I will get debug logs this week and post them here. I'm waiting on a Lyra test board to arrive so I can easily work with this. I found the issue on our custom hardware and can't reflash it at this time with the example code to repro the issue. I will post here as soon as I can.

sidwarkd, 2022-08-18:
Ok, I have logs and some additional information on this issue @ahhfzhang . I was not able to reproduce the issue with the voip example sdkconfig.defaults file. So I ported portions of the defaults file we are using in our product. Note that it includes the CONFIG_NEWLIB_NANO_FORMAT=y option. This will cause the out-of-the-box voip example to crash every time. If that option is turned off everything works fine. So it seems there is some other combination happening with other config settings we have. I've attached the sdkconfig file and here is the log output you requested:

```txt
I (27) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (27) boot: compile time 15:47:18
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00200000
I (90) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=570a8h (356520) map
I (247) esp_image: segment 1: paddr=000670d0 vaddr=3ffbdb60 size=0585ch ( 22620) load
I (256) esp_image: segment 2: paddr=0006c934 vaddr=40080000 size=036e4h ( 14052) load
I (262) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=fb16ch (1028460) map
I (635) esp_image: segment 4: paddr=0016b194 vaddr=400836e4 size=16e78h ( 93816) load
I (674) esp_image: segment 5: paddr=00182014 vaddr=50000000 size=00010h (    16) load
I (687) boot: Loaded app from partition at offset 0x10000
I (687) boot: Disabling RNG early entropy source...
I (699) psram: This chip is ESP32-D0WD
I (701) spiram: Found 64MBit SPI RAM device
I (701) spiram: SPI RAM mode: flash 40m sram 40m
I (705) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (712) cpu_start: Pro cpu up.
I (716) cpu_start: Starting app cpu, entry point is 0x400818d8
0x400818d8: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (1609) spiram: SPI SRAM memory test OK
I (1618) cpu_start: Pro cpu start user code
I (1618) cpu_start: cpu freq: 240000000
I (1618) cpu_start: Application information:
I (1621) cpu_start: Project name:     voip_app
I (1626) cpu_start: App version:      1
I (1631) cpu_start: Compile time:     Aug 18 2022 15:47:10
I (1637) cpu_start: ELF file SHA256:  feded6a8b679f310...
I (1643) cpu_start: ESP-IDF:          v4.4.1-dirty
I (1649) heap_init: Initializing. RAM available for dynamic allocation:
I (1656) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (1662) heap_init: At 3FFC6050 len 00019FB0 (103 KiB): DRAM
I (1668) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1675) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1681) heap_init: At 4009A55C len 00005AA4 (22 KiB): IRAM
I (1688) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1697) spi_flash: detected chip: generic
I (1700) spi_flash: flash io: dio
I (1706) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1715) spiram: Reserving pool of 18K of internal memory for DMA/internal allocations
I (1755) VOIP_EXAMPLE: [1.0] Initialize peripherals management
I (1765) VOIP_EXAMPLE: [1.1] Initialize and start peripherals
I (1765) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1775) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1785) VOIP_EXAMPLE: [1.2] Create and start input key service
I (1785) VOIP_EXAMPLE: [ 1.3 ] Create display service instance
W (1805) PERIPH_TOUCH: _touch_init
I (1805) VOIP_EXAMPLE: [ 2 ] Start codec chip
I (1805) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
E (1815) gpio: gpio_install_isr_service(449): GPIO isr service already installed
I (1855) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1855) ES8388_DRIVER: init,out:02, in:00
I (1865) AUDIO_HAL: Codec mode is 3, Ctrl:1
[... 330 lines cut ...]
#
# mbedTLS
#
#CONFIG_WPA_MBEDTLS_CRYPTO=n
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=n
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_ECDH_LEGACY_CONTEXT=y
CONFIG_MBEDTLS_ECP_RESTARTABLE=y
CONFIG_MBEDTLS_CMAC_C=y
# end of mbedTLS

#
# Newlib
#
CONFIG_NEWLIB_NANO_FORMAT=y
# end of Newlib

#
# Amazon Web Services IoT Platform
#
CONFIG_AWS_IOT_MQTT_TX_BUF_LEN=2048
CONFIG_AWS_IOT_MQTT_RX_BUF_LEN=4096
CONFIG_AWS_IOT_MQTT_NUM_SUBSCRIBE_HANDLERS=10
# end of Amazon Web Services IoT Platform
```