espressif/esp-adf #856: Proper HTTPS Audio Streaming Example

sidwarkd, 2022-07-27:
**Is your feature request related to a problem? Please describe.**
As far as I can tell every streaming audio example for the ADF requires the following sdkconfig settings:

  - CONFIG_ESP_TLS_INSECURE=y
  - CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
 
This means the connection is not properly secured. I can't seem to find documentation that shows how to do a proper TLS connection to stream audio using the ADF. The documentation does state that HTTPS is supported but provides no examples. From the documentation:

> `AUDIO_STREAM_READER` supports HTTP, HTTPS, HTTP Live Stream, and other protocols.

**Describe the solution you'd like**

Clear documentation and example code showing how to do an audio stream over properly secured and verified HTTPS.

**Describe alternatives you've considered**

It seems like this may be possible by manually digging through the struct members. For example `http_stream_init` returns an `audio_element_handle_t` struct. The function `http_stream_init` calls `audio_element_setdata` which sets the `data` member of the struct to an `http_stream_t*` struct. Therefore the return value from `http_stream_init` has a `data` member which is the `http_stream_t*` struct. The problem is `http_stream_t` is defined in http_stream.c and not in a header file that can be included.

That struct, in turn, has an `esp_http_client_handle_t` member named `client`. That client handle is just a typedef of `esp_http_client` which has a `connection_info_t` member named `connection_info`. It is finally on that struct buried deep in the hierarchy where we find a `char*` member named `cert_pem`. Presumably if we set that to a valid cert then true, verified HTTPS communication would occur. I have not had time to test this theory but something along the lines of the following may work:

```cpp
// define an equivalent http_stream_t struct in this file or in an included header

audio_element_handle_t http_stream_reader;
http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
http_cfg.multi_out_num = 1;
http_stream_reader = http_stream_init(&http_cfg);

http_stream_t *http_stream = (http_stream_t*)http_stream_reader->data;
http_stream->client->connection_info.cert_pem = "[A VALID CERT]";

// At this point when the http client is used to connect it would presumably use the cert to validate the connection.
```

sidwarkd, 2022-11-14:
@jason-mao Any word on whether this will be included in a future release or if there is a way to do HTTPS streaming today that I've just missed?

jason-mao, 2022-11-18:
@sidwarkd Sorry for reply late. Thank you for your suggestions.  I think we can provide an API such as `http_stream_set_server_cert` to change the certfication.

sidwarkd, 2022-11-19:
@jason-mao That would be very helpful. Will you update this thread when the change is available?

shootao, 2022-11-29:
Hi @sidwarkd 
Please use the patch below to make sure it solves your problem
[0001-feat-http_stream-support-https-links-for-http_stream.zip](https://github.com/espressif/esp-adf/files/10110964/0001-feat-http_stream-support-https-links-for-http_stream.zip)

sidwarkd, 2022-11-29:
@shootao Thank you for the quick response. I have applied the patch but cannot successfully stream. Perhaps I am doing something wrong in the code. Here are snippets of my code that I built using the streaming example from the docs but adding the new cert option.

```c
const char * digicert_ca = 
"-----BEGIN CERTIFICATE-----\n\
MIIEvjCCA6agAwIBAgIQBtjZBNVYQ0b2ii+nVCJ+xDANBgkqhkiG9w0BAQsFADBh\n\
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\n\
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD\n\
QTAeFw0yMTA0MTQwMDAwMDBaFw0zMTA0MTMyMzU5NTlaME8xCzAJBgNVBAYTAlVT\n\
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxKTAnBgNVBAMTIERpZ2lDZXJ0IFRMUyBS\n\
U0EgU0hBMjU2IDIwMjAgQ0ExMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKC\n\
AQEAwUuzZUdwvN1PWNvsnO3DZuUfMRNUrUpmRh8sCuxkB+Uu3Ny5CiDt3+PE0J6a\n\
qXodgojlEVbbHp9YwlHnLDQNLtKS4VbL8Xlfs7uHyiUDe5pSQWYQYE9XE0nw6Ddn\n\
g9/n00tnTCJRpt8OmRDtV1F0JuJ9x8piLhMbfyOIJVNvwTRYAIuE//i+p1hJInuW\n\
raKImxW8oHzf6VGo1bDtN+I2tIJLYrVJmuzHZ9bjPvXj1hJeRPG/cUJ9WIQDgLGB\n\
Afr5yjK7tI4nhyfFK3TUqNaX3sNk+crOU6JWvHgXjkkDKa77SU+kFbnO8lwZV21r\n\
eacroicgE7XQPUDTITAHk+qZ9QIDAQABo4IBgjCCAX4wEgYDVR0TAQH/BAgwBgEB\n\
/wIBADAdBgNVHQ4EFgQUt2ui6qiqhIx56rTaD5iyxZV2ufQwHwYDVR0jBBgwFoAU\n\
A95QNVbRTLtm8KPiGxvDl7I90VUwDgYDVR0PAQH/BAQDAgGGMB0GA1UdJQQWMBQG\n\
CCsGAQUFBwMBBggrBgEFBQcDAjB2BggrBgEFBQcBAQRqMGgwJAYIKwYBBQUHMAGG\n\
GGh0dHA6Ly9vY3NwLmRpZ2ljZXJ0LmNvbTBABggrBgEFBQcwAoY0aHR0cDovL2Nh\n\
Y2VydHMuZGlnaWNlcnQuY29tL0RpZ2lDZXJ0R2xvYmFsUm9vdENBLmNydDBCBgNV\n\
HR8EOzA5MDegNaAzhjFodHRwOi8vY3JsMy5kaWdpY2VydC5jb20vRGlnaUNlcnRH\n\
bG9iYWxSb290Q0EuY3JsMD0GA1UdIAQ2MDQwCwYJYIZIAYb9bAIBMAcGBWeBDAEB\n\
MAgGBmeBDAECATAIBgZngQwBAgIwCAYGZ4EMAQIDMA0GCSqGSIb3DQEBCwUAA4IB\n\
AQCAMs5eC91uWg0Kr+HWhMvAjvqFcO3aXbMM9yt1QP6FCvrzMXi3cEsaiVi6gL3z\n\
ax3pfs8LulicWdSQ0/1s/dCYbbdxglvPbQtaCdB73sRD2Cqk3p5BJl+7j5nL3a7h\n\
qG+fh/50tx8bIKuxT8b1Z11dmzzp/2n3YWzW2fP9NsarA4h20ksudYbj/NhVfSbC\n\
EXffPgK2fPOre3qGNm+499iTcc+G33Mw+nur7SpZyEKEOxEXGlLzyQ4UfaJbcme6\n\
ce1XR2bFuAJKZTRei9AqPCCcUZlM51Ke92sRKw2Sfh3oius2FkOH6ipjv3U/697E\n\
A7sKPPcw7+uvTPyLNhBzPvOk\n\
-----END CERTIFICATE-----";

audio_element_handle_t http_stream_reader, i2s_stream_writer, mp3_decoder;

audio_err_t audio_player_stream_mp3(const char *url)
{
    ESP_LOGI(TAG, "[2.0] Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline);

    ESP_LOGI(TAG, "[2.1] Create http stream to read data");
    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.cert_pem = digicert_ca;
    http_stream_reader = http_stream_init(&http_cfg);

    ESP_LOGI(TAG, "[2.2] Create i2s stream to write data to codec chip");
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_WRITER;
    i2s_stream_writer = i2s_stream_init(&i2s_cfg);

    ESP_LOGI(TAG, "[2.3] Create mp3 decoder to decode mp3 file");
    mp3_decoder_cfg_t mp3_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_decoder = mp3_decoder_init(&mp3_cfg);

    ESP_LOGI(TAG, "[2.4] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline, http_stream_reader, "http");
    audio_pipeline_register(pipeline, mp3_decoder,        "mp3");
    audio_pipeline_register(pipeline, i2s_stream_writer,  "i2s");

    ESP_LOGI(TAG, "[2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]");
    const char *link_tag[3] = {"http", "mp3", "i2s"};
    audio_pipeline_link(pipeline, &link_tag[0], 3);

    ESP_LOGI(TAG, "[2.6] Set up  uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)");
    audio_element_set_uri(http_stream_reader, "https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3");

    audio_pipeline_run(pipeline);
    return ESP_OK;
}
```
And I get the following error output:

```txt
E (8312) esp-tls: couldn't get hostname for :dl.espressif.com: getaddrinfo() returns 202, addrinfo=0x0
E (8312) esp-tls: Failed to open new connection
E (8322) TRANSPORT_BASE: Failed to open a new connection
E (8322) HTTP_CLIENT: Connection failed, sock < 0
E (8332) HTTP_STREAM: Failed to open http stream
E (8332) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,28674
```

Please let me know what I am doing wrong and thank you again for your help.

sidwarkd, 2022-11-29:
@shootao My apologies, the patch is working fine. My previous comment was because the device had not yet fully connected to Wi-Fi. Once I fixed that in the test I was able to stream just fine. Leaving my previous comment in case someone else gets a similar error they can know what to look for.

Thanks for providing a quick solution. I will do more testing and let you know if anything is not working properly.

closed 2022-12-02 by espressif-bot