espressif/esp-adf #857: About periph_sdcard_mode_t incorect define

Fuhua-Chen, 2022-07-28:
1. I wrote an init function for sdcard 4 line mode, it work well in esp-idf.
2. When i try to move to esp-adf, and run example **PIPELINR_REC_WAV_AMR_SDCARD**, i found sd card work fail. sdmmc send comm fail( same board with same micro sd card).

i found that the different between my init function and esp-adf api is **periph_sdcard_mode_t** 

**esp-adf**
```
/**
 * @brief  SD card mode, SPI, 1-line SD mode, 4-line SD mode
 *
 */
typedef enum {
    SD_MODE_SPI    = 0x0,  /*!< sd_card SPI*/
    SD_MODE_1_LINE = 0x1,  /*!< sd_card 1-line SD mode*/
    SD_MODE_4_LINE = 0x2,  /*!< sd_card 4-line SD mode*/
    SD_MODE_MAX,
} periph_sdcard_mode_t;
```

**code from sdcard.c**
```
sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
// slot_config.gpio_cd = g_gpio;
slot_config.width = mode & 0X01;
```
**It mean that esp-adf only support  1 line mode?**

**my own init code**
```
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
    // Set bus width to use:
    slot_config.width = 4;
    // On chips where the GPIOs used for SD card can be configured, set them in
    // the slot_config structure:
#if CONFIG_IDF_TARGET_ESP32S3 == 1
    slot_config.clk = SDMMC_PIN_CLK;
    slot_config.cmd = SDMMC_PIN_CMD;
    slot_config.d0 = SDMMC_PIN_D0;
    #if SDMMC_BUS_WIDTH == 4
    slot_config.d1 = SDMMC_PIN_D1;
    slot_config.d2 = SDMMC_PIN_D2;
    slot_config.d3 = SDMMC_PIN_D3;
    #endif  // CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
#endif
```

now i have fixed the issue by change as follow（mode left shift 1bit，not and 0x01）, and mount fatfs successfull:
```
sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
// slot_config.gpio_cd = g_gpio;
slot_config.width = mode << 0X01;
```

i think the correct define is below?

```
typedef enum {
    SD_MODE_SPI    = 0x0,  /*!< sd_card SPI*/
    SD_MODE_1_LINE = 0x2,  /*!< sd_card 1-line SD mode*/
    SD_MODE_4_LINE = 0x4,  /*!< sd_card 4-line SD mode*/
    SD_MODE_MAX,
} periph_sdcard_mode_t;
```

Fuhua-Chen, 2022-07-28:
i wrote an incorrect code, the last should be 
```
typedef enum {
    SD_MODE_SPI    = 0x0,  /*!< sd_card SPI*/
    SD_MODE_1_LINE = 0x1,  /*!< sd_card 1-line SD mode*/
    SD_MODE_4_LINE = 0x4,  /*!< sd_card 4-line SD mode*/
    SD_MODE_MAX,
} periph_sdcard_mode_t;
```

jason-mao, 2022-08-04:
@Fuhua-Chen Thanks for your report. Let we check and fix.

closed 2022-08-05 by espressif-bot

Fuhua-Chen, 2022-08-11:
i found code in sdmmc_periph.h,  line 31
`
typedef struct {
    uint8_t width;          /*!< Maximum supported slot width (1, 4, 8) */
    uint8_t card_detect;    /*!< Card detect signal in GPIO Matrix */
    uint8_t write_protect;  /*!< Write protect signal in GPIO Matrix */
    uint8_t card_int;       /*!< Card interrupt signal in GPIO Matrix */
} sdmmc_slot_info_t;
`
when use 4-line, width must be SD_MODE_4_LINE = 0x4, and 
`
slot_config.width = mode & 0X01;  
`
should not mode & 0x01;

jason-mao, 2022-08-11:
> i found code in sdmmc_periph.h, line 31
>
> ```
> typedef struct {
>     uint8_t width;          /*!< Maximum supported slot width (1, 4, 8) */
>     uint8_t card_detect;    /*!< Card detect signal in GPIO [...]

Yes, you are right.  Change to `slot_config.width = mode ;  `