espressif/esp-adf #859: ESP-ADF getting started project build error

alibukharai, 2022-07-31:
----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      ESP32-LyraT 4.3
- Operating system:     Windows 10

## Problem Description

I am new to LyraT board. I am trying to run the getting started commands from documents, it gives me a strange error related to cmake. I have attached a screenshot.  
[image]

### Code to Reproduce This Issue

in Cmd 

cd C:\esp
git clone --recursive https://github.com/espressif/esp-adf.git   or 
git clone --recursive https://gitee.com/EspressifSystems/esp-adf.git 
set ADF_PATH=C:\esp\esp-adf
echo %ADF_PATH%

in idfcmd 

C:\esp\esp-idf\export.bat
cd C:\esp
xcopy /e /i C:\esp\esp-adf\examples\get-started\play_mp3_control

cd C:\esp\play_mp3_control
idf.py set-target esp32
idf.py menuconfig

jason-mao, 2022-08-04:
@alibukharai I not have found any problems as your commands, did you saw right path with `echo %ADF_PATH%`and `echo %IDF_PATH%` before `idf.py set-target esp32`?

alibukharai, 2022-08-04:
> @alibukharai I not have found any problems as your commands, did you saw right path with `echo %ADF_PATH%`and `echo %IDF_PATH%` before `idf.py set-target esp32`?

after restarting the system this problem is solved

jason-mao, 2022-08-04:
@alibukharai Thanks for your reply.

closed 2022-08-04 by jason-mao