espressif/esp-adf #860: Error in cloning ESP-ADF

alibukharai, 2022-07-31:
## Environment

- Audio development kit:      [ESP32-LyraT|
- Operating system:     Windows10

## Problem Description

during the installation and cloning process of ADF, there shows an error. I have attached a screenshot.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

1. git clone --recursive https://github.com/espressif/esp-adf.git

[image]

jason-mao, 2022-08-04:
@alibukharai It's looks like can't connect to GitHub server. Can you open GitHub in your browser？

alibukharai, 2022-08-04:
> 

yes I resolved the issue by setting IP address

jason-mao, 2022-08-04:
@alibukharai Thanks for your update.

closed 2022-08-04 by jason-mao