espressif/esp-adf #862: Crash from resample task

espradio, 2022-08-03:
Observed following crash from resample task in long run

Env:
ESP32-S3-N8R8
ADF version: 478513b592d55a1ceef46acb20bd61dd610569b1 (latest as of Aug 03 2022)
IDF verison: 
```

commit 1329b19fe494500aeb79d19b27cfd99b40c37aec (HEAD, tag: v4.4.1)
Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
Date:   Sat Apr 16 09:43:14 2022 +0800

    versions: Update version to 4.4.1

```

Crash log

```
==================== THREAD 1 (TCB: 0x3fcba940, name: 'resample') =====================
#0  0x40382cf9 in compare_and_set_native (set=<synthetic pointer>, compare=3007315967, addr=0xaaaf1f6e) at esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
#1  spinlock_acquire (timeout=-1, lock=0xaaaf1f6e) at esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
#2  xPortEnterCriticalTimeout (mux=0xaaaf1f6e, timeout=-1) at esp-idf/components/freertos/port/xtensa/port.c:288
#3  0x4037fad2 in vPortEnterCritical (mux=0xaaaf1f6e) at esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
#4  xQueueReceive (xQueue=0xaaaf1f1a, pvBuffer=0x3fcba760, xTicksToWait=<optimized out>) at esp-idf/components/freertos/queue.c:1400
#5  0x4200e949 in audio_event_iface_waiting_cmd_msg (evt=0x3d86f6c0) at esp-adf/components/audio_pipeline/audio_event_iface.c:244
#6  0x4200df1a in audio_element_task (pv=0x3d80f5e4) at esp-adf/components/audio_pipeline/audio_element.c:473
#7  0x40382b90 in vPortTaskWrapper (pxCode=0x4200de78 <audio_element_task>, pvParameters=0x3d80f5e4) at esp-idf/components/freertos/port/xtensa/port.c:131
```
```

==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fcba940, name: 'resample', GDB name: 'process 1070311744'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0xaaaf1f6e
epc1           0x4203b3fb
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40382cf9          0x40382cf9 <xPortEnterCriticalTimeout+69>
lbeg           0x40056f5c          1074098012
lend           0x40056f72          1074098034
lcount         0x0                 0
sar            0x4                 4
ps             0x60523             394531
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8037fad2          -2143814958
a1             0x3fcba700          1070311168
a2             0xaaaf1f6e          -1431363730
a3             0xffffffff          -1
a4             0x400               1024
a5             0x3d833b98          1032010648
a6             0x400               1024
a7             0x3d80f520          1031861536
a8             0xabab              43947
a9             0xabab              43947
a10            0x60520             394528
a11            0xcdcd              52685
a12            0xb33fffff          -1287651329
a13            0x60523             394531
a14            0x1                 1
a15            0x400               1024
```

espradio, 2022-08-04:
@jason-mao Looks like it is issue from power. It was usb self power.

Is there  anyway programmatically to detect power quality is not good .Some reason brownout detector not fired. Is monitoring voltage via divider + ADC scheme helps?

jason-mao, 2022-08-05:
@espradio The main power consumption is WiFi and sounds. As your said, I think it's a method, but the practical effect should not be good.

closed 2022-08-05 by jason-mao