espressif/esp-adf #864: wwe工程不用唤醒词不能进行英文语音识别吗

Zhengqinze05, 2022-08-10:
你好，我使用的是esp32_s3_korvo2_v3的板子，IDF的版本是idf4.4
我想使用英文语音识别，并且不使用唤醒词，但是我在menuconfig中发现关闭了唤醒词。并且选择了英文语音识别，它没有英文的关键词模型选择
[image]

jason-mao, 2022-08-11:
@Zhengqinze05 你说的情况可以在 1mic 的情况下使用。但不推荐。

Zhengqinze05, 2022-08-11:
我目前是使用1mic的，请问如何开启呢，有什么风险吗

jason-mao, 2022-08-11:
参考[esp_afe_config.h](https://github.com/espressif/esp-sr/blob/98900ad8ef5c24f95a31fca22e331634627c7338/include/esp32s3/esp_afe_config.h#L77)。识别效果不理想。

closed 2022-08-17 by jason-mao

Zhengqinze05, 2022-08-18:
具体如何操作呢，我是使用的adf版本里面的esp-sr库，没有esp_afe_config.h这个文件，在adf版本里，我使用menuconfig关闭了唤醒词，在wwe工程中#define WAKENET_ENABLE      (false) ，然而提示我
wakenet is NULL, please use set_wakenet function to set wakenet.并且说出关键词也没有反应
[image]
请问还需要修改什么地方吗