espressif/esp-adf #866: [PLAY MUTIPLE FILE MP3]

quoctrung99, 2022-08-12:
I have a problem with run mutiple file mp3 esp adf. 
please help me

jason-mao, 2022-08-12:
@quoctrung99 Please add more information for your issue, such as what is file? how to play?

quoctrung99, 2022-08-12:
For example i have 3 mp3 files,  I need to read each file in turn.

jason-mao, 2022-08-15:
@quoctrung99 What kind of behavior do you expect？

quoctrung99, 2022-08-15:
i'm following the example play_mp3_control_example.c

quoctrung99, 2022-08-15:
i need to concatenate all mp3 files to play. For example I have a number as input: 101. I need play file1 ->file0->file1

jason-mao, 2022-08-17:
Can there be silence between file1 and file0?

quoctrung99, 2022-08-17:
yes it is possible

and-elf, 2022-09-10:
Use a playlist?

quoctrung99, 2022-09-12:
yes, please help me

and-elf, 2022-09-12:
Have you looked at this example: https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_sdcard_mp3_control/main/play_sdcard_mp3_control_example.c You'll also have to learn how to use the playlist API. Basically create a playlist and then add paths to it based on command/user input. Have you written the parser (code that handles input)? Split a string into characters and then use sprintf () to turn them into file names should be easy enough

closed 2022-11-18 by jason-mao