espressif/esp-adf #873: wwe出现AFE_SR: ERROR! rb_out slow!!!

Zhengqinze05, 2022-08-23:
你好，我使用的是esp32_s3_korvo2_v3板子，wwe工程。我将语音识别设置成单麦，并且不使用唤醒词时，当命令词被检测到时，出现了AFE_SR: ERROR! rb_out slow!!!错误，请问这是什么情况呢
[image]

jason-mao, 2022-08-26:
这个信息是AFE数据处理不及时导致的溢出，所以他体现的是CPU loading不均衡。`audio_sys_get_real_time_stats` 可以获取CPU 的状态。

jason-mao, 2022-08-26:
Same as #870

Zhengqinze05, 2022-08-26:
@jason-mao 感谢您的回复，我仅仅是在esp_afe_config.h将配置改为单麦克风
#define AFE_CONFIG_DEFAULT() { \
    .aec_init = true, \
    .se_init = true, \
    .vad_init = true, \
    .wakenet_init = true, \
    .voice_communication_init = false, \
    .voice_communication_agc_init = false, \
    .voice_communication_agc_gain = 15, \
    .vad_mode = VAD_MODE_3, \
    .wakenet_model_name = NULL, \
    .wakenet_mode = DET_MODE_2CH_90, \
    .afe_mode = SR_MODE_LOW_COST, \
    .afe_perferred_core = 0, \
    .afe_perferred_priority = 5, \
    .afe_ringbuf_size = 50, \
    .memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM, \
    .agc_mode = AFE_MN_PEAK_AGC_MODE_2, \
    .pcm_config.total_ch_num = 2, \
    .pcm_config.mic_num = 1, \
    .pcm_config.ref_num = 1, \
    .pcm_config.sample_rate = 16000, \
}
#endif

在i2s_stream.h文件中将.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,设置为单通道，并且是wwe的main.c中设置了i2s_cfg.i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
其他并没有明显更改，但是当唤醒之后，要进行命令词识别时，依旧会有这样的报错。也没有其他core报错的log
使用audio_sys_get_real_time_stats报的信息如下
[0;32mI (4554) AUDIO_SYS: | Task              | Run Time    | Per | Prio | HWM       | State   | CoreId   | Stack [0m
[0;32mI (4554) AUDIO_SYS: | recorder_task     | 2438        | 0%  | 10   | 3568      | Ready   | 1        | Extr[0m
[0;32mI (4564) AUDIO_SYS: | feed_task         | 131977      | 6%  | 5    | 5116      | Ready   | 1        | Extr[0m
[0;32mI (4574) AUDIO_SYS: | main              | 1567        | 0%  | 1    | 920       | Running | 0        | Intr[0m
[0;32mI (4584) AUDIO_SYS: | IDLE              | 722264      |36%  | 0    | 736       | Ready   | 1        | Intr[0m
[0;32mI (4594) AUDIO_SYS: | IDLE              | 686526      |34%  | 0    | 732       | Ready   | 0        | Intr[0m
[0;32mI (4604) AUDIO_SYS: | esp_periph        | 2858        | 0%  | 5    | 3112      | Blocked | 0        | Extr[0m
[0;32mI (4614) AUDIO_SYS: | afe_mase          | 204456      |10%  | 5    | 7432      | Blocked | 0        | Intr[0m
[0;32mI (4624) AUDIO_SYS: | fetch_task        | 139032      | 6%  | 5    | 4312      | Blocked | 1        | Extr[0m
[0;32mI (4634) AUDIO_SYS: | button_task       | 21428       | 1%  | 10   | 1468      | Blocked | 0        | Intr[0m
[0;32mI (4644) AUDIO_SYS: | Tmr Svc           | 0           | 0%  | 1    | 1364      | Blocked | 0        | Intr[0m
[0;32mI (4664) AUDIO_SYS: | ipc0              | 0           | 0%  | 1    | 764       | Blocked | 0        | Intr[0m
[0;32mI (4674) AUDIO_SYS: | media_task        | 0           | 0%  | 6    | 2992      | Blocked | 0        | Extr[0m
[0;32mI (4684) AUDIO_SYS: | ipc1              | 0           | 0%  | 24   | 704       | Blocked | 1        | Intr[0m
[0;32mI (4694) AUDIO_SYS: | i2s               | 11540       | 0%  | 23   | 2540      | Blocked | 0        | Intr[0m
[0;32mI (4704) AUDIO_SYS: | filter            | 67514       | 3%  | 5    | 2868      | Blocked | 0        | Extr[0m
[0;32mI (4714) AUDIO_SYS: | esp_timer         | 0           | 0%  | 22   | 3348      | Suspended | 0        | Intr[0m
请问是ESP32S3不支持单麦语音识别吗

closed 2022-09-05 by espressif-bot