espressif/esp-adf #874: voip can not normal running when enable ota code

hnhkj, 2022-08-26:
I reference examples/korvo_du1960 enable app_ota_start() function. and reference examples/advanced_examples/voip code enable voip function.
now when I enable app_ota_start() function. voip can not normal running. when I disable ota function. voip running. I dont know if they can not running same time. I found examples/korvo_du1960 default disable app_ota_start() function. if then can not running in same time?

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x17c0
load:0x403b6000,len:0xe64
load:0x403ba000,len:0x30e4
entry 0x403b6280
I (25) boot: ESP-IDF v4.4.1-149-g009e66e625-dirty 2nd stage bootloader
I (25) boot: compile time 08:46:03
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00010000 002b0000
I (94) boot:  4 ota_1            OTA app          00 11 002c0000 002b0000
I (102) boot:  5 model            Unknown data     01 82 00570000 0040e000
I (109) boot:  6 flash_tone       Unknown data     01 ff 0097e000 00080000
I (117) boot: End of partition table
I (121) boot: No factory image, trying OTA 0
I (126) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=44974h (280948) map
I (177) esp_image: segment 1: paddr=0005499c vaddr=3fca0d80 size=04778h ( 18296) load
I (181) esp_image: segment 2: paddr=0005911c vaddr=40378000 size=06efch ( 28412) load
I (188) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=ffdc0h (1048000) map
I (350) esp_image: segment 4: paddr=0015fde8 vaddr=4037eefc size=11e80h ( 73344) load
I (364) esp_image: segment 5: paddr=00171c70 vaddr=50000000 size=00010h (    16) load
I (374) boot: Loaded app from partition at offset 0x10000
I (404) boot: Set actual ota_seq=1 in otadata[0]
I (404) boot: Disabling RNG early entropy source...
I (415) opi psram: vendor id : 0x0d (AP)
I (415) opi psram: dev id    : 0x02 (generation 3)
I (416) opi psram: density   : 0x03 (64 Mbit)
I (419) opi psram: good-die  : 0x01 (Pass)
I (424) opi psram: Latency   : 0x01 (Fixed)
I (429) opi psram: VCC       : 0x01 (3V)
I (433) opi psram: SRF       : 0x01 (Fast Refresh)
I (439) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (444) opi psram: BurstLen  : 0x01 (32 Byte)
I (449) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (455) opi psram: DriveStrength: 0x00 (1/1)
W (460) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (471) spiram: Found 64MBit SPI RAM device
I (475) spiram: SPI RAM mode: sram 80m
I (479) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (487) cpu_start: Pro cpu up.
I (490) cpu_start: Starting app cpu, entry point is 0x403795e4
0x403795e4: call_start_cpu1 at /root/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (782) spiram: SPI SRAM memory test OK
I (791) cpu_start: Pro cpu start user code
I (791) cpu_start: cpu freq: 240000000
I (791) cpu_start: Application information:
I (794) cpu_start: Project name:     voice_extender
I (799) cpu_start: App version:      v1.5.2
I (804) cpu_start: Compile time:     Aug 25 2022 08:46:04
I (810) cpu_start: ELF file SHA256:  1761300cbf9d6052...
I (816) cpu_start: ESP-IDF:          v4.4.1-149-g009e66e625-dirty
[... 207 lines cut ...]
    EventBits_t bits = xEventGroupWaitBits(WIFI_CONNECTED_FLAG, WIFI_CONNECTED_BIT, true, false, WIFI_WAIT_CONNECT_TIME_MS);
    if (bits & WIFI_CONNECTED_BIT) {
        app_ota_start();
    } else {
        ESP_LOGW(TAG, "WIFI  connection timeout(%dms), skipped OTA service", WIFI_WAIT_CONNECT_TIME_MS);
    }
    vEventGroupDelete(WIFI_CONNECTED_FLAG);
    WIFI_CONNECTED_FLAG = NULL;
    ota_flag = false;

    // voip
    ESP_LOGI(TAG,"app_voip_start [%d]",__LINE__);
    app_voip_start();

    // step 7. setup the esp_player
#if CONFIG_BT_ENABLED
    //app_player_init(NULL, audio_player_callback, set, user_a2dp_sink_cb);
#else
    //app_player_init(NULL, audio_player_callback, set, NULL);
#endif

    //audio_player_vol_set(40);
}

```

jason-mao, 2022-08-26:
@hnhkj  As your log 
```c
E (18060) AUDIO_ELEMENT: [filter] audio_thread_create failed
E (18060) AUDIO_THREAD: Error creating task i2s
```
looks like do not use the [patches](https://github.com/espressif/esp-adf/tree/master/idf_patches), please check that.

hnhkj, 2022-08-30:
master branch, the latest commit( b733b05341c6933fb492d2c56080f601d77ab818) , apply idf_patches/idf_v4.4_i2s_c3_pdm_tx.patch fail.

```
root@ba95e0759df2:~/esp-adf/esp-idf# git apply ../idf_patches/idf_v4.4_i2s_c3_pdm_tx.patch 
error: patch failed: components/hal/i2s_hal.c:133
error: components/hal/i2s_hal.c: patch does not apply
```

hnhkj, 2022-09-01:
Enable follow config. OTA and voip working. maybe memory overflow.
```
idf.py menuconfig->Component Config->ESPS32S3-Specific->Support for external, SPI-connected RAM->SPI RAM config->Try to allocate memories of WiFi 
```