espressif/esp-adf #875: VoIP项目,无法链接SIP服务器

jiafeng-1, 2022-08-29:
报错信息,一直无法注册. SIP服务器采用的freeswitch.
```c
I (8638) SIP: [1970-01-01/00:00:03]=======================>>
W (11648) SIP: Retransmit again
I (11648) SIP: [1970-01-01/00:00:05]=======WRITE 0603 bytes>>
I (11648) SIP: 

REGISTER sip:1000@192.168.81.133:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.31.71:15342;branch=z9hG4bK--99381185;rport
From: <sip:1000@192.168.81.133:5060>;tag=1274637133
To: <sip:1000@192.168.81.133:5060>
Contact: <sip:1000@192.168.31.71:15342;transport=UDP>
Max-Forwards: 70
Call-ID: 0B22143B58E3A09352C6BA753304DAF64C2D2F164BE1
CSeq: 1 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml

I (11698) SIP: [1970-01-01/00:00:05]=======================>>
W (14708) SIP: read timeout
E (14708) SIP: Error register device
W (14708) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1689
W (15708) SIP: CHANGE STATE FROM 0, TO 0, :func: sip_reconnect:355
I (16908) SIP: Conecting...
W (16908) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1674
I (16908) SIP: [1970-01-01/00:00:08]=======WRITE 0605 bytes>>
I (16908) SIP: 
```

jiafeng-1, 2022-08-29:
1.关闭了防火墙sudo service firewalld status
● firewalld.service - firewalld - dynamic firewall daemon
   Loaded: loaded (/lib/systemd/system/firewalld.service; disabled; vendor prese
   Active: inactive (dead)
     Docs: man:firewalld(1)
2.使用eyeBeam软件，能够注册。在freeswitch查询到只有这一个注册信息。
Registrations:
=================================================================================================
Call-ID:    	OTk4ZDRjODE0YzI2ZTVkYmZlOGI5MzRmNDVhNjAwYTE.
User:       	1001@192.168.81.133
Contact:    	"1001" <sip:1001@192.168.81.1:25504;rinstance=0a4d3c60b397156d>
Agent:      	eyeBeam release 1011d stamp 40820
Status:     	Registered(UDP)(unknown) EXP(2022-08-29 01:09:53) EXPSECS(2625)
Ping-Status:	Reachable
Ping-Time:	0.00
Host:       	ubuntu
IP:         	192.168.81.1
Port:       	25504
Auth-User:  	1001
Auth-Realm: 	192.168.81.133
MWI-Account:	1001@192.168.81.133

Total items returned: 1

jiafeng-1, 2022-08-29:
3.freeswitch所在虚拟机，可以ping通esp32-s3。
```c
ping 192.168.31.71
PING 192.168.31.71 (192.168.31.71) 56(84) bytes of data.
From 192.168.31.227 icmp_seq=3 Destination Host Unreachable
64 bytes from 192.168.31.71: icmp_seq=5 ttl=128 time=960 ms
64 bytes from 192.168.31.71: icmp_seq=6 ttl=128 time=3.52 ms
64 bytes from 192.168.31.71: icmp_seq=9 ttl=128 time=95.0 ms
64 bytes from 192.168.31.71: icmp_seq=10 ttl=128 time=117 ms
64 bytes from 192.168.31.71: icmp_seq=11 ttl=128 time=38.0 ms
64 bytes from 192.168.31.71: icmp_seq=14 ttl=128 time=59.9 ms
^C
```
请问这会是什么原因导致无法注册呢？一直报错下面的内容
```c
I (438208) SIP: [1970-01-01/00:07:08]=======================>>
W (441218) SIP: read timeout
E (441218) SIP: Error register device
W (441218) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1689
W (531218) SIP: CHANGE STATE FROM 0, TO 0, :func: sip_reconnect:355
I (532418) SIP: Conecting...
W (532418) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1674
I (532418) SIP: [1970-01-01/00:08:25]=======WRITE 0605 bytes>>
I (532418) SIP: 

```

ahhfzhang, 2022-09-02:
Hi @jiafeng-1 

这个问题是设备发送给服务器的包没有收到回复，也可能是服务器没有收到。
帮忙做一些测试

1. 看起来服务器和客户端并不在一个网段，确认是否能够访问到？ -- ping的地址(192.168.31.x)和服务器(192.168.81.x)也不在一个网段
2. 保证在同一网段下测一下
3. 确认一下 media_lib_add_default_adapter 是不是没有调用，这个没有调用，网络包会发送不出去。

Thanks.

jiafeng-1, 2022-09-06:
hi@ahhfzhang

问题找到了，我的freeswitch搭在了虚拟机上，虚拟机网络，跟主机不在一个网段。虽然从虚拟机ping 板卡能ping通。但是板子到虚拟机是不通的。

解决办法就是将虚拟机网络与主机设置为同一网段。

closed 2022-09-06 by jiafeng-1