espressif/esp-adf #877: ota 升级错误

hnhkj, 2022-08-29:
OTA升级错误，我参考examples/korvo_du1906/main/app_ota_upgrade.c代码实现对ESP32-S3-korvo2升级。

我的固件升级2部分，tone数据和app固件。在之前的版本中，固件升级正常。但是我更新esp-adf到最新的commit后，更新功能失败。

```c
void app_ota_start()
{
    ESP_LOGI(TAG, "Create OTA service");
    OTA_FLAG = xEventGroupCreate();
    ota_service_config_t ota_service_cfg = OTA_SERVICE_DEFAULT_CONFIG();
    ota_service_cfg.task_stack = OTA_SERVICE_STACK_SIZE;
    ota_service_cfg.evt_cb = ota_service_cb;
    ota_service_cfg.cb_ctx = NULL;
    periph_service_handle_t ota_service = ota_service_create(&ota_service_cfg);
    ota_upgrade_ops_t upgrade_list[] = {
        {
            {
                ESP_PARTITION_TYPE_DATA,
                "flash_tone",
                "https://xxx.com/ESSFirmwareUpdate/ve/tone/audio_tone.bin",
                NULL
            },
            NULL,
            NULL,
            NULL,
            NULL,
            false,
            false
        },
        //{
        //    {
        //        ESP_PARTITION_TYPE_DATA,
        //        "profile",
        //        "file://sdcard/profile.bin",
        //        NULL
        //    },
        //    NULL,
        //    NULL,
        //    NULL,
        //    NULL,
        //    false,
        //    false
        //},
        {
            {
                ESP_PARTITION_TYPE_APP,
                NULL,
                "https://xxx.com/ESSFirmwareUpdate/ve/firmware/app.bin",
                (char *)server_cert_pem_start,
            },
            NULL,
            NULL,
            NULL,
            NULL,
            true,
            false
        }
    };
```

#### 更新失败消息

```sh
I (3915) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
W (3925) WIFI_SERV: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
W (3935) wifi:<ba-add>idx:0 (ifx:0, 9c:a6:15:e9:aa:53), tid:0, ssn:2601, winSize:64
I (3975) wifi:BcnInt:102400, DTIM:1
I (4925) esp_netif_handlers: sta ip: 192.168.0.107, mask: 255.255.255.0, gw: 192.168.0.1
I (4925) WIFI_SERV: Got ip:192.168.0.107
W (4925) WIFI_SERV: STATE type:2, pdata:0x0, len:0
I (4945) VOICE_EXTEMDER: PERIPH_WIFI_CONNECTED [116]
I (4945) APP_OTA_UPGRADE: Create OTA service
I (4955) APP_OTA_UPGRADE: Start OTA service
I (4965) OTA_DEFAULT: data upgrade uri https://call4help.uk/ESSFirmwareUpdate/ve/tone/audio_tone.bin
W (4975) TONE_STREAM: No more data,ret:0 ,info.byte_pos:6120
I (8175) HTTP_STREAM: total_bytes=379658
I (11275) APP_OTA_UPGRADE: format 1 : 1
I (11275) APP_OTA_UPGRADE: imcoming magic_word F55F9876, project_name ESP_TONE_BIN
I (11275) APP_OTA_UPGRADE: current version v1.2.0, incoming version v1.2.0
W (11285) APP_OTA_UPGRADE: The incoming version is same as or lower than the running version
E (11295) OTA_SERVICE: No need to upgrade
E (11305) APP_OTA_UPGRADE: List id: 0, OTA failed, result: 589828
E (11595) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (11595) esp-tls: create_ssl_handle failed
E (11595) esp-tls: Failed to open new connection
E (11595) TRANSPORT_BASE: Failed to open a new connection
E (11605) HTTP_CLIENT: Connection failed, sock < 0
E (11605) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (11615) esp_https_ota: Failed to establish HTTP connection
E (11625) OTA_DEFAULT: ESP HTTPS OTA Begin failed
E (11625) OTA_SERVICE: OTA prepared fail
E (11635) APP_OTA_UPGRADE: List id: 1, OTA failed, result: -1
W (11635) OTA_SERVICE: OTA_END!
W (11645) APP_OTA_UPGRADE: upgrade finished, Please check the result of OTA
I (11655) VOICE_EXTEMDER: app_voip_start [284]
I (11655) APP_VOIP: Create SIP Service
I (11715) APP_VOIP: SIP_EVENT_REGISTERED
W (11725) TONE_STREAM: No more data,ret:0 ,info.byte_pos:12427
I (31665) APP_SYS_TOOLS: Func:sys_monitor_task, Line:36, MEM Total:8353435 Bytes, Inter:22575 Bytes, Dram:22575 Bytes

I (51665) APP_SYS_TOOLS: Func:sys_monitor_task, Line:36, MEM Total:8353419 Bytes, Inter:22575 Bytes, Dram:22575 Bytes

I (71665) APP_SYS_TOOLS: Func:sys_monitor_task, Line:36, MEM Total:8353419 Bytes, Inter:22575 Bytes, Dram:22575 Bytes

I (91665) APP_SYS_TOOLS: Func:sys_monitor_task, Line:36, MEM Total:8353403 Bytes, Inter:22575 Bytes, Dra
```

jason-mao, 2022-08-30:
@hnhkj 您这边的软件环境帮提供一下，korvo_du1906 在最近的IDF版本上已经不支持了。

hnhkj, 2022-08-30:
esp-adf master branch: b733b05341c6933fb492d2c56080f601d77ab818

hnhkj, 2022-08-30:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17c0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x30d4
entry 0x403c9980
I (25) boot: ESP-IDF v4.4.2-155-g23d5a582cb-dirty 2nd stage bootloader
I (25) boot: compile time 01:20:46
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00010000 002b0000
I (94) boot:  4 ota_1            OTA app          00 11 002c0000 002b0000
I (102) boot:  5 model            Unknown data     01 82 00570000 0040e000
I (109) boot:  6 flash_tone       Unknown data     01 ff 0097e000 00080000
I (117) boot: End of partition table
I (121) boot: No factory image, trying OTA 0
I (126) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=44e4ch (282188) map
I (177) esp_image: segment 1: paddr=00054e74 vaddr=3fca6120 size=06a64h ( 27236) load
I (182) esp_image: segment 2: paddr=0005b8e0 vaddr=40378000 size=04738h ( 18232) load
I (186) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=10a8d4h (1091796) map
I (357) esp_image: segment 4: paddr=0016a8fc vaddr=4037c738 size=199e4h (104932) load
I (377) esp_image: segment 5: paddr=001842e8 vaddr=50000000 size=00010h (    16) load
I (390) boot: Loaded app from partition at offset 0x10000
I (417) boot: Set actual ota_seq=1 in otadata[0]
I (418) boot: Disabling RNG early entropy source...
I (428) opi psram: vendor id : 0x0d (AP)
I (429) opi psram: dev id    : 0x02 (generation 3)
I (429) opi psram: density   : 0x03 (64 Mbit)
I (432) opi psram: good-die  : 0x01 (Pass)
I (437) opi psram: Latency   : 0x01 (Fixed)
I (442) opi psram: VCC       : 0x01 (3V)
I (447) opi psram: SRF       : 0x01 (Fast Refresh)
I (452) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (457) opi psram: BurstLen  : 0x01 (32 Byte)
I (462) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (468) opi psram: DriveStrength: 0x00 (1/1)
W (473) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (484) spiram: Found 64MBit SPI RAM device
I (488) spiram: SPI RAM mode: sram 80m
I (493) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (500) cpu_start: Pro cpu up.
I (503) cpu_start: Starting app cpu, entry point is 0x40379608
0x40379608: call_start_cpu1 at /root/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (795) spiram: SPI SRAM memory test OK
I (804) cpu_start: Pro cpu start user code
I (804) cpu_start: cpu freq: 240000000
I (804) cpu_start: Application information:
I (807) cpu_start: Project name:     voice_extender
I (812) cpu_start: App version:      v1.5.2
I (817) cpu_start: Compile time:     Aug 30 2022 08:32:33
I (823) cpu_start: ELF file SHA256:  a0bca83bbff8191c...
I (829) cpu_start: ESP-IDF:          v4.4.2-155-g23d5a582cb-dirty
I (836) heap_init: Initializing. RAM available for dynamic allocation:
I (843) heap_init: At 3FCB2338 len 000373D8 (220 KiB): D/IRAM
I (850) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[... 84 lines cut ...]
I (8323) APP_OTA_UPGRADE: imcoming magic_word F55F9876, project_name ESP_TONE_BIN
I (8323) APP_OTA_UPGRADE: current version v1.2.0, incoming version v1.2.0
W (8323) APP_OTA_UPGRADE: The incoming version is same as or lower than the running version
E (8333) OTA_SERVICE: No need to upgrade
E (8343) APP_OTA_UPGRADE: List id: 0, OTA failed, result: 589828
W (8343) esp_https_ota: Continuing with insecure option because CONFIG_OTA_ALLOW_HTTP is set.
E (8693) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (8693) esp-tls: create_ssl_handle failed
E (8693) esp-tls: Failed to open new connection
E (8703) TRANSPORT_BASE: Failed to open a new connection
E (8713) HTTP_CLIENT: Connection failed, sock < 0
E (8713) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (8723) esp_https_ota: Failed to establish HTTP connection
E (8723) OTA_DEFAULT: ESP HTTPS OTA Begin failed
E (8733) OTA_SERVICE: OTA prepared fail
E (8733) APP_OTA_UPGRADE: List id: 1, OTA failed, result: -1
W (8743) OTA_SERVICE: OTA_END!
W (8743) APP_OTA_UPGRADE: upgrade finished, Please check the result of OTA
I (8753) VOICE_EXTEMDER: app_voip_start [284]
I (8763) APP_VOIP: Create SIP Service
I (8803) APP_VOIP: SIP_EVENT_REGISTERED
W (8823) TONE_STREAM: No more data,ret:0 ,info.byte_pos:12427
I (28763) APP_SYS_TOOLS: Func:sys_monitor_task, Line:36, MEM Total:8353459 Bytes, Inter:22595 Bytes, Dram:22595 Bytes

```

hnhkj, 2022-08-31:
examples/ota sample error same.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f74
entry 0x403c9954
I (25) boot: ESP-IDF v4.4.2-155-g23d5a582cb-dirty 2nd stage bootloader
I (25) boot: compile time 11:08:51
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 16MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 ota_0            OTA app          00 10 00010000 00100000
I (89) boot:  4 ota_1            OTA app          00 11 00110000 00100000
I (96) boot:  5 flash_tone       Unknown data     01 ff 00210000 00100000
I (104) boot: End of partition table
I (108) boot: No factory image, trying OTA 0
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=22e68h (142952) map
I (147) esp_image: segment 1: paddr=00032e90 vaddr=3fc97a40 size=042f4h ( 17140) load
I (151) esp_image: segment 2: paddr=0003718c vaddr=40374000 size=08e8ch ( 36492) load
I (161) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=ad3e8h (709608) map
I (289) esp_image: segment 4: paddr=000ed410 vaddr=4037ce8c size=0abb4h ( 43956) load
I (299) esp_image: segment 5: paddr=000f7fcc vaddr=50000000 size=00010h (    16) load
I (307) boot: Loaded app from partition at offset 0x10000
I (334) boot: Set actual ota_seq=1 in otadata[0]
I (335) boot: Disabling RNG early entropy source...
I (345) cpu_start: Pro cpu up.
I (345) cpu_start: Starting app cpu, entry point is 0x40375368
0x40375368: call_start_cpu1 at /root/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (359) cpu_start: Pro cpu start user code
I (360) cpu_start: cpu freq: 160000000
I (360) cpu_start: Application information:
I (362) cpu_start: Project name:     ota_example
I (368) cpu_start: App version:      ve/1.5.2-2-gb733b05-dirty
I (374) cpu_start: Compile time:     Aug 31 2022 11:08:43
I (380) cpu_start: ELF file SHA256:  462d3feb792b6632...
I (386) cpu_start: ESP-IDF:          v4.4.2-155-g23d5a582cb-dirty
I (393) heap_init: Initializing. RAM available for dynamic allocation:
I (400) heap_init: At 3FC9FBC0 len 00049B50 (294 KiB): D/IRAM
I (407) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (413) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (419) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (426) spi_flash: detected chip: gd
I (430) spi_flash: flash io: dio
I (435) sleep: Configure to isolate all GPIO pins in sleep state
I (441) sleep: Enable automatic switching of GPIO sleep configuration
I (448) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (489) HTTPS_OTA_EXAMPLE: [1.0] Initialize peripherals management
I (489) HTTPS_OTA_EXAMPLE: [1.1] Start and wait for Wi-Fi network
I (489) pp: pp rom version: e7ae62f
I (489) net80211: net80211 rom version: e7ae62f
I (509) wifi:wifi driver task: 3fceea8c, prio:23, stack:6656, core=0
I (509) system_api: Base MAC address is not set
I (509) system_api: read default base MAC address from EFUSE
I (519) wifi:wifi firmware version: cfcc3b2
[... 49 lines cut ...]
I (4999) OTA_DEFAULT: data upgrade uri file://sdcard/flash_tone.bin
I (5009) FATFS_STREAM: File size: 15310 byte, file position: 0
E (5009) TONE_PARTITION: Not flash tone partition
I (6049) OTA_DEFAULT: write_offset 0, r_size 2048
I (6069) OTA_DEFAULT: write_offset 2048, r_size 2048
I (6069) OTA_DEFAULT: write_offset 4096, r_size 2048
I (6079) OTA_DEFAULT: write_offset 6144, r_size 2048
I (6099) OTA_DEFAULT: write_offset 8192, r_size 2048
I (6099) OTA_DEFAULT: write_offset 10240, r_size 2048
I (6109) OTA_DEFAULT: write_offset 12288, r_size 2048
I (6119) OTA_DEFAULT: write_offset 14336, r_size 974
W (6129) FATFS_STREAM: No more data, ret:0
I (6129) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (6129) OTA_DEFAULT: partition flash_tone upgrade successes
W (6139) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
I (6149) HTTPS_OTA_EXAMPLE: List id: 0, OTA success
E (6149) esp_https_ota: No option for server verification is enabled in esp_http_client config.
E (6159) OTA_DEFAULT: ESP HTTPS OTA Begin failed
E (6169) OTA_SERVICE: OTA prepared fail
E (6169) HTTPS_OTA_EXAMPLE: List id: 1, OTA failed
W (6179) OTA_SERVICE: OTA_END!
I (6179) HTTPS_OTA_EXAMPLE: [2.3] Finish OTA service
I (6189) HTTPS_OTA_EXAMPLE: [2.4] Clear OTA service

```

shootao, 2022-09-01:
1.    ` (https://github.com/espressif/esp-adf/commit/35d24221461d0bd7a8ea6c3c1912e5dd348f17f7) is okay `  指的是打上了上面patch之后的吗？
2.  `latest commit do not work` 是否提供一下 log， 还是和之前的一样 ？

hnhkj, 2022-09-02:
sorry. It's okay when I apply your patch. Thanks.

closed 2022-09-02 by hnhkj