espressif/esp-adf #879: 使用voip模板，搭建ble_gatt_server报错。

jiafeng-1, 2022-09-06:
```bash
#项目地址
https://gitee.com/jiafeng-1/esp-adf-voip-ble-error.git
```
报错内容
```bash
/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/blufi/blufi_security.c:45:10: fatal error: esp32s2beta/rom/crc.h: No such file or directory
 #include "esp32s2beta/rom/crc.h"
```
adf版本2.4 
idf版本4.4.2

板卡使用的[ESP32-S3-Korvo-2](https://docs.espressif.com/projects/esp-adf/zh_CN/latest/get-started/user-guide-esp32-s3-korvo-2.html)

报错原因我找到了，但是不知道怎么解决。
```bash
#注释掉voip根目录下的cmakelist.txt中的include($ENV{ADF_PATH}/CMakeLists.txt)就可以了，
#但是我要在工程中集成voip+wifi+ble。所以不知道怎么解决了。请各位大佬看下gitee工程，帮忙解决下。
include($ENV{ADF_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```
详细报错信息
```bash
FAILED:
espdf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/lib/blufi/blufi_security.c.obj

/home/jiafeng/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-DADF_VER=\"672433d-dirty\"
-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"
/home/jiafeng/jiafeng/ESP32/esp-s3/guardian1/build/config
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/include
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/adc_button
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/gpio_isr
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/button
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/blufi
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/IS31FL3216
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/aw2013
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/tca9554
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/driver/i2c_bus
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/sdcard
-I/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/touch
-I/home/jiafeng/esp/esp-idf/components/newlib/platform_include
-I/home/jiafeng/esp/esp-idf/components/freertos/include
-I/home/jiafeng/esp/esp-idf/components/freertos/include/esp_additions/freertos
-I/home/jiafeng/esp/esp-idf/components/freertos/port/xtensa/include
-I/home/jiafeng/esp/esp-idf/components/freertos/include/esp_additions
-I/home/jiafeng/esp/esp-idf/components/esp_hw_support/include
-I/home/jiafeng/esp/esp-idf/components/esp_hw_support/include/soc
-I/home/jiafeng/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3
-I/home/jiafeng/esp/esp-idf/components/esp_hw_support/port/esp32s3/.
-I/home/jiafeng/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include
-I/home/jiafeng/esp/esp-idf/components/heap/include
-I/home/jiafeng/esp/esp-idf/components/log/include
-I/home/jiafeng/esp/esp-idf/components/lwip/include/apps
-I/home/jiafeng/esp/esp-idf/components/lwip/include/apps/sntp
-I/home/jiafeng/esp/esp-idf/components/lwip/lwip/src/include
-I/home/jiafeng/esp/esp-idf/components/lwip/port/esp32/include
-I/home/jiafeng/esp/esp-idf/components/lwip/port/esp32/include/arch
-I/home/jiafeng/esp/esp-idf/components/soc/include
-I/home/jiafeng/esp/esp-idf/components/soc/esp32s3/.
-I/home/jiafeng/esp/esp-idf/components/soc/esp32s3/include
-I/home/jiafeng/esp/esp-idf/components/hal/esp32s3/include
-I/home/jiafeng/esp/esp-idf/components/hal/include
-I/home/jiafeng/esp/esp-idf/components/hal/platform_port/include
-I/home/jiafeng/esp/esp-idf/components/esp_rom/include
-I/home/jiafeng/esp/esp-idf/components/esp_rom/include/esp32s3
-I/home/jiafeng/esp/esp-idf/components/esp_rom/esp32s3
-I/home/jiafeng/esp/esp-idf/components/esp_common/include
-I/home/jiafeng/esp/esp-idf/components/esp_system/include
-I/home/jiafeng/esp/esp-idf/components/esp_system/port/soc
[... 108 lines cut ...]
-D_POSIX_READER_WRITER_LOCKS
-MD
-MT
esp-idf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/lib/blufi/blufi_security.c.obj
-MF
esp-idf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/lib/blufi/blufi_security.c.obj.d
-o
esp-idf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/lib/blufi/blufi_security.c.obj
-c
/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/blufi/blufi_security.c
/home/jiafeng/esp/esp-adf/components/esp_peripherals/lib/blufi/blufi_security.c:45:10:
fatal
error:
esp32s2beta/rom/crc.h:
No
such
file
or
directory

#include
"esp32s2beta/rom/crc.h"

```
@projectgus @justintconroy @pedrominatel @kedars @mahavirj 麻烦各位大佬啦！voip+wifi已经跑通了，在加ble组件时候报错的。

jason-mao, 2022-09-07:
这个问题在 d44e7418ba8e52a8822723f6d593d32a9977f8ec 上已修复，请更新代码再验证。

jiafeng-1, 2022-09-07:
您好，目前我的adf版本已经是2.4版本了。请问还需要怎么更新呢？
```bash
:~/esp/esp-adf$ git describe --tags --dirty
v2.4-dirty
~/esp/esp-idf$ git describe --tags --dirty
v4.4.2-dirty
```
@jason-mao

jason-mao, 2022-09-07:
您可以使用以下命令更新到最新的master代码
```c
git checkout master
git pull
git submodule update --init --recursive
```

jiafeng-1, 2022-09-07:
@jason-mao 
问题已经解决，需要更新下adf固件。
非常感谢，你们的支持。谢谢

closed 2022-09-07 by jiafeng-1