espressif/esp-adf #883: get music msg from mobile phone over BT's a2dp

Kingwulin, 2022-09-08:
Hi,I am using the BT a2dp to recevie music from mobile phone，how can get the music title，article or some other msg from phone

and-elf, 2022-09-10:
Take a look at this code:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_av.c
static void bt_av_hdl_avrc_ct_evt()
 case ESP_AVRC_CT_METADATA_RSP_EVT:

Basically, I use all these files in the bt_sink folder in my projects Not sure if there is a better way, though

Kingwulin, 2022-09-10:
thank your reply,It is a good way to achive it,but I am using the example in adf sdk,Is it possible to achive this function in this [demo](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream)

and-elf, 2022-09-11:
I'm pretty sure a demo is supposed to show a specific part, not all possible features of an API. I showed you how it is possible to extend the demo to enable your request. 
The easy answer is: No

Kingwulin, 2022-09-22:
> bt_avrc_ct_cb

it is what need，Thanks very Much
But I have not idea to close and reopen the it
Is there any api to achive it

and-elf, 2022-09-22:
What do you mean by close and reopen?
The API to change callback function is esp_avrc_register_callback()

Kingwulin, 2022-09-22:
Hi,I am using the example from [here](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_sink)
But it still can got the music lyric, some detail at the pic below
[image]

and-elf, 2022-09-22:
Yeah. It depends on the a2dp source. Your device may not send that event. What avrc capabilities does it say? Also, the sink actually ask for the information through esp_avrc_ct_send_metadata_cmd. Basically you listen to "new track" and request new track into
Take a look at https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/main/bt_app_av.c again

Kingwulin, 2022-09-22:
> Yeah. It depends on the a2dp source. Your device may not send that event. What avrc capabilities does it say during device connect sequence? Also, the sink actually ask for the information through esp_avrc_ct_send_metadata_cmd. Basically you listen to "new track" and request new track into Take a look at https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex/main/bt_app_av.c again

the demo you say before,I found it work well,below the pic
using the same source device,but the adf demo can't work well
[image]

Kingwulin, 2022-09-22:
you mean the speaker device can't receive lyric massage is because the source device have not send  lyric msg?
And how to make the source device to send lyric massage
But I am using the same source deveice
I think there may some diff setting between the idf demo and adf demo,But I can't found it

and-elf, 2022-09-22:
IDF implements the request part, whereas adf doesn't
case ESP_AVRC_CT_CHANGE_NOTIFY_EVT:
and
case ESP_AVRC_CT_GET_RN_CAPABILITIES_RSP_EVT

Kingwulin, 2022-09-23:
do you mean those:  
  btc_avrc_ct_cb_to_app(ESP_AVRC_CT_CHANGE_NOTIFY_EVT, &param); [link](https://github.com/espressif/esp-idf/blob/5c33570524118873f7bd32490c7a0442fede4bf8/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c#L608)

btc_avrc_ct_cb_to_app(ESP_AVRC_CT_GET_RN_CAPABILITIES_RSP_EVT, &param);  [link](https://github.com/espressif/esp-idf/blob/5c33570524118873f7bd32490c7a0442fede4bf8/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c#L621)

Kingwulin, 2022-09-24:
> IDF implements the request part, whereas adf doesn't case ESP_AVRC_CT_CHANGE_NOTIFY_EVT: and case ESP_AVRC_CT_GET_RN_CAPABILITIES_RSP_EVT

I using the method the same as idf demo do,But the code running into dump error
Do you know what function should to achive in adf demo

Kingwulin, 2022-09-26:
Hi, I modify the bluethooth_service.c file as below，but it still can‘t get the music lyric ，would pealse give me  a  hand?
 

#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "audio_mem.h"
#include "sdkconfig.h"

#include "audio_idf_version.h"

#if (defined CONFIG_CLASSIC_BT_ENABLED)
#include "esp_a2dp_api.h"
#include "esp_avrc_api.h"
#include "esp_bt.h"
#include "esp_bt_device.h"
#include "esp_bt_main.h"
#include "esp_gap_bt_api.h"

#include "bluetooth_service.h"
#include "bt_keycontrol.h"

static const char *TAG = "BLUETOOTH_SERVICE";

#define BT_SOURCE_DEFAULT_REMOTE_NAME "ESP-ADF-SPEAKER"

#define VALIDATE_BT(periph, ret)                                         \
    if (!(periph && esp_periph_get_id(periph) == PERIPH_ID_BLUETOOTH)) { \
        ESP_LOGE(TAG, "Invalid Bluetooth periph, at line %d", __LINE__); \
        return ret;                                                      \
    }

/* AVRCP used transaction labels */
#define APP_RC_CT_TL_GET_CAPS            (0)
#define APP_RC_CT_TL_GET_META_DATA       (1)
#define APP_RC_CT_TL_RN_TRACK_CHANGE     (2)
#define APP_RC_CT_TL_RN_PLAYBACK_CHANGE  (3)
#define APP_RC_CT_TL_RN_PLAY_POS_CHANGE  (4)
/* A2DP source global state */
typedef enum {
    BT_SOURCE_STATE_IDLE,
    BT_SOURCE_STATE_DISCOVERING,
    BT_SOURCE_STATE_DISCOVERED,
    BT_SOURCE_STATE_UNCONNECTED,
    BT_SOURCE_STATE_CONNECTING,
    BT_SOURCE_STATE_CONNECTED,
    BT_SOURCE_STATE_DISCONNECTING,
} esp_a2d_source_state_t;

typedef uint8_t esp_peer_bdname_t[ESP_BT_GAP_MAX_BDNAME_LEN + 1];

typedef struct bluetooth_service {
    audio_element_handle_t stream;
    esp_periph_handle_t periph;
    audio_stream_type_t stream_type;
    esp_bd_addr_t remote_bda;
    esp_peer_bdname_t peer_bdname;
    esp_a2d_connection_state_t connection_state;
    esp_a2d_source_state_t source_a2d_state;
    esp_a2d_audio_state_t audio_state;
    uint64_t pos;
    uint8_t tl;
    bool avrc_connected;
    int a2dp_sample_rate;
} bluetooth_service_t;

[... 718 lines cut ...]
        g_bt_service->source_a2d_state = BT_SOURCE_STATE_IDLE;
        return esp_bt_gap_cancel_discovery();
    }
    return ESP_OK;
}

esp_err_t periph_bluetooth_connect(esp_periph_handle_t periph, bluetooth_addr_t remote_bda)
{
    AUDIO_NULL_CHECK(TAG, g_bt_service, return ESP_FAIL);
    if (g_bt_service->stream_type == AUDIO_STREAM_READER) {
        g_bt_service->source_a2d_state = BT_SOURCE_STATE_DISCOVERED;
        memcpy(&g_bt_service->remote_bda, remote_bda, BLUETOOTH_ADDR_LEN);
        ESP_LOGI(TAG, "Cancel device discovery and connect remote device.");
        return esp_bt_gap_cancel_discovery();
    }
    return ESP_OK;
}

int periph_bluetooth_get_a2dp_sample_rate()
{
    AUDIO_NULL_CHECK(TAG, g_bt_service, return -1);
    return g_bt_service->a2dp_sample_rate;
}

#endif

Kingwulin, 2022-09-28:
> IDF implements the request part, whereas adf doesn't case ESP_AVRC_CT_CHANGE_NOTIFY_EVT: and case ESP_AVRC_CT_GET_RN_CAPABILITIES_RSP_EVT

Hi, I modify the bluethooth_service.c file as below，but it still can‘t get the music lyric ，would pealse give me a hand on how to deal with it? Thanks

```
#include <stdlib.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "audio_mem.h"
#include "sdkconfig.h"

#include "audio_idf_version.h"

#if (defined CONFIG_CLASSIC_BT_ENABLED)
#include "esp_a2dp_api.h"
#include "esp_avrc_api.h"
#include "esp_bt.h"
#include "esp_bt_device.h"
#include "esp_bt_main.h"
#include "esp_gap_bt_api.h"

#include "bluetooth_service.h"
#include "bt_keycontrol.h"

static const char *TAG = "BLUETOOTH_SERVICE";

#define BT_SOURCE_DEFAULT_REMOTE_NAME "ESP-ADF-SPEAKER"

#define VALIDATE_BT(periph, ret)                                         \
    if (!(periph && esp_periph_get_id(periph) == PERIPH_ID_BLUETOOTH)) { \
        ESP_LOGE(TAG, "Invalid Bluetooth periph, at line %d", __LINE__); \
        return ret;                                                      \
    }

/* AVRCP used transaction labels */
#define APP_RC_CT_TL_GET_CAPS            (0)
#define APP_RC_CT_TL_GET_META_DATA       (1)
#define APP_RC_CT_TL_RN_TRACK_CHANGE     (2)
#define APP_RC_CT_TL_RN_PLAYBACK_CHANGE  (3)
#define APP_RC_CT_TL_RN_PLAY_POS_CHANGE  (4)
/* A2DP source global state */
typedef enum {
    BT_SOURCE_STATE_IDLE,
    BT_SOURCE_STATE_DISCOVERING,
    BT_SOURCE_STATE_DISCOVERED,
    BT_SOURCE_STATE_UNCONNECTED,
    BT_SOURCE_STATE_CONNECTING,
    BT_SOURCE_STATE_CONNECTED,
    BT_SOURCE_STATE_DISCONNECTING,
} esp_a2d_source_state_t;

typedef uint8_t esp_peer_bdname_t[ESP_BT_GAP_MAX_BDNAME_LEN + 1];

typedef struct bluetooth_service {
    audio_element_handle_t stream;
    esp_periph_handle_t periph;
    audio_stream_type_t stream_type;
    esp_bd_addr_t remote_bda;
    esp_peer_bdname_t peer_bdname;
    esp_a2d_connection_state_t connection_state;
    esp_a2d_source_state_t source_a2d_state;
    esp_a2d_audio_state_t audio_state;
    uint64_t pos;
    uint8_t tl;
    bool avrc_connected;
    int a2dp_sample_rate;
} bluetooth_service_t;
[... 724 lines cut ...]
        return esp_bt_gap_cancel_discovery();
    }
    return ESP_OK;
}

esp_err_t periph_bluetooth_connect(esp_periph_handle_t periph, bluetooth_addr_t remote_bda)
{
    AUDIO_NULL_CHECK(TAG, g_bt_service, return ESP_FAIL);
    if (g_bt_service->stream_type == AUDIO_STREAM_READER) {
        g_bt_service->source_a2d_state = BT_SOURCE_STATE_DISCOVERED;
        memcpy(&g_bt_service->remote_bda, remote_bda, BLUETOOTH_ADDR_LEN);
        ESP_LOGI(TAG, "Cancel device discovery and connect remote device.");
        return esp_bt_gap_cancel_discovery();
    }
    return ESP_OK;
}

int periph_bluetooth_get_a2dp_sample_rate()
{
    AUDIO_NULL_CHECK(TAG, g_bt_service, return -1);
    return g_bt_service->a2dp_sample_rate;
}

#endif
```

Kingwulin, 2022-09-28:
[image]
[image]

Kingwulin, 2022-09-29:
Hi