espressif/esp-adf #887: Run VoIP Call and MQTT together

wenxiao11, 2022-09-15:
## Environment

- Audio development kit: ESP32-LyraT
- Audio kit version: v4.3
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4.1
- ADF version: v2.4
- Build system: idf.py
- Operating system: Windows

## Problem Description
I try to enable both MQTT and VoIP.
- Run voip function without MQTT connected -> No error
- Connect MQTT first, then make voip call -> `ERROR! rb_in slow!!!`
- Make VoIP call first, connect MQTT during the call -> No error currently, will get `AUDIO_THREAD: Error creating task algo_fetch` for the next call.
- The system status during the call and MQTT connection:
```
I (54040) AUDIO_SYS: | Task              | Run Time    | Per | Prio | HWM       | State   | CoreId   | Stack
I (54040) AUDIO_SYS: | algo              | 317495      |15%  | 5    | 2868      | Ready   | 1        | Intr
I (54050) AUDIO_SYS: | at_process_task   | 3799        | 0%  | 4    | 1360      | Running | 7fffffff | Intr
I (54060) AUDIO_SYS: | uTask             | 27          | 0%  | 1    | 408       | Ready   | 7fffffff | Intr
I (54070) AUDIO_SYS: | IDLE              | 884684      |44%  | 0    | 416       | Ready   | 0        | Intr
I (54080) AUDIO_SYS: | IDLE              | 259875      |13%  | 0    | 420       | Ready   | 1        | Intr
I (54090) AUDIO_SYS: | _rtp_rx_task      | 20060       | 1%  | 16   | 1208      | Blocked | 0        | Extr
I (54100) AUDIO_SYS: | tiT               | 38495       | 1%  | 18   | 1044      | Blocked | 7fffffff | Intr
I (54110) AUDIO_SYS: | mqtt_task         | 1757        | 0%  | 5    | 652       | Blocked | 0        | Intr
I (54120) AUDIO_SYS: | sip_task          | 4827        | 0%  | 20   | 1780      | Blocked | 0        | Extr
I (54130) AUDIO_SYS: | Tmr Svc           | 0           | 0%  | 1    | 1564      | Blocked | 0        | Intr
I (54140) AUDIO_SYS: | ipc0              | 0           | 0%  | 4    | 972       | Blocked | 0        | Intr
I (54150) AUDIO_SYS: | voip_serv         | 0           | 0%  | 10   | 596       | Blocked | 1        | Extr
I (54170) AUDIO_SYS: | ipc1              | 0           | 0%  | 24   | 964       | Blocked | 1        | Intr
I (54170) AUDIO_SYS: | filter            | 28998       | 1%  | 5    | 2064      | Blocked | 1        | Extr
I (54180) AUDIO_SYS: | sip_enc           | 16999       | 0%  | 5    | 1180      | Blocked | 1        | Extr
I (54190) AUDIO_SYS: | esp_timer         | 2136        | 0%  | 22   | 3344      | Suspended | 0        | Intr
I (54210) AUDIO_SYS: | _rtp_tx_task      | 20429       | 1%  | 16   | 2308      | Blocked | 0        | Extr
I (54220) AUDIO_SYS: | filter            | 28642       | 1%  | 5    | 2076      | Blocked | 1        | Extr
I (54230) AUDIO_SYS: | sip_dec           | 9779        | 0%  | 5    | 960       | Blocked | 1        | Extr
I (54240) AUDIO_SYS: | afe_mase          | 217721      |10%  | 5    | 1708      | Blocked | 1        | Intr
I (54250) AUDIO_SYS: | algo_fetch        | 19237       | 0%  | 5    | 1368      | Blocked | 0        | Intr
I (54260) AUDIO_SYS: | wifi              | 68029       | 3%  | 23   | 4040      | Blocked | 1        | Intr
I (54270) AUDIO_SYS: | i2s               | 17660       | 0%  | 23   | 1488      | Blocked | 1        | Intr
I (54280) AUDIO_SYS: | i2s               | 13704       | 0%  | 23   | 1612      | Blocked | 1        | Intr
I (54290) AUDIO_SYS: | filter_r          | 14246       | 0%  | 5    | 2072      | Blocked | 1        | Extr
I (54300) AUDIO_SYS: | at_recv_task      | 0           | 0%  | 6    | 2980      | Blocked | 7fffffff | Intr
I (54310) AUDIO_SYS: | at_socket_task    | 0           | 0%  | 4    | 5176      | Blocked | 7fffffff | Intr
I (54320) AUDIO_SYS: | sys_evt           | 0           | 0%  | 20   | 1220      | Blocked | 0        | Intr

I (54330) VOIP_CMD: Func:at_queryCmdVoipStatus, Line:776, MEM Total:3860819 Bytes, Inter:44551 Bytes, Dram:3347 Bytes
```
- The [sdkconfig](https://github.com/espressif/esp-adf/files/9577027/sdkconfig.txt) file for my project.
I know that `ERROR! rb_in slow!!!` is related to a CPU problem, but in some situation CPU can handle them together. Do other errors also CPU problems or memory problems? I want to know it is a software/config problem or hardware limitation. Thanks!

wenxiao11, 2022-09-21:
I found that the problem is the lack of internal memory. Is there any way to increase the amount of use for SPI RAM to decrease the pressure of IRAM?

jason-mao, 2022-09-23:
@wenxiao11 You can test with follow configurations. I saw your CPU core 1 is busy than core 0, move some tasks to core 0 to make the two cores have same load. 

```c
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096    # Can be adjusted a little lower， too low may cause crash
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=18432
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
```

hnhkj, 2022-10-17:
esp32-s3-korvo-2 has same problem. I modify the CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL. problem have not improved.

closed 2022-11-24 by wenxiao11