espressif/esp-adf #889: Non RFC 7616 complain Authorization header causes troubles with VOIP SIP servers

lukakama, 2022-09-18:
Hi,

I'm trying to use an online VOIP service to register an ESP32 device (based on voip-example), but the authentication protocol goes in a sort of loop, with the server returning a "401 Unauthorized" response to every REGISTER method invocation of the client, even if containing an Authorization header build with correct credential and information (verified):

```
I (4129) SIP: 

REGISTER sip:<omississ>@sip.linphone.org:5060 SIP/2.0
Via: SIP/2.0/TCP 192.168.1.236:52820;branch=z9hG4bK-479632323;rport
From: <sip:<omississ>@sip.linphone.org:5060>;tag=1365422751
To: <sip:<omississ>@sip.linphone.org:5060>
Contact: <sip:<omississ>@192.168.1.236:52820;transport=TCP>
Max-Forwards: 70
Call-ID: CDCF755A440EB9D51B94439E6DDBF79D12F4BD12A426
CSeq: 1 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml

I (4199) SIP: [1970-01-01/00:00:01]=======================>>
I (4239) SIP: [1970-01-01/00:00:01]<<=====READ 0555 bytes==
I (4239) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/TCP 192.168.1.236:52820;branch=z9hG4bK-479632323;rport=52820;received=93.42.72.166
From: <sip:<omississ>@sip.linphone.org:5060>;tag=1365422751
To: <sip:<omississ>@sip.linphone.org:5060>;tag=j1Z6m71Fr8t5p
Call-ID: CDCF755A440EB9D51B94439E6DDBF79D12F4BD12A426
CSeq: 1 REGISTER
Server: Flexisip/2.2.0-alpha-91-g78f7e3c (sofia-sip-nta/2.0)
WWW-Authenticate: Digest realm="sip.linphone.org", nonce="/mXR5gAAAAB6kaNOAAA4GVbAaZQAAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
Content-Length: 0

I (4289) SIP: [1970-01-01/00:00:01]<<======================
I (4289) SIP: Required authentication
I (4299) SIP: [1970-01-01/00:00:01]=======WRITE 0929 bytes>>
I (4299) SIP: 

REGISTER sip:<omississ>@sip.linphone.org:5060 SIP/2.0
Via: SIP/2.0/TCP 192.168.1.236:52820;branch=z9hG4bK--1752212970;rport
From: <sip:<omississ>@sip.linphone.org:5060>;tag=-1860452768
To: <sip:<omississ>@sip.linphone.org:5060>
Contact: <sip:<omississ>@192.168.1.236:52820;transport=TCP>
Max-Forwards: 70
Call-ID: CDCF755A440EB9D51B94439E6DDBF79D12F4BD12A426
CSeq: 2 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="<omississ>", realm="sip.linphone.org", nonce="/mXR5gAAAAB6kaNOAAA4GVbAaZQAAAAA", uri="sip:sip.linphone.org:5060", response="6a42b3d16acf7dd586b08defdec03aa28187eed", qop="auth"

I (4389) SIP: [1970-01-01/00:00:01]=======================>>
I (4429) SIP: [1970-01-01/00:00:01]<<=====READ 0558 bytes==
I (4429) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/TCP 192.168.1.236:52820;branch=z9hG4bK--1752212970;rport=52820;received=93.42.72.166
From: <sip:<omississ>@sip.linphone.org:5060>;tag=-1860452768
To: <sip:<omississ>@sip.linphone.org:5060>;tag=S0gm14Q836p8Q
Call-ID: CDCF755A440EB9D51B94439E6DDBF79D12F4BD12A426
CSeq: 2 REGISTER
Server: Flexisip/2.2.0-alpha-91-g78f7e3c (sofia-sip-nta/2.0)
WWW-Authenticate: Digest realm="sip.linphone.org", nonce="/mXR5gAAAADntO1iAABpr8rucFMAAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
Content-Length: 0
```

Investigating the issue, it seems that the Authorization header sent by the ESP32 contains, at least, an invalid value for the "qop" parameter: according to [RFC 7616](https://tools.ietf.org/html/rfc7616) this parameter in Authorization header MUST not be quoted.

So, the following Authorization header:
```
Authorization: Digest username="<omississ>", realm="sip.linphone.org", nonce="/mXR5gAAAAB6kaNOAAA4GVbAaZQAAAAA", uri="sip:sip.linphone.org:5060", response="6a42b3d16acf7dd586b08defdec03aa28187eed", qop="auth"
```
should be sent as follows:
```
Authorization: Digest username="<omississ>", realm="sip.linphone.org", nonce="/mXR5gAAAAB6kaNOAAA4GVbAaZQAAAAA", uri="sip:sip.linphone.org:5060", response="6a42b3d16acf7dd586b08defdec03aa28187eed", qop=auth
```

This is the [assertion in the RFC](https://tools.ietf.org/html/rfc7616#section-3.4):

```
   For historical reasons, a sender MUST only generate the quoted string
   syntax for the following parameters: username, realm, nonce, uri,
   response, cnonce, and opaque.

   For historical reasons, a sender MUST NOT generate the quoted string
   syntax for the following parameters: algorithm, qop, and nc.

```
This is also reported in [RFC requests examples](https://tools.ietf.org/html/rfc7616#section-3.9).

jason-mao, 2022-09-23:
@lukakama Thanks for your report, let us check then repy.

ahhfzhang, 2022-09-23:
Hi @lukakama 

Thanks for the detailed report and we will submit a fix soon.

br.

lukakama, 2022-09-23:
Hi @jason-mao and @ahhfzhang, thanks for the reply. 

Please let me update you with results of some experiments I made by modifying SIP request messages sent by the ESP32 device (using an http reverse proxy): the non RFC compliant "qop" parameter seems to not be the only cause of the authentication loop with some SIP services, but there is another issue caused by the absence of the "opaque" parameter in Authorization headers, not sent by the ESP32 library, when it is present in the WWW-Authenticate header provided by the SIP server.

In the same RFC 7619, it is stated that the "opaque" parameter SHOULD be included in the Authorization header sent by the client with value untouched if the server provide it in the WWW-Authenticate header. As RFC SHOULD keyword means, it is a recommended behavior for better specification compatibility and its omission can lead to not working integrations. In this case, as the RFC itself report, the "opaque" parameter may be used by servers to keep some sort of authentication session state, so its absence in the Authorization header can break Digest authentication if the server need its value and cannot work without it.

I can confirm that adding the missing "opaque" parameter in Authorization headers (modifying them in the http reverse proxy), the authentication toward the SIP server completes successfully without further issues, so it would be nice if it can be managed inside the ESP-ADF library.

Let me know if it is needed to open another issue for the "opaque" parameter or if it can be managed in the current one.

ahhfzhang, 2022-09-23:
Hi @lukakama

No need to open another issue, I will fix it too.

closed 2023-01-19 by espressif-bot