espressif/esp-adf #892: 语音识别命令词的内存方式改变

Zhengqinze05, 2022-09-26:
你好，我现在使用的是esp32_s3_korvo2_v3的板子，在进行语音识别的时候，使用了4系列的模型mn4_cn，这个的命令词初始化是在外部内存，但是我在使用5系列模型mn5q8_en，发现了他的命令词初始化在内部内存。本身语音识别占用的内存就比较大，这些命令词的初始化是否可以放在外部。另外，我在使用5系列模型mn5q8_en时，动态修改命令词，我的程序是有暂停语音识别的，cfg.sr_iface->base.enable(cfg.sr_handle,false);当我进行2-3次暂停开启和修改命令词时，系统发生了重启，但是使用mn4_cn模型时不会有这些问题。
[image]
[image]
是否是命令词初始化在内部内存，我进行动态修改时触发了什么问题，还有可不可以将命令词初始化放在外部内存，这可以使得我内部内存可以减少占用。

jason-mao, 2022-10-25:
@Zhengqinze05 不同的模型对CPU的要求不一样，我们会根据情况调整内存位置。一般放置在内部的RAM是为了提高大数据的处理速度。建议您把其他内存放置在外部。您说的第二问题复现的条件是“进行2-3次暂停开启和修改命令词 cfg.sr_iface->base.enable(cfg.sr_handle,false)系统就发生了重启”？

Zhengqinze05, 2022-10-25:
> @Zhengqinze05命令的条件是“进行2-3次暂停开启和修改词cfg.sr_iface->base.enable(cfg.sr_handle,false)”开始重启了”？

感谢你的回复，已经知道原因了，还是内存问题。语音识别占用的内存太多了，开启了其他东西，再进行暂停开启很容易导致内存问题而重启

closed 2022-10-25 by Zhengqinze05