espressif/esp-adf #898: Resample examples do not work in latest stable release

AristarchosOfSamos, 2022-10-08:
Tested the esp-adf resample examples,
audio_processing/pipeline_resample
audio_processing/pipeline_spiffs_amr_resample
in order to evaluate hte resample feature.
Unfortunately the examples although built ok, upon running they did not work, not performing any resample and showing errors in monitor.
Environment used is a Ubuntu 22.04 LTS and esp-adf used is the latest (at the time) stable tag release v2.4.1 from here
https://github.com/espressif/esp-adf/releases/tag/v2.4.1
Devkits used are
ESP32-LyraT
ESP32-LyraT-Mini
ESP32-S2-Kaluga-1
Problems of not working and showing errors are in all three devkits tested

Following are the monitor logs of each of the devkits for the example "pipeline_spiffs_amr_resample".
This is not modified in any way, is is used verbatim as it exists in espressif esp-adf repository.
Also, environment is not altered in any way, just followed the install instructions in order to have as clean of an environment as possible.

The log from ESP32-LyraT
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6676
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.2-155-g23d5a582cb 2nd stage bootloader
I (29) boot: compile time 23:16:39
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 3, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1dfd4h (122836) map
I (165) esp_image: segment 1: paddr=0002dffc vaddr=3ffb0000 size=0201ch (  8220) load
I (168) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=47f9ch (294812) map
I (277) esp_image: segment 3: paddr=00077fc4 vaddr=3ffb201c size=003cch (   972) load
I (278) esp_image: segment 4: paddr=00078398 vaddr=40080000 size=0e414h ( 58388) load
I (307) esp_image: segment 5: paddr=000867b4 vaddr=50000000 size=00010h (    16) load
I (314) boot: Loaded app from partition at offset 0x10000
I (314) boot: Disabling RNG early entropy source...
I (327) cpu_start: Pro cpu up.
I (327) cpu_start: Starting app cpu, entry point is 0x400812f0
0x400812f0: call_start_cpu1 at /home/aristarchos/Espressif/framework/esp-adf-v2.4.1/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (341) cpu_start: Pro cpu start user code
I (342) cpu_start: cpu freq: 160000000
I (342) cpu_start: Application information:
I (346) cpu_start: Project name:     spiffs_amr_resample_app
I (352) cpu_start: App version:      1
I (357) cpu_start: Compile time:     Oct  8 2022 23:16:32
I (363) cpu_start: ELF file SHA256:  c781b30699d99fe2...
I (369) cpu_start: ESP-IDF:          v4.4.2-155-g23d5a582cb
I (375) heap_init: Initializing. RAM available for dynamic allocation:
[... 290 lines cut ...]
W (139419) AUDIO_ELEMENT: IN-[file_writer] AEL_IO_ABORT
W (139429) AUDIO_ELEMENT: IN-[amrnb_encoder] AEL_IO_ABORT
E (139439) AUDIO_ELEMENT: [file_writer] Element already stopped
I (139439) SPIFFS_AMR_RESAMPLE_EXAMPLE: Link audio elements to make playback pipeline ready
W (139449) AUDIO_PIPELINE: There are no listener registered
I (139459) SPIFFS_AMR_RESAMPLE_EXAMPLE: Setup file path to read the amr audio to play
W (139469) SPIFFS_STREAM: No more data, ret:0
E (139469) FIR_RESAMPLE: failed to allocate p_out buffer.size 196608 line 2010
E (139479) RSP_FILTER: Failed to create the resample handler
E (139479) AUDIO_ELEMENT: [filter_upsample] AEL_STATUS_ERROR_OPEN,-1
W (139489) AUDIO_ELEMENT: [filter_upsample] audio_element_on_cmd_error,7
W (139499) AUDIO_ELEMENT: IN-[i2s_writer] AEL_IO_ABORT

```

## Environment

- Audio development kits:      ESP32-LyraT v4.3, ESP32-LyraT-Mini v1.2, ESP32-S2-Kaluga-1 v1.3
-  IDF version :   v4.4.2-155-g23d5a582cb
-  ADF version :   v2.4.1
- Build system:         idf.py
- Compiler version :    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating system:     Linux Ubuntu 22.04 LTS
- Using an IDE?: No
- Power supply:         external 5V

jason-mao, 2022-10-25:
@AristarchosOfSamos The `E (113389) FIR_RESAMPLE: failed to allocate p_out buffer.size 196608 line 2010` means is lacking memory. If you have PSRAM, please enable it at menuconfig first.

AristarchosOfSamos, 2022-10-27:
Hi @jason-mao,
thank you for answering.
I do understand that the PSRAM needs to be enabled in each of the three Espresif audio dev kits.
The problem though is that after this other errors come, eg in the case of LyraT mini the examples do not work since the board itself as a hardware is a mono and not stereo, or even after all debugging that a plain user could do in the so called 'examples' the audio is choppy or with many problems.
And that is the issue, a user should not struggle to debug the 'examples', they suppose to be ready and working showcasing the functionality they advertise in their associated readme files.
I do not know if they were working sometime in the past, but as they are now in the v2.4.1 they simply need someone's attention from Espressif to run them and test or ammend them in order to serve their purpose as 'examples'.
They should work out of the box, or got very specific instructions on how to make them work per each given audio devkit.

Regards,

pad52, 2022-11-02:
I have the same problem.
Since I am using a ESP32 WROOM without PSRAM for my projects I am obliged to use an old esp-adf version to use the resampling library.

houhaiyan, 2022-11-18:
@pad52 @AristarchosOfSamos Thank you for your feedback.   I want to know some detail about resample. 
Can  you tell me resample configure in your project?
resample configure:
1. src_rate
2. dest_rate
3. src_bits
4. dest_bits
5. complexity

pad52, 2022-11-22:
Hello,
I am resampling from 44100Hz, 2 channels, 16 bit.
to 48000Hz, same channels and bit.
Complexity is the one defined in DEFAULT_RESAMPLE_FILTER_CONFIG, so I think it's 2.

yujong-lee, 2023-05-03:
@houhaiyan  I am also using facing a problem. (I am running example)

```cpp
// pipeline_a2dp_sink_stream/main/play_bt_music_example.c
#if (CONFIG_ESP_LYRATD_MSC_V2_1_BOARD || CONFIG_ESP_LYRATD_MSC_V2_2_BOARD)
    rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg.src_rate = 44100;
    rsp_cfg.src_ch = 2;
    rsp_cfg.dest_rate = 48000;
    rsp_cfg.dest_ch = 2;
    rsp_cfg.task_prio = 19;
    audio_element_handle_t filter = rsp_filter_init(&rsp_cfg);
    audio_pipeline_register(pipeline, filter, "filter");
    i2s_stream_set_clk(i2s_stream_writer, 48000, 16, 2);
    const char *link_tag[3] = {"bt", "filter", "i2s"};
    audio_pipeline_link(pipeline, &link_tag[0], 3);
#else
    const char *link_tag[2] = {"bt", "i2s"};
    audio_pipeline_link(pipeline, &link_tag[0], 2);
#endif
```

I am sure that the audio is changed to mono from stereo, which do not match the config.

yujong-lee, 2023-05-03:
@jason-mao 
@houhaiyan  

I am also using facing a problem. (I am running an example)

```cpp
// pipeline_a2dp_sink_stream/main/play_bt_music_example.c
#if (CONFIG_ESP_LYRATD_MSC_V2_1_BOARD || CONFIG_ESP_LYRATD_MSC_V2_2_BOARD)
    rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg.src_rate = 44100;
    rsp_cfg.src_ch = 2;
    rsp_cfg.dest_rate = 48000;
    rsp_cfg.dest_ch = 2;
    rsp_cfg.task_prio = 19;
    audio_element_handle_t filter = rsp_filter_init(&rsp_cfg);
    audio_pipeline_register(pipeline, filter, "filter");
    i2s_stream_set_clk(i2s_stream_writer, 48000, 16, 2);
    const char *link_tag[3] = {"bt", "filter", "i2s"};
    audio_pipeline_link(pipeline, &link_tag[0], 3);
#else
    const char *link_tag[2] = {"bt", "i2s"};
    audio_pipeline_link(pipeline, &link_tag[0], 2);
#endif
```

I am sure that the audio is changed to mono from the stereo, which does not match the config.
(I tested with mono audio and stereo audio)

pad52, 2023-06-13:
any news on this?

houhaiyan, 2023-06-14:
@pad52 The newest version has fixed some bug, please use new library.

closed 2023-09-05 by jason-mao