espressif/esp-adf #899: ESP32-S3-Korvo-2 V3.0 的voip 中的AEC不起作用

jiafeng-1, 2022-10-09:
我使用的版本库
* idf(HEAD, tag: v4.4.2)
* adf(HEAD -> master）
```c
ESP32-S3-Korvo-2 V3.0开发板的AEC好像并没有起作用。

其中宏定义默认为true
#define RECORD_HARDWARE_AEC         (true)

导致下面的代码是不执行的。

#if !RECORD_HARDWARE_AEC
    //Please reference the way of ALGORITHM_STREAM_INPUT_TYPE2 in "algorithm_stream.h"
    ringbuf_ref = rb_create(2 * ESP_RING_BUFFER_SIZE, 1);
    audio_element_set_multi_input_ringbuf(element_algo, ringbuf_ref, 0);

    /* When the playback signal far ahead of the recording signal,
        the playback signal needs to be delayed */
    algo_stream_set_delay(element_algo, ringbuf_ref, DEFAULT_REF_DELAY_MS);
#endif

我将RECORD_HARDWARE_AEC         改为false
发现无法听清ESP32-S3-Korvo-2 V3.0开发板说什么。
修改延时也是没有任何效果。
```
请问这个问题如何解决呢？假如选择一款外置的音频采集板卡来做AEC处理，请问有什么注意的地方吗？

jason-mao, 2022-10-25:
@jiafeng-1 AEC 的是将MIC 采集的数据与采集的自身播放的数据进行抵消，从而获得较为清晰的mic数据。所以设备自身播放的数据采集的质量是AEC的效果基本保证。如果您要求高质量的AEC，建议使用ESP32-S3-Korvo-2 V3.0推荐的硬件设计方式。对于您反馈的不起作用的问题，我们验证过后回复您。

jason-mao, 2023-09-05:
关闭这个不活跃问题，如有需要，请重新打开或者创建新 issue.

closed 2023-09-05 by jason-mao