espressif/esp-adf #902: lyrat-mini关于例子中wwe和google_translate_device的合并后不断重启

HongYan12345, 2022-10-12:
我使用的是lyrat-mini-v1.2, 我希望通过唤醒词来唤醒板子之后使用google的sr和tts功能，但是当我合并代码flash后出现循环重启并出现一下错误
`
I (32591) wwe_example: Recorder has been created
Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
failed to calloc fft cos/sin windows table
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40100699  PS      : 0x00060930  A0      : 0x800fc2f6  A1      : 0x3ffc0b60
0x40100699: WebRtcNs_set_policy_core at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/noise_suppression/webrtc_ns.c:1713
 (inlined by) WebRtcNs_set_policy at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/noise_suppression/webrtc_ns.c:2115

A2      : 0x00002000  A3      : 0x00000002  A4      : 0x00000000  A5      : 0x000007fc
A6      : 0x00000400  A7      : 0x3ffe3e8c  A8      : 0x3f8ccccd  A9      : 0x00000000
A10     : 0xffffffff  A11     : 0x00003e80  A12     : 0x00000804  A13     : 0x00002008
A14     : 0x00000000  A15     : 0x000003ff  SAR     : 0x00000017  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000205c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5

Backtrace:0x40100696:0x3ffc0b600x400fc2f3:0x3ffc0b80 0x400fb22d:0x3ffc0ba0 0x400df79d:0x3ffc0bd0 0x400d87d5:0x3ffc0c00 0x400d8b9c:0x3ffc0d20 0x400d8db4:0x3ffc0d50 0x4008fcb9:0x3ffc0e70
0x40100696: WebRtcNs_set_policy_core at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/noise_suppression/webrtc_ns.c:1713
 (inlined by) WebRtcNs_set_policy at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/noise_suppression/webrtc_ns.c:2115

0x400fc2f3: ns_pro_create at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/esp_ns.c:45

0x400fb22d: afe_create_from_config at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_front_end/esp_afe_sr_1mic.c:129

0x400df79d: recorder_sr_create at C:/Users/a3761/Desktop/esp-idf/esp-adf/components/audio_recorder/recorder_sr.c:557 (discriminator 2)

0x400d87d5: start_recorder at c:\users\a3761\desktop\esp-idf\esp-adf\examples\cloud_services\google_translate_device\build/../main/wake.c:343

0x400d8b9c: wake_start at c:\users\a3761\desktop\esp-idf\esp-adf\examples\cloud_services\google_translate_device\build/../main/wake.c:409

0x400d8db4: translate_task at c:\users\a3761\desktop\esp-idf\esp-adf\examples\cloud_services\google_translate_device\build/../main/translate_device_example.c:142 (discriminator 13)

0x4008fcb9: vPortTaskWrapper at C:/Users/a3761/Desktop/esp-idf/components/freertos/port/xtensa/port.c:131
`

这是由于什么导致的？两个项目我单独跑都是没有问题的，我是将wwe的代码放到了google_translate_device里面，因为wwe的menuconfig不全

HongYan12345, 2022-10-27:
我在menuconfig中设置了psram，启动时已经不会出现这个错误了，但是又有了新的问题。当我使用唤醒词叫醒lyrat后，它进去连接wifi时显示内存不足，我在idf.py size-components 中看我的内存是足够的，我不知道是哪里冲突了
下面是我的错误
`I (6272) LYRAT: [ 1 ] Initialize Buttons & Connect to Wi-Fi network, ssid=AndroidAP2614
E (6282) gpio: gpio_install_isr_service(449): GPIO isr service already installed
I (6282) LYRAT: debug4
I (6292) wifi:wifi driver task: 3ffc6474, prio:23, stack:6656, core=0
I (6292) system_api: Base MAC address is not set
I (6302) system_api: read default base MAC address from EFUSE
I (6322) wifi:wifi firmware version: 7679c42
I (6332) wifi:wifi certification version: v7.0
I (6332) wifi:config NVS flash: enabled
I (6332) wifi:config nano formating: disabled
I (6332) wifi:Init data frame dynamic rx buffer num: 32
I (6332) wifi:Init management frame dynamic rx buffer num: 32
I (6342) wifi:Init management short buffer num: 32
I (6352) wifi:Init static tx buffer num: 16
I (6352) wifi:Init tx cache buffer num: 32
I (6352) wifi:Init static rx buffer size: 1600
W (6362) wifi:malloc buffer fail
I (6362) wifi:Init static rx buffer num: 0
I (6362) wifi:Init dynamic rx buffer num: 32
E (6372) wifi:Expected to init 10 rx buffer, actual is 0
I (6372) wifi:Deinit lldesc rx mblock:0
I (6382) wifi:Deinit lldesc rx mblock:0
I (6382) wifi:Deinit lldesc rx mblock:0
I (6382) wifi:Deinit lldesc rx mblock:0
I (6422) wifi_init: rx ba win: 6
I (6422) wifi_init: tcpip mbox: 32
I (6422) wifi_init: udp mbox: 6
I (6422) wifi_init: tcp mbox: 6
I (6432) wifi_init: tcp tx win: 5744
I (6432) wifi_init: tcp rx win: 5744
I (6432) wifi_init: tcp mss: 1440
I (6432) wifi_init: WiFi IRAM OP enabled
I (6442) wifi_init: WiFi RX IRAM OP enabled
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4008e27c
0x4008e27c: _esp_error_check_failed at C:/Users/a3761/Desktop/esp-idf/components/esp_system/esp_err.c:42

file: "IDF/esp-adf/components/esp_peripherals/periph_wifi.c" line 416
func: _wifi_init
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x4008e27f on core 0
0x4008e27f: _esp_error_check_failed at C:/Users/a3761/Desktop/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40081ff5:0x3fff3c900x4008e289:0x3fff3cb0 0x400934c9:0x3fff3cd0 0x4008e27f:0x3fff3d40 0x400dd1d1:0x3fff3d60 0x400dcdbd:0x3fff3ed0
0x40081ff5: panic_abort at C:/Users/a3761/Desktop/esp-idf/components/esp_system/panic.c:402

0x4008e289: esp_system_abort at C:/Users/a3761/Desktop/esp-idf/components/esp_system/esp_system.c:121

0x400934c9: abort at C:/Users/a3761/Desktop/esp-idf/components/newlib/abort.c:46

0x4008e27f: _esp_error_check_failed at C:/Users/a3761/Desktop/esp-idf/components/esp_system/esp_err.c:43

0x400dd1d1: _wifi_init at C:/Users/a3761/Desktop/esp-idf/esp-adf/components/esp_peripherals/periph_wifi.c:404 (discriminator 1)

0x400dcdbd: esp_periph_task at C:/Users/a3761/Desktop/esp-idf/esp-adf/components/esp_peripherals/esp_peripherals.c:104 (discriminator 15)
`

以下是我的代码主体部分
`
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"

#include "esp_http_client.h"
#include "sdkconfig.h"
[... 509 lines cut ...]
    log_clear();

    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    periph_cfg.extern_stack = true;
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    if (set != NULL) {
        esp_periph_set_register_callback(set, periph_callback, NULL);
    }
    //audio_board_sdcard_init(set, SD_MODE_1_LINE);
    audio_board_key_init(set);
    //audio_board_init();
    setup_player();
    start_recorder();
    rec_q = xQueueCreate(3, sizeof(int));
    audio_thread_create(NULL, "read_task", voice_read_task, NULL, 4 * 1024, 5, true, 0);
}

void app_main(void)
{
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_INFO);
    wake_start();
}`

当运行到 periph_wifi_wait_for_connected(wifi_handle, portMAX_DELAY);时出现了ESP_ERR_NO_MEM错误并且重启了，是什么原因导致的？究竟是component开启的太多了我要去menuconfig里面修改还是代码里哪个函数冲突了？

jason-mao, 2022-11-18:
@HongYan12345 wifi 初始化放置在start_recorder前面试试。

HongYan12345, 2022-11-18:
@jason-mao 我现在有个想法，我使用的是google-sr来判断用户说的话，也就是不使用wwe中的multinet，那么那个player只用来播放dingdong.mp3，所以我把player去掉，把dingdong.MP3放到sd卡里然后用audio_pipeline_run来播放是不是可行？

HongYan12345, 2022-11-20:
我在start_recorder前后加了AUDIO_MEM_SHOW(TAG);
[image]
会不会是这个dram不足导致的？有没有什么办法

HongYan12345, 2022-11-22:
我打开了menuconfig里面的Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate inter
[image]
然后再运行
[image]
可以看到有两个task成功创建了，但是还差一点

HongYan12345, 2022-11-23:
我参考了https://blog.csdn.net/weixin_44821644/article/details/109207305，关掉了一些东西，已经能成功启用唤醒词和wifi了

closed 2022-11-23 by HongYan12345