espressif/esp-adf #903: ESP-IDF V4.4.2 patch idf_v4.4_freertos.patch doesn't match

Vxs14, 2022-10-13:
Hello,

trying to apply the freertos patch on windows :

idf V4.4.2
adf Release v2.4.1
ESP32 

to reproduce :

C:\>cd %IDF_PATH%

C:\Users\user\esp\esp-idf>git apply %ADF_PATH%/idf_patches/idf_v4.4_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:351
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:981
error: components/freertos/tasks.c: patch does not apply

shootao, 2022-10-14:
Hi @Vxs14 
I use the same `esp-adf` and `esp-idf` as you, so I can patch normally. Please keep the `esp-idf` in a clean environment before patching. If there is any problem, run git status under the `esp-idf` path to check. 
My test `esp-idf` commit is '1b16ef6cfc2479a08136782f9dc57effefa86f66 (HEAD, tag: v4.4.2)'

Vxs14, 2022-10-15:
Hi shootao,

thank you very much for your answer, then I did a fresh install with git , and indeed there isn't problem with the patch.
My first install of idf 4.4.2 was done by vscode's idf extension, the only difference between the two installs.
I couldn't find any further explanation.

closed 2022-10-15 by Vxs14

shootao, 2023-02-23:
Hi @Vxs14 
Has your problem been solved？

Bhavesh-Moradiya, 2024-08-06:
madarchod andar jake change karle