espressif/esp-adf #908: 如何释放OTA占用的资源

hnhkj, 2022-10-27:
我添加了新的俄
```
I (5005) AUDIO_THREAD: The Audio_forge task allocate stack on external memory
I (4945) VOICE_EXTENDER: Func:app_init, Line:310, MEM Total:8443931 Bytes, Inter:163079 Bytes, Dram:163079 Bytes

I (5005) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (5025) APP_OTA_UPGRADE: Create OTA service
I (5025) APP_OTA_UPGRADE: Start OTA service
I (5035) OTA_DEFAULT: data upgrade uri https://call4help.uk/ESSFirmwareUpdate/ve/tone/audio_tone.bin
I (7185) HTTP_STREAM: total_bytes=384482
I (7945) APP_OTA_UPGRADE: format 1 : 1
I (7955) APP_OTA_UPGRADE: imcoming magic_word F55F9876, project_name ESP_TONE_BIN
I (7955) APP_OTA_UPGRADE: current version v1.7.1, incoming version v1.7.0
W (7955) APP_OTA_UPGRADE: The incoming version is same as or lower than the running version
E (7965) OTA_SERVICE: No need to upgrade
E (7975) APP_OTA_UPGRADE: List id: 0, OTA failed, result: 589828
W (7975) esp_https_ota: Continuing with insecure option because CONFIG_OTA_ALLOW_HTTP is set.
I (10135) esp_https_ota: Starting OTA...
I (10135) esp_https_ota: Writing to partition subtype 17 at offset 0x2c0000
I (12725) OTA_DEFAULT: Running firmware version: v1.8.0, the incoming firmware version v1.7.0
W (12725) OTA_DEFAULT: Current running version is the same as or higher than a new. We will not continue the update.
E (12735) OTA_SERVICE: No need to upgrade
E (12745) APP_OTA_UPGRADE: List id: 1, OTA failed, result: 589828
W (12755) OTA_SERVICE: OTA_END!
W (12755) APP_OTA_UPGRADE: upgrade finished, Please check the result of OTA
E (12765) OTA_SERVICE: OTA_DESTROY
I (12765) VOICE_EXTENDER: Func:app_init, Line:312, MEM Total:8352211 Bytes, Inter:85427 Bytes, Dram:85427 Bytes

I (12775) VOICE_EXTENDER: app_voip_start [324]
I (12785) APP_VOIP: Create SIP Service
E (12785) DISPATCHER: exe first list: 0x0
```

jason-mao, 2022-10-31:
@hnhkj 您的调用流程是什么样的？

hnhkj, 2022-11-02:
main.c

```
void app_init(void)
{
    // Clear the debug message
    log_clear();
    media_lib_add_default_adapter();

    WIFI_CONNECTED_FLAG = xEventGroupCreate();

    // step 1. init the peripherals set and peripherls
    ESP_LOGI(TAG, "[ 1 ] Initialize peripherals management");
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    periph_cfg.extern_stack = true;
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    
    // step 2. setup the input key service
    ESP_LOGI(TAG, "[ 2 ] Initialize and start peripherals");
    audio_board_key_init(set);
    input_key_service_info_t input_info[] = INPUT_KEY_DEFAULT_INFO();
    input_key_service_cfg_t key_serv_info = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
    key_serv_info.based_cfg.extern_stack = false;
    key_serv_info.handle = set;
//    key_serv_info.based_cfg.task_stack = 4 * 1024;
    periph_service_handle_t input_key_handle = input_key_service_create(&key_serv_info);

    // step 3. Start codec chip
    ESP_LOGI(TAG, "[ 3 ] Start codec chip");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
    audio_hal_set_volume(board_handle->audio_hal, 70);

    AUDIO_NULL_CHECK(TAG, input_key_handle, return);
    input_key_service_add_key(input_key_handle, input_info, INPUT_KEY_NUM);
    periph_service_set_callback(input_key_handle, input_key_service_cb, (void *)board_handle);

    // step 4. Start codec chip
    ESP_LOGI(TAG, "[ 4 ] Initialize tone player");
    audio_player_int_tone_init();

    // step 5. initialize sdcard
    //audio_board_sdcard_init(set, SD_MODE_1_LINE);
    //periph_led_test();
    
    // step 6. setup Wi-Fi service
    setup_wifi();
    ota_flag = true;
    
    // step 7. wait for wifi connected, and start ota service
    EventBits_t bits = xEventGroupWaitBits(WIFI_CONNECTED_FLAG, WIFI_CONNECTED_BIT, true, false, portMAX_DELAY);
    if (bits & WIFI_CONNECTED_BIT) {
        app_ota_start();
    } else {
        ESP_LOGW(TAG, "WIFI  connection timeout(%dms), skipped OTA service", WIFI_WAIT_CONNECT_TIME_MS);
    }
    vEventGroupDelete(WIFI_CONNECTED_FLAG);
    WIFI_CONNECTED_FLAG = NULL;
    ota_flag = false;

    //app_mdns_main();
    //mqtt_app_start();

    // step 8. start voip
    ESP_LOGI(TAG,"app_voip_start [%d]",__LINE__);
    app_voip_start();
    //char *uri="udp://100:100@192.168.0.104:6560";
    //app_voip_restart(uri);
    //rest_app_main();
}
[... 202 lines cut ...]
    for (int i = 0; i < sizeof(upgrade_list) / sizeof(upgrade_list[0]); i++) {
        if (upgrade_list[i].node.type == ESP_PARTITION_TYPE_DATA) {
            ota_data_get_default_proc(&upgrade_list[i]);
            if (strcmp(upgrade_list[i].node.label, "flash_tone") == 0 && strstr(upgrade_list[i].node.uri, "http")) {
                upgrade_list[i].need_upgrade = ota_http_tone_image_need_upgrade;
            } else {
                upgrade_list[i].need_upgrade = ota_sdcard_image_need_upgrade;
            }
        } else if (upgrade_list[i].node.type == ESP_PARTITION_TYPE_APP) {
            ota_app_get_default_proc(&upgrade_list[i]);
        } else {
            ESP_LOGE(TAG, "unknown type");
        }
    }
    ota_service_set_upgrade_param(ota_service, upgrade_list, sizeof(upgrade_list) / sizeof(upgrade_list[0]));
    ESP_LOGI(TAG, "Start OTA service");
    periph_service_start(ota_service);
    EventBits_t bits = xEventGroupWaitBits(OTA_FLAG, OTA_FINISH, true, false, portMAX_DELAY);
    if (bits & OTA_FINISH) {
        ESP_LOGW(TAG, "upgrade finished, Please check the result of OTA");
    }
    vEventGroupDelete(OTA_FLAG);
    periph_service_destroy(ota_service);
}
```

closed 2022-11-22 by espressif-bot