espressif/esp-adf #910: Can the codec MCLK pin be attached to any gpio on esp32s3?

casitelwithdean, 2022-11-02:
Hi,recently i am designing a pcb which uses es8311 codec,I'm not sure if the code pin can be attached to any gpio on esp32s3?
Thanks

NARRob, 2022-11-03:
+1 on this question.  seem components/audio_stream/i2s_stream.c file hardcodes GPIO_NUM_0 for MCLK, that's probably for the ADF supported kits, but what about a custom hardware?   I am unable to use another GPIO for MCLK unless other changes required.  Anyone able to use another pin for MCLK.  I am also using ESP32-S3.

NARRob, 2022-11-03:
@casitelwithdean 
I can confirm MCLK can be changed for the codec on ESP32-S3, even when the i2s_stream.c hardcoded it.
in function get_i2s_pins(...) I included the following my case (own codec driver and board_pins_config.c 

```
        i2s_config->bck_io_num = GPIO_NUM_13;
        i2s_config->ws_io_num = GPIO_NUM_11;
        i2s_config->data_out_num = GPIO_NUM_12;
        i2s_config->data_in_num = GPIO_NUM_10;
        i2s_config->mck_io_num = GPIO_NUM_45;

```

jason-mao, 2022-11-18:
@casitelwithdean  MCLK can map to any pins with ESP32s3.

closed 2022-11-18 by jason-mao