espressif/esp-adf #911: A2DP Sink task is breaking support with ESP32-WROOM

pad52, 2022-11-02:
## Environment

- Audio development kit:    none
- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.3.4
- ADF version: from commit d86e186c9fbc0580d91fc1388cf68142c8b25d2f 
- Build system: idf.py

## Problem Description

The commit d86e186c9fbc0580d91fc1388cf68142c8b25d2f - "a2dp_stream: add task to handle a2dp sink data"
is breaking all my projects on ESP32 WROOM. 

I found many errors "A2DP_STREAM: No ineffecitive memory to allocate" and sounds very badly.

The project is a "pipeline_a2dp_sink_stream" example with resampling from 44.1khz to 48khz
the pipeline is as follows:
bt_stream_reader -> rsp_filter -> i2s_stream_writer

Since I am using a ESP32 WROOM I cannot enable PSRAM,
Using esp-adf v2.4 and substituting "a2dp_stream.c" & "a2dp_stream.h" with the files in the commit just before d86e186 everything works.

jason-mao, 2022-11-18:
@pad52 Let us check and then reply.

JosephTang, 2023-09-14:
Hi @pad52, if you don't want to modify the esp-adf library, maybe you can check the `gest_now` flag when `AUDIO_REC_WAKEUP_START` triggered in `rec_engine_cb`, invoke `audio_recorder_trigger_stop` to stop the wake up process if the flag is not set.

pad52, 2023-09-14:
I didn't understand the solution you are proposing, can you explain it to me?

However, updating esp-adf (tag v2.6) and esp-idf (I tested from release/4.4 to tag v5.1) AND  **disabling logging completely**,
It seems to me that I no longer receive  "A2DP_STREAM: No ineffecitive memory to allocate".
If I enable logging, I start to receive the error and receiving bad sound. 
For me we can also close the issue.

JosephTang, 2023-09-14:
@pad52  sorry, wrong answer, not for this issue.

pad52, 2024-05-16:
Since the issue is still there and loosing my mind I need to reopen

pad52, 2024-05-17:
Fixed enabling 
Flash SPI mode as QIO
Flash SPI speed as 80MHz
Refer to the guide: https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/design-considerations.html

closed 2024-05-17 by pad52