espressif/esp-adf #912: example pipeline_embed_flash_tone - unable to reproduce two files

pad52, 2022-11-03:
## Environment

- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.4 branch
- ADF version: v2.4 branch 

## Problem Description

Using the example pipeline_embed_flash_tone i am unable to reproduce two files.
It play only one file and then fails.
I am using the same procedure of "play_sdcard_mp3_control_example"

### Code to Reproduce This Issue

Just modify the pipeline_embed_flash_tone in the i2s stream writer event like this:

```cpp
            // Advance to the next song when previous finishes
            if (msg.source == (void *) i2s_stream_writer
                && msg.cmd == AEL_MSG_CMD_REPORT_STATUS) {
                audio_element_state_t el_state = audio_element_get_state(i2s_stream_writer);
                if (el_state == AEL_STATE_FINISHED) {
                    ESP_LOGI(TAG, "[ * ] Finished, advancing to the next song");

                    audio_element_set_uri(embed_flash_stream_reader, embed_tone_url[ALARM_MP3]);
                    audio_pipeline_reset_ringbuffer(pipeline);
                    audio_pipeline_reset_elements(pipeline);
                    audio_pipeline_change_state(pipeline, AEL_STATE_INIT);
                    audio_pipeline_run(pipeline);
                }
                continue;
```

## Debug Logs

```
D (33802) EMBED_FLASH_STREAM: Wanted read flash flash index is 1
D (33802) AUDIO_ELEMENT: [flash] el opened
D (33802) AUDIO_ELEMENT: REPORT_STATUS,[flash]evt out cmd = 8,status:12
D (33822) AUDIO_PIPELINE: resume,linked:1, state:1,[mp3-0x3ffd79c8]
I (33822) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (33832) MP3_DECODER: MP3 opened
D (33832) AUDIO_PIPELINE: resume,linked:0, state:5,[filter-0x3ffd8950]
I (33852) AUDIO_PIPELINE: Pipeline started
W (33892) EMBED_FLASH_STREAM: No more data,ret:-627 ,info.byte_pos:11910
I (33902) AUDIO_ELEMENT: IN-[flash] AEL_IO_DONE,0
I (33912) AUDIO_ELEMENT: IN-[mp3] AEL_IO_DONE,-2
E (33922) MP3_DECODER: failed to read audio data (line 117)
```

jason-mao, 2022-11-18:
@pad52 Let us reproduce your issue and reply.

shootao, 2022-11-21:
Hi @pad52 Thank you for your feedback
The fixes for the problem are as follows, and we will combine them with the latest version
```c
diff --git a/components/audio_stream/embed_flash_stream.c b/components/audio_stream/embed_flash_stream.c
index 467ff2cc..346fbd18 100644
--- a/components/audio_stream/embed_flash_stream.c
+++ b/components/audio_stream/embed_flash_stream.c
@@ -117,6 +117,7 @@ static esp_err_t _embed_close(audio_element_handle_t self)
     if (stream->is_open) {
        stream->is_open = false;
     }
+    audio_element_set_byte_pos(self, 0);
     return ESP_OK;
 }

```

closed 2022-11-25 by espressif-bot