espressif/esp-adf #913: WWE例程无法运行，报错和重启

Ouyangtengfang, 2022-11-07:
您好！

按照指导文档（https://github.com/espressif/esp-adf/blob/ba18b938297c4a0faaded7508a9fedb29d0277c8/examples/speech_recognition/wwe/README_CN.md）编译WWE例程，开发板是（ESP32-S3-Korvo-2 v3.0），但是运行后，说“HI，Lexin”就会报错和重启，日志如下，请指导，谢谢！

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.2

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.2\tools

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Espressif\frameworks\esp-idf-v4.4.2>cd c:\Espressif\esp\wwe

c:\Espressif\esp\wwe>idf.py -p COM4 monitor
Executing action: monitor
Running idf_monitor in directory c:\espressif\esp\wwe
Executing "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4.2\tools/idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\espressif\esp\wwe\build\example_wwe.elf -m 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Espressif\frameworks\esp-idf-v4.4.2\tools\idf.py' '-p' 'COM4'"...
[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx[0m
[0;33m--- Using \\.\COM4 instead...[0m
[0;33m--- idf_monitor on \\.\COM4 115200 ---[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17b0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x30e8
entry 0x403c9980
I (25) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (25) boot: compile time 17:07:52
I (25) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  3 ota_0            OTA app          00 10 00010000 00290000
I (93) boot:  4 model            Unknown data     01 82 002a0000 0040e000
I (100) boot:  5 flash_tone       Unknown data     01 27 006ae000 00032000
I (108) boot: End of partition table
[... 192 lines cut ...]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17b0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x30e8
entry 0x403c9980
I (25) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (25) boot: compile time 17:07:52
I (25) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 ota_0            OTA app          00 10 00010000 00290000
I (93) boot:  4 model            Unknown data     01 82 002a0000 0040e000
I (101) boot:  5 flash_tone       Unknown data     01 27 006ae000 00032000
I (108) boot: End of partition table
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=3b550h (243024) map
I (158) esp_image: segment 1: paddr=0004b578 vaddr=3fca21b0 size=04aa0h ( 19104) load

jason-mao, 2022-11-18:
@Ouyangtengfang 上面问题的ADF版本是多少？另外你帮用最新的master 试一下

jason-mao, 2023-01-09:
我们在0cbeef7f6d34b0db1ecd813876781be2ad279e22分支上check 该问题，没有复现。该问题先行关闭，如还有问题请重新打开或者新建Issue。

closed 2023-01-09 by jason-mao