espressif/esp-adf #916: Link的时候，提示：undefined reference to `ch_sort_16bit_4ch'

Ouyangtengfang, 2022-11-09:
您好！

我在编译dueros例程的时候，提示undefined reference to `ch_sort_16bit_4ch'， 但是我在头文件ch_sort.h的确看到函数ch_sort_16bit_4ch的定义，但是不知道为什么link的时候会报错，请问是不是哪里配置不对，请指导，谢谢！

项目很紧，请看到信息尽快回复，谢谢！

c:\Espressif\esp\voip7>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\espressif\esp\voip7\build
Executing "ninja all"...
[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\esp\voip7\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Espressif/esp/voip7/build/bootloader/bootloader.bin"
Bootloader binary size 0x5750 bytes. 0x28b0 bytes (32%) free.
[2/4] Linking CXX executable voip_app.elf
FAILED: voip_app.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\voip_app.elf.rsp -o voip_app.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/audio_recorder/libaudio_recorder.a(recorder_sr.c.obj):(.literal.feed_task+0x18): undefined reference to `ch_sort_16bit_4ch'
c:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/audio_recorder/libaudio_recorder.a(recorder_sr.c.obj): in function `feed_task':
C:/Espressif/esp/esp-adf/components/audio_recorder/recorder_sr.c:232: undefined reference to `ch_sort_16bit_4ch'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

JosephTang, 2022-11-10:
@Ouyangtengfang 你好，基于ADF中的dueros和voip例程，都未能复现你所说的问题，请问是有修改代码或者配置吗？

Ouyangtengfang, 2022-11-12:
Hi Joseph & all

您好！ 首先，非常感谢您的回信！我是佛山市台风网络科技的欧阳腾方，我们之前用了你们ESP32开发过产品，感觉你们的产品挺好，所以新产品的研发，我们想采用贵司另外一款ESP32S3模组开发。目前，为了做技术可行性分析，所以购买了你们的开发板和下载你们的demo来测试，但是遇到了一些技术问题，所以想你们指导，谢谢！

是的，我在之前提的两个关于dueros和voip的问题，是我修改过代码，为了排除是我修改代码的影响，我重新下载ADF中dueros代码，然后重新编译运行，代码能够正常运行，但是，代码会不定时出现报错，然后开发板自动重启，我把我的操作步骤和具体的软件和硬件信息描述如下，麻烦你帮忙分析一下，是什么问题，万分感激，谢谢！

一、软件和硬件信息
1、IDF版本：IDF4.4.2
2、ADF版本：直接在Github下载安装
3、开发板：ESP32-S3-Korvo-2  在你们的淘宝官网购买（329元）

二、操作步骤：
1、设置ADF的环境变量
cd c:\Espressif\esp\dueros
set ADF_PATH=c:\Espressif\esp\esp-adf
echo %ADF_PATH%

2、配置项目参数

idf.py set-target esp32s3

idf.py menuconfig
说明：在这里，我只是简单配置了WiFi的账号和密码，另外把唤醒词改成了“小爱同学”，其他配置没有更改。

3、编译工程
idf.py build

4、烧录到设备并且监视
idf.py -p COM4 flash monitor

三、错误日志
说明：运行一段时间后，会不定时出现如下错误，然后开发板重启，日志信息如下：

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00050031  A0      : 0x00000000  A1      : 0x3fca3410
A2      : 0x00000002  A3      : 0x3fca972c  A4      : 0x00000000  A5      : 0x3fca3380
A6      : 0x00000000  A7      : 0x3fcc7d68  A8      : 0x8208af6e  A9      : 0x4038bd2a
0x4038bd2a: _frxt_int_enter at C:/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:119

A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x400408bc
A14     : 0x00060023  A15     : 0x00000001  SAR     : 0x00000005  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0xfffffffd:0x3fca3410

ELF file SHA256: 0d5919cdf1d8969d

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379ac8
0x40379ac8: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17b0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x30e8
entry 0x403c9980
I (29) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (29) boot: compile time 09:04:36
I (29) boot: chip revision: 0
I (31) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
[... 79 lines cut ...]
I (1037) wifi:Init data frame dynamic rx buffer num: 32
I (1047) wifi:Init management frame dynamic rx buffer num: 32
I (1047) wifi:Init management short buffer num: 32
I (1057) wifi:Init static tx buffer num: 16
I (1057) wifi:Init tx cache buffer num: 32
I (1067) wifi:Init static tx FG buffer num: 2
I (1067) wifi:Init static rx buffer size: 1600
I (1067) wifi:Init static rx buffer num: 16
I (1077) wifi:Init dynamic rx buffer num: 32
I (1077) wifi_init: rx ba win: 16
I (1087) wifi_init: tcpip mbox: 32
I (1087) wifi_init: udp mbox: 6
I (1087) wifi_init: tcp mbox: 6
I (1097) wifi_init: tcp tx win: 5744
I (1097) wifi_init: tcp rx win: 5744
I (1107) wifi_init: tcp mss: 1440
I (1107) wifi_init: WiFi/LWIP prefer SPIRAM
I (1107) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (1147) wifi:mode : sta (68:b6:b3:22:8f:10)
I (1147) wifi:enable tsf
I (108,tid:3fcc2584) lightduer_session.c(  44):random = 75933

欧阳腾方
13825571213
佛山市台风网络科技有限公司

JosephTang, 2022-11-14:
@Ouyangtengfang  请确认下dueros 的代码和adf是否配套，建议直接编译esp-adf/example/dueros 来测试。（本地测试同样未能复现你所描述的问题）。

如果是修改了代码的dueros，请确认下代码中是否有调用函数指针，但指针值为NULL的情况。

Ouyangtengfang, 2022-11-14:
Hi Joseph & All

非常感谢您及时的回复！

我们没有修改过代码，是直接 编译esp-adf/example/dueros 代码来运行的，只是因为我们的没有SDcard，所以屏蔽了一行代码 //audio_board_sdcard_init(set, SD_MODE_1_LINE);，其他的代码没有做任何的修改。

为了能够让您重现问题，我把我们的代码和详细的日志文件一起发给您，请您帮忙分析以下是什么问题，谢谢！

另外，因为重启是不定时发生，我们开着一个小时左右，就出现了8次左右的重启，会不会跟WiFi有关系？ 还是跟开发板本身有问题？

期待您的回复，谢谢!

欧阳腾方
13825571213
台风网络科技

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.2

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.2\tools

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Espressif\frameworks\esp-idf-v4.4.2>cd c:\Espressif\esp

c:\Espressif\esp>dir
 驱动器 C 中的卷是 Windows
 卷的序列号是 C0A0-A8AB

 c:\Espressif\esp 的目录

2022/11/12  09:00    <DIR>          .
2022/11/12  09:00    <DIR>          ..
2022/11/08  16:01    <DIR>          dueros
2022/11/09  15:42    <DIR>          dueros2
2022/11/10  13:38    <DIR>          dueros7
2022/11/11  19:58    <DIR>          dueros8
2022/11/12  08:58    <DIR>          dueros_original
2022/11/03  16:28    <DIR>          esp-adf
2022/11/05  16:43    <DIR>          esp-box
2022/11/05  15:52    <DIR>          esp-box-backup
2022/11/09  15:04    <DIR>          esp_dispatcher_dueros
2022/11/03  15:35    <DIR>          hello_world
2022/11/04  09:30    <DIR>          play_mp3_control
2022/11/08  09:36    <DIR>          vad
2022/11/07  13:58    <DIR>          voip
2022/11/09  17:06    <DIR>          voip2
2022/11/08  14:13    <DIR>          voip3
[... 678 lines cut ...]
I (13754) AUDIO_WRAPPER: Recorder has been created

Done

c:\Espressif\esp\dueros_original>idf.py -p COM4 flash monitor
Executing action: flash
Running ninja in directory c:\espressif\esp\dueros_original\build
Executing "ninja flash"...
[1/12] Running move model...
C:/Espressif/esp/dueros_original/sdkconfig
C:/Espressif/esp/esp-adf/components/esp-sr/model
['wn9_xiaoaitongxue']
[]
Recommended model partition size:  1184KB
[2/12] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\esp\dueros_original\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Espressif/esp/dueros_original/build/bootloader/bootloader.bin"
[8/10] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated C:/Espressif/esp/dueros_original/build/esp_dueros.bin
[9/10] cmd.exe /C "cd /D C:\Espressif\esp\dueros_original\...bin C:/Espressif/esp/dueros_original/build/esp_dueros.bin"
esp_dueros.bin binary size 0x1bf460 bytes. Smallest app partition is 0x310000 bytes. 0x150ba0 bytes (43%) free.
[9/10] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf

JosephTang, 2022-11-14:
@Ouyangtengfang  请提供下开机到重启之后的完整log。

Ouyangtengfang, 2022-11-14:
Hi Joseph,

您好！ 我今天早上的邮件里面附件有完整的从开机到重启的日志，我再发一次给你，附件有代码和日志，谢谢！

另外，问题要运行一段时间才会出现，我运行1个小时左右，出现了8次左右，不是一运行就马上出现，谢谢！

还有，可以告诉你的电话吗？ 我想跟你电话沟通一下我们的情况，谢谢！

欧阳腾方
13825571213
台风网络科技

[9/10] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v...sp-idf-v4.4.2/components/esptool_py/run_serial_tool.cmake"

JosephTang, 2022-11-14:
@Ouyangtengfang github 页面上并不能显示log的全部，也没有附件的

jason-mao, 2022-11-18:
@Ouyangtengfang 这个问题是解决了吗？

jason-mao, 2023-09-05:
关闭这个不活跃问题，如有需要，请重新打开或者创建新 issue.

closed 2023-09-05 by jason-mao