espressif/esp-adf #919: 关于google_translate项目合并wwe后无法使用tts

HongYan12345, 2022-11-27:
我之前将google_translate项目和wwe合并，遇到了dram不足的问题，根据https://blog.csdn.net/weixin_44821644/article/details/109207305 在menuconfig中调整后成功初始化了两个项目，成功连接wifi后使用唤醒词hi lexin也成功唤醒，但是当我运行google-sr或者google-tts时，出现了下列问题
`I (5718) LYRAT: [ 5 ] Listen for all pipeline events
I (8218) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
I (8218) GOOGLE_TTS: TTS text = hola, soy demo
I (8218) AUDIO_ELEMENT: [tts_http-0x3f83079c] Element task created
I (8228) AUDIO_ELEMENT: [tts_mp3-0x3f830980] Element task created
I (8228) AUDIO_ELEMENT: [tts_i2s-0x3f830178] Element task created
I (8238) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:3982339 Bytes, Inter:47643 Bytes, Dram:18907 Bytes

I (8248) AUDIO_ELEMENT: [tts_http] AEL_MSG_CMD_RESUME,state:1
I (8258) GOOGLE_TTS: [ + ] HTTP client HTTP_STREAM_PRE_REQUEST, lenght=0
I (8278) AUDIO_ELEMENT: [tts_mp3] AEL_MSG_CMD_RESUME,state:1
I (8278) AUDIO_ELEMENT: [tts_i2s] AEL_MSG_CMD_RESUME,state:1
I (8288) AUDIO_PIPELINE: Pipeline started
I (8288) I2S_STREAM: AUDIO_STREAM_WRITER
I (8278) MP3_DECODER: MP3 opened
E (8738) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (8748) esp-tls: create_ssl_handle failed
E (8748) esp-tls: Failed to open new connection
E (8748) TRANSPORT_BASE: Failed to open a new connection
E (8768) HTTP_CLIENT: Connection failed, sock < 0
E (8768) HTTP_STREAM: Failed to open http stream
E (8768) AUDIO_ELEMENT: [tts_http] AEL_STATUS_ERROR_OPEN,28674
W (8778) AUDIO_ELEMENT: [tts_http] audio_element_on_cmd_error,7
W (8788) AUDIO_ELEMENT: IN-[tts_mp3] AEL_IO_ABORT
E (8788) MP3_DECODER: failed to read audio data (line 119)
W (8798) AUDIO_ELEMENT: [tts_mp3] AEL_IO_ABORT, -3
I (8798) MP3_DECODER: Closed
W (8828) AUDIO_ELEMENT: IN-[tts_i2s] AEL_IO_ABORT
E (10588) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10588) task_wdt:  - IDLE (CPU 1)
E (10588) task_wdt: Tasks currently running:
E (10588) task_wdt: CPU 0: afe_mase
E (10588) task_wdt: CPU 1: feed_task
E (10588) task_wdt: Print CPU 0 (current core) backtrace`

是不是因为内存还是不够？我该怎么办

HongYan12345, 2022-12-01:
我使用了esp_get_free_heap_size()和AUDIO_MEM_SHOW(TAG);查看，发现内存应该是充裕的，应该是start_recorde 和tts_start 里面的pipeline冲突，我尝试在唤醒词唤醒后将pipeline关掉再启动google-tts，可是发送了以下错误
[image]
I (16188) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
W (16188) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (16188) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (16198) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (16198) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
I (16208) GOOGLE_TTS: TTS text = hola, soy demo
I (16228) AUDIO_ELEMENT: [tts_http-0x3f8302c0] Element task created
I (16228) AUDIO_ELEMENT: [tts_mp3-0x3f831528] Element task created
I (16228) AUDIO_ELEMENT: [tts_i2s-0x3f830178] Element task created
I (16238) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:3982275 Bytes, Inter:47643 Bytes, Dram:18907 Bytes

I (16248) AUDIO_ELEMENT: [tts_http] AEL_MSG_CMD_RESUME,state:1
I (16258) GOOGLE_TTS: [ + ] HTTP client HTTP_STREAM_PRE_REQUEST, lenght=0
I (16268) AUDIO_ELEMENT: [tts_mp3] AEL_MSG_CMD_RESUME,state:1
I (16268) MP3_DECODER: MP3 opened
I (16268) AUDIO_ELEMENT: [tts_i2s] AEL_MSG_CMD_RESUME,state:1
I (16278) AUDIO_PIPELINE: Pipeline started
I (16278) I2S_STREAM: AUDIO_STREAM_WRITER
W (16378) AUDIO_ELEMENT: IN-[raw] AEL_IO_ABORT
W (16378) AUDIO_ELEMENT: IN-[raw] AEL_IO_ABORT
W (16378) AUDIO_ELEMENT: IN-[raw] AEL_IO_ABORT
W (16378) AUDIO_EVT: There is no space in external queue
W (16388) AUDIO_ELEMENT: IN-[raw] AEL_IO_ABORT
同时这个错误任然存在
[image]
以下是我添加google-tts的地方,在唤醒词wakeup_start里面
` 
if (AUDIO_REC_WAKEUP_START == type) {
        ESP_LOGI(TAG2, "rec_engine_cb - REC_EVENT_WAKEUP_START");

        if (voice_reading) {
            int msg2 = REC_CANCEL;
            if (xQueueSend(rec_q, &msg2, 0) != pdPASS) {
                ESP_LOGE(TAG2, "rec cancel send failed");
            }
        }
        audio_pipeline_stop(pipeline);
        audio_pipeline_wait_for_stop(pipeline);
        //ESP_LOGE(TAG2, "free heap size: %d",  esp_get_free_heap_size());
        //ESP_LOGE(TAG2, "minimum free heap size: %d",  esp_get_minimum_free_heap_size());
        google_tts_start(tts, "hola, soy demo", GOOGLE_TTS_LANG);
        vTaskDelay(20);
        while(google_tts_check_event_finish(tts, &msg)){
            ESP_LOGI(TAG2, "%d",google_tts_check_event_finish(tts, &msg));
        }`

jason-mao, 2022-12-02:
@HongYan12345 `AEL_IO_ABORT` 事件要即使处理，它的初现意味着pipeline即可停止了。

HongYan12345, 2022-12-02:
@jason-mao 我看google-sr和google-tts的切换也只是用了audio_pipeline_stop(sr->pipeline);
    audio_pipeline_wait_for_stop(sr->pipeline);这两个函数，我以为wwe换到google-tts也是一样的，但是不行，那么怎么才能处理掉AEL_IO_ABORT事件呢？

HongYan12345, 2022-12-03:
我发现我将start_recorder中的`cfg.sr_handle = recorder_sr_create(&recorder_sr_cfg, &cfg.sr_iface);` 注释掉就能成功调用google_tts, 但是少了这个函数就无法成功使用唤醒词，有什么办法吗？

HongYan12345, 2022-12-04:
我尝试在tts_start前面加上了
```
audio_recorder_destroy (recorder);
audio_pipeline_stop(pipeline);
audio_pipeline_wait_for_stop(pipeline);
```
但还是出现了以下错误
[image]

HongYan12345, 2022-12-11:
我修改了menuconfig让tls使用外部内存，可以调用tts了，但是重复出现ERROR! rb_in slow!!!错误，该怎么更改呢？

jason-mao, 2023-09-05:
ERROR! rb_in slow!!! 参见其他issue https://github.com/espressif/esp-adf/issues/873

closed 2023-09-05 by jason-mao