espressif/esp-adf #920: ESP32-Audio-kit VOIP Crash

pierluigizagaria, 2022-11-29:
## Environment

- Audio development kit: ESP32-Audio-kit v2.297
- ESP-ADF: 2.4.1

## Problem Description
When a call is established with the voip example, the board crashes.

## Debug Logs
https://gist.github.com/pierluigizagaria/303aaf1c89cbd7386174ddf0f1d27b77

jason-mao, 2022-12-02:
@pierluigizagaria What is your IDF version and toolchain version? Did you test this example with espresif boards?

pierluigizagaria, 2022-12-02:
I was working with ESP-IDF 4.4.3 and ESP-ADF 2.4.1. I was using the Audiokit porting in this MR #764 but it seems to not work now. I've fixed the issue by reimporting the board using the newer board.h template. I'll post here the branch I'm working on if it can help someone.

pierluigizagaria, 2022-12-03:
The example now works with the board porting in this branch [aithinker-audio-kit](https://github.com/pierluigizagaria/esp-adf-sip-intercom/tree/aithinker-audio-kit).

closed 2022-12-03 by pierluigizagaria