espressif/esp-adf #922: SIP Registration failed with Ethernet

wenxiao11, 2022-11-29:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version : v4.3
- Module or chip used:  ESP32-WROVER-E
- IDF version: 4.4
- ADF version: 2.3
- Build system:         idf.py

## Problem Description

The SIP Connect cannot detect the ethernet connection.

### Expected Behavior

SIP Connect with only ethernet connection

### Actual Behavior

SIP cannot connect with only ethernet connection

### Steps to Reproduce

1. Ethernet Connect without WIFI
2. Run VoIP registration, fail
3. Connect to WIFI, VoIP registration success
4. Disconnect WIFI, make VoIP call, success

## Timeline
- Ethernet Connect: 2022-11-28 16:12:11.056 (4986)
- VOIP Registration fail: 2022-11-28 16:12:37.793 (31706)
- Connect WIFI: 2022-11-28 16:13:04.121 (58046)
- VoIP Registration success: 2022-11-28 16:13:11.390 (65306)
- Disconnect WIFI: 2022-11-28 16:13:56.549 (110476)
- VOIP call success: 2022-11-28 16:14:14.384 (128286)
- Keep Alive: 2022-11-28 16:15:22.009 (195916)

## Debug Logs

```
[2022-11-28 16:12:06.121] I (29) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
[2022-11-28 16:12:06.121] I (29) boot: compile time 13:36:21
[2022-11-28 16:12:06.121] I (29) boot: chip revision: 3
[2022-11-28 16:12:06.121] I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
[2022-11-28 16:12:06.121] I (30) qio_mode: Enabling default flash chip QIO
[2022-11-28 16:12:06.121] I (31) boot.esp32: SPI Speed      : 80MHz
[2022-11-28 16:12:06.121] I (32) boot.esp32: SPI Mode       : QIO
[2022-11-28 16:12:06.121] I (32) boot.esp32: SPI Flash Size : 8MB
[2022-11-28 16:12:06.121] I (33) boot: Enabling RNG early entropy source...
[2022-11-28 16:12:06.121] I (34) boot: Partition Table:
[2022-11-28 16:12:06.121] I (34) boot: ## Label            Usage          Type ST Offset   Length
[2022-11-28 16:12:06.121] I (35) boot:  0 phy_init         RF data          01 01 0000f000 00001000
[2022-11-28 16:12:06.121] I (36) boot:  1 otadata          OTA data         01 00 00010000 00002000
[2022-11-28 16:12:06.121] I (37) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
[2022-11-28 16:12:06.121] I (38) boot:  3 ota_0            OTA app          00 10 00020000 00220000
[2022-11-28 16:12:06.121] I (39) boot:  4 ota_1            OTA app          00 11 00240000 00220000
[2022-11-28 16:12:06.121] I (39) boot:  5 at_customize     unknown          40 00 00460000 00250000
[2022-11-28 16:12:06.121] I (40) boot:  6 flash_tone       Unknown data     01 ff 006b0000 0007d000
[2022-11-28 16:12:06.121] I (41) boot: End of partition table
[2022-11-28 16:12:06.121] I (42) boot_comm: chip revision: 3, min. application chip revision: 0
[2022-11-28 16:12:06.121] I (43) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=70068h (458856) map
[2022-11-28 16:12:06.276] I (170) esp_image: segment 1: paddr=00090090 vaddr=3ffbdb60 size=05c1ch ( 23580) load
[2022-11-28 16:12:06.276] I (178) esp_image: segment 2: paddr=00095cb4 vaddr=40080000 size=0a364h ( 41828) load
[2022-11-28 16:12:06.276] I (192) esp_image: segment 3: paddr=000a0020 vaddr=400d0020 size=13d230h (1298992) map
[2022-11-28 16:12:06.631] I (550) esp_image: segment 4: paddr=001dd258 vaddr=4008a364 size=0c1f8h ( 49656) load
[2022-11-28 16:12:06.669] I (566) esp_image: segment 5: paddr=001e9458 vaddr=400c0000 size=00068h (   104) load
[2022-11-28 16:12:06.669] I (566) esp_image: segment 6: paddr=001e94c8 vaddr=50000000 size=00010h (    16) load
[2022-11-28 16:12:06.669] I (578) boot: Loaded app from partition at offset 0x20000
[2022-11-28 16:12:06.669] I (578) boot: Disabling RNG early entropy source...
[2022-11-28 16:12:06.669] D (580) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
[... 583 lines cut ...]
[2022-11-28 16:15:22.009] Expires: 3600
[2022-11-28 16:15:22.009] User-Agent: ESP32 SIP/2.0
[2022-11-28 16:15:22.009] Content-Length: 0
[2022-11-28 16:15:22.009] Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
[2022-11-28 16:15:22.009] Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
[2022-11-28 16:15:22.009] Allow-Events: presence, kpml
[2022-11-28 16:15:22.009] Authorization: Digest username="+17786576992", realm="sip.twilio.com", nonce="CJWsY_Ft90O-rPkxhAPQN9SGGfr4zZv34N0ZrDX0zxFZOqOc", uri="sip:epicsafetytest.sip.twilio.com:5060", response="5966262eb37708808060329e9cb9bd3f", algorithm=md5, nc=00000004, cnonce="cb9481668e0d93d8", qop="auth"
[2022-11-28 16:15:22.024] 
[2022-11-28 16:15:22.024] 
[2022-11-28 16:15:22.024] I (195926) SIP: [1970-01-01/00:03:07]=======================>>
[2022-11-28 16:15:22.093] I (196006) SIP: [1970-01-01/00:03:07]<<=====READ 0406 bytes==
[2022-11-28 16:15:22.093] I (196006) SIP: 
[2022-11-28 16:15:22.093] 
[2022-11-28 16:15:22.093] SIP/2.0 200 OK
[2022-11-28 16:15:22.093] Via: SIP/2.0/UDP 0.0.0.0:16661;branch=z9hG4bK-1671688047;rport=44432;received=207.6.10.148
[2022-11-28 16:15:22.093] From: <sip:+17786576992@epicsafetytest.sip.twilio.com:5060>;tag=-128716691
[2022-11-28 16:15:22.093] To: <sip:+17786576992@epicsafetytest.sip.twilio.com:5060>;tag=24af9585fd65ee6f2a61b710bfe5956a.e9f018b3
[2022-11-28 16:15:22.093] Call-ID: 0BEA4AC79B2900F9F5EA2781E69CD440B31ACF7C453E
[2022-11-28 16:15:22.093] CSeq: 7 OPTIONS
[2022-11-28 16:15:22.093] Server: Twilio Gateway
[2022-11-28 16:15:22.093] Content-Length: 0
[2022-11-28 16:15:22.093] 
[2022-11-28 16:15:22.093] 
[2022-11-28 16:15:22.093] I (196016) SIP: [1970-01-01/00:03:07]<<======================
```

jason-mao, 2022-12-02:
@wenxiao11 Thanks for your report, let we check  then reply.

ahhfzhang, 2022-12-05:
Hi @wenxiao11 

Your eth been adapted with the esp_netif interface ?
and you can sent or receive data though eth ?

btw, you need config sip local_addr with your eth address.

thanks.

wenxiao11, 2022-12-05:
Hi @ahhfzhang 
Sending and receiving data through ethernet works correctly on other functions. For VoIP part, we remove the wifi_service and create "voip_service" similar to it but without WIFI. We also use ESP-AT commands and ethernet will be connected automatically when plugging in. 
For the second part, how can we config sip local_addr? esp_sip does not provide any related function.

Thanks

ahhfzhang, 2022-12-07:
Hi @wenxiao11 

You can change the local ip [here](https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/voip/main/voip_app.c#L260).

All sip socket send or recevie by these [API](https://github.com/espressif/esp-adf-libs/blob/master/media_lib_sal/port/media_lib_socket_default.c).

wenxiao11, 2022-12-16:
Hi @ahhfzhang 

Sorry for the reopen. After I fix the local IP problem, the SIP registration is ok with Ethernet. When I try to make the VoIP call, Invite and Hangout works correctly but there is no sound during the call. The reason is that for the SIP handler:
[image]
There is no WRITE_EVENT, only READ_EVENT. 
Here are the logs for WIFI (success) and Ethernet (fail).
[WIFI Success.log](https://github.com/espressif/esp-adf/files/10241668/WIFI.Success.log)
[Ethernet Fail.log](https://github.com/espressif/esp-adf/files/10241671/Ethernet.Fail.log)

Thanks

ahhfzhang, 2023-03-23:
Hi @wenxiao11 ,

ADF v2.5 has fix this issue，only open [use_public_addr](https://github.com/espressif/esp-adf-libs/blob/master/esp_media_protocols/include/esp_rtc.h#L106)，use public address communicate with server.

thanks.

closed 2023-06-21 by espressif-bot