espressif/esp-adf #932: Cannot Hang up VoIP Call Correctly

wenxiao11, 2022-12-24:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version : v4.3
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4
- ADF version: v2.3

## Problem Description

The VoIP call session is not ended correctly when caller is not the ESP device. That is, if I use MicroSIP to call the device, the device accept the call and then hangup, the session cannot be ended.
However, if I do it reversely, use device to call MicroSIP and MicroSIP hangup, the session is closed correctly.

### Expected Behavior

Receive 200 OK message

### Actual Behavior

Receive 481 Call/Transaction Does not Exist

### Steps to Reproduce

1. Register both MicroSIP (101) and ESP device (100)
2. 101 send call to 100
3. 100 accept the call, the session start
4. 100 hangup, the audio pipeline closed, the session is not closed

## Debug Logs

[log.txt](https://github.com/espressif/esp-adf/files/10297759/log.txt): The SIP BYE messages for both success and fail cases
[success.log](https://github.com/espressif/esp-adf/files/10297762/Success.log): Full log for the successful case. (Device call, MicroSIP hangup)
[fail.log](https://github.com/espressif/esp-adf/files/10297765/Fail.log): Full log for the fail case. (MicroSIP call, Device hangup)

ahhfzhang, 2023-01-12:
Hi @wenxiao11 

Can you try to update to release/v2.4, because the new version has some sip bugfixes.

But I also see some errors like the CSeq should not be 3, due to the fail.log is incomplete and some SIP fields are missing, so it's not easy to analyze the reasons.

Thanks.

charon22, 2023-01-17:
Hi @ahhfzhang 
I am Sungsu working with Wenxiao11. First of all, we were using the 2022 Feb version of esp-adf.
We tested with ESP32-LyraT 4.3 with the latest version 2.4 as you recommened. 
The hangup problem still occurs.
To reproduce this problem, a more detailed explanation is as follows.

Test Environment : 
-	It is currently being developed using ESP32-LyraT, and MicroSIP is installed on the development PC to test the call.
                       PC (MicroSIP)  <--> Twilio  SIP server <--> ESP32-LyraT
Problem : 
-	The VoIP call session is not ended correctly when caller is not the ESP device. 
-	That is, if I use MicroSIP to call the device, the device accept the call and then hangup, the session cannot be ended.
o	PC (MicroSIP)  -----> Twilio  SIP server --> ESP32-LyraT
                 Call         ---------------------------->
                                                                             Accept call
                                <--------------------------     Hang up
                                                                              Fail
         Call status is maintained.

-	However, if I do it reversely, use device to call MicroSIP and MicroSIP hangup, the session is closed correctly.
o	PC (MicroSIP)  <------Twilio  SIP server  <----- ESP32-LyraT
                                <--------------------------------- Call
       Accept call
       Hang up            --------------------------------->
       Success

Thanks
Sungsu

ahhfzhang, 2023-01-28:
Hi @charon22 

As twillo replied, we should use the 'INVITE' contact header, but it was cleared by ‘OPTION’, so it replaced by new one.
I think the latest ADF master had fix this issue, we use keep-alive instead of 'option' to keep the NAT hole opened.
So you can set 'send_options = false' and try again.

Thanks.

charon22, 2023-02-01:
Hi @ahhfzhang 
As you said, I set send_option to false and tested it. 
However, the same problem still occurred. 
I will send you the corresponding device logo.

You can check that send_option is set to false in the log as follows in the log.
I (4549) VOIP_EXAMPLE: [ 5 ] Create SIP Service
I (4549) VOIP_EXAMPLE: **send_options (0)**

Lyrat device log

[teraterm_COM7_20230201_092223.log](https://github.com/espressif/esp-adf/files/10559789/teraterm_COM7_20230201_092223.log)

Thanks

ahhfzhang, 2023-02-07:
Hi @charon22 

Thanks, we will fix it soon.

charon22, 2023-02-07:
Hi @ahhfzhang
We look forward to your corrections.
Thank you

closed 2023-06-21 by espressif-bot