espressif/esp-adf #934: 关于lyrat_mini无法使用google-sr

HongYan12345, 2022-12-26:
我使用lyrat-mini在调用例子中的google-traslate时发现google-sr无法调用，google-tts没有问题，但是google-sr调用后返回的是空值
[image]
api调用成功了但是并没有值在里面

jason-mao, 2023-01-09:
@HongYan12345 您看下服务端是否参数发生了变化导致请求无效？

HongYan12345, 2023-01-09:
> @HongYan12345 您看下服务端是否参数发生了变化导致请求无效？

@jason-mao 我用v4.3执行一样的代码就没有任何问题，返回空的原因是没有声音录进去

PrathamG, 2023-12-20:
Hi! Were you able to solve this? I have the same issue :(

shootao, 2023-12-21:
Hi @PrathamG 
Please refer to  [Issue 939](https://github.com/espressif/esp-adf/issues/939)

PrathamG, 2024-01-24:
Hi @shootao , I tried with the patch, but still getting the same result :(

> I (73258) GOOGLE_SR: Got HTTP Response = {
> "totalBilledTime": "3s",
>  "requestId": "1085991790643842495"
> }

PrathamG, 2024-02-02:
Solved! After going through the sdcard recording example, I realized the I2S configuration is different for this board.

For LyraT-Mini I had to change the following in google_sr.c, function google_sr_init(google_sr_config_t *config):

Replace this part:
`
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    sr->i2s_reader = i2s_stream_init(&i2s_cfg);
`

with this:
`
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_cfg.i2s_port = 1;
    i2s_cfg.i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
    sr->i2s_reader = i2s_stream_init(&i2s_cfg);
`

> google-sr gives a proper response now:
> 
>  I (31408) GOOGLE_SR: Got HTTP Response = {
>   "results": [
>     {
>       "alternatives": [
>         {
>           "transcript": "hello hello hello how are you hello",
>           "confidence": 0.93190706
>         }
>       ],
>       "resultEndTime": "2.890s",
>       "languageCode": "en-us"
>     }
>   ],
>   "totalBilledTime": "3s",
>   "requestId": "1111744560565211375"
> }

closed 2024-07-31 by jason-mao