espressif/esp-adf #936: MP3播放任务无法停止

ForgetCSX, 2023-01-06:
我使用ESP32-LyraT-Mini_V1.2开发板运行play_sdcard_mp3_control_example例子，无法停止播放任务，
。
我直接复制一份，然后屏蔽掉while直接执行后面释放资源的代码，将会导致异常复位。
我认为可以这样操作直接停止播放任务的，是我操作不当，还是不支持完全停止释放资源？麻烦帮忙看下，谢谢了。

下面是完整日志：
ELF file SHA256: 4cfb14e2b890d413

Rebooting...
I (13) boot: ESP-IDF v4.4.3-347-g9ee3c8337d-dirty 2nd stage bootloader
I (13) boot: compile time 17:58:03
I (13) boot: chip revision: 3
I (16) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (23) boot.esp32: SPI Speed      : 40MHz
I (28) boot.esp32: SPI Mode       : DIO
I (32) boot.esp32: SPI Flash Size : 2MB
I (37) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (61) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 00100000
I (76) boot: End of partition table
I (80) boot_comm: chip revision: 3, min. application chip revision: 0
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=463e8h (287720) map
I (200) esp_image: segment 1: paddr=00056410 vaddr=3ffb0000 size=02818h ( 10264) load
I (204) esp_image: segment 2: paddr=00058c30 vaddr=40080000 size=073e8h ( 29672) load
I (218) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=47548h (292168) map
I (324) esp_image: segment 4: paddr=000a7570 vaddr=400873e8 size=0e998h ( 59800) load
I (359) boot: Loaded app from partition at offset 0x10000
I (359) boot: Disabling RNG early entropy source...
I (371) psram: This chip is ESP32-D0WD
I (373) spiram: Found 64MBit SPI RAM device
I (373) spiram: SPI RAM mode: flash 40m sram 40m
I (376) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (383) cpu_start: Pro cpu up.
I (387) cpu_start: Starting app cpu, entry point is 0x400816f8
0x400816f8: call_start_cpu1 at /home/xiang/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (387) cpu_start: App cpu up.
I (1281) spiram: SPI SRAM memory test OK
I (1289) cpu_start: Pro cpu start user code
I (1289) cpu_start: cpu freq: 160000000
I (1289) cpu_start: Application information:
I (1292) cpu_start: Project name:     play_sdcard_mp3_control
I (1298) cpu_start: App version:      v20220417-12-g909eec5-dirty
I (1305) cpu_start: Compile time:     Dec 30 2022 17:57:58
I (1311) cpu_start: ELF file SHA256:  4cfb14e2b890d413...
I (1317) cpu_start: ESP-IDF:          v4.4.3-347-g9ee3c8337d-dirty
I (1324) heap_init: Initializing. RAM available for dynamic allocation:
I (1331) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1338) heap_init: At 3FFB33C0 len 0002CC40 (179 KiB): DRAM
I (1344) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1350) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1357) heap_init: At 40095D80 len 0000A280 (40 KiB): IRAM
I (1363) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1373) spi_flash: detected chip: generic
I (1376) spi_flash: flash io: dio
W (1380) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1394) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1404) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1414) SDCARD_MP3_CONTROL_EXAMPLE: [1.0] Initialize peripherals management
I (1424) SDCARD_MP3_CONTROL_EXAMPLE: [1.1] Initialize and start peripherals
I (1934) SDCARD_MP3_CONTROL_EXAMPLE: [1.2] Set up a sdcard playlist and scan sdcard music save to it
I (2144) SDCARD_MP3_CONTROL_EXAMPLE: [ 2 ] Start codec chip
W (2174) I2C_BUS: i2c_bus_create:58: I2C bus has been already created, [port:0]
I (2184) SDCARD_MP3_CONTROL_EXAMPLE: [ 3 ] Create and start input key service
I (2184) SDCARD_MP3_CONTROL_EXAMPLE: [4.0] Create audio pipeline for playback
I (2194) SDCARD_MP3_CONTROL_EXAMPLE: [4.1] Create i2s stream to write data to codec chip
I (2204) SDCARD_MP3_CONTROL_EXAMPLE: [4.2] Create mp3 decoder to decode mp3 file
I (2204) SDCARD_MP3_CONTROL_EXAMPLE: [4.3] Create resample filter
I (2214) SDCARD_MP3_CONTROL_EXAMPLE: [4.4] Create fatfs stream to read data from sdcard
I (2224) SDCARD_MP3_CONTROL_EXAMPLE: [4.5] Register all elements to audio pipeline
I (2234) SDCARD_MP3_CONTROL_EXAMPLE: [4.6] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->resample-->i2s_stream-->[codec_chip]
I (2244) SDCARD_MP3_CONTROL_EXAMPLE: [5.0] Set up  event listener
I (2254) SDCARD_MP3_CONTROL_EXAMPLE: [5.1] Listen for all pipeline events
W (2254) SDCARD_MP3_CONTROL_EXAMPLE: [ 6 ] Press the keys to control music player:
W (2264) SDCARD_MP3_CONTROL_EXAMPLE:       [Play] to start, pause and resume, [Set] next song.
W (2274) SDCARD_MP3_CONTROL_EXAMPLE:       [Vol-] or [Vol+] to adjust volume.
I (2284) SDCARD_MP3_CONTROL_EXAMPLE: [ 7 ] Stop audio_pipeline
W (2294) AUDIO_PIPELINE: Without stop, st:1
W (2294) AUDIO_PIPELINE: Without wait stop, st:1
W (2304) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2304) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2314) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2324) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2534) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2534) AUDIO_PIPELINE: There are no listener registered
W (2534) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2544) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2554) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2564) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x400821e9:0x3ffb5520 0x4008d801:0x3ffb5540 0x4009494d:0x3ffb5560 0x400913b7:0x3ffb5680 0x4008e135:0x3ffb56c0 0x400df190:0x3ffb5700 0x400df1ec:0x3ffb5750 0x400dae06:0x3ffb5780 0x400daceb:0x3ffb57b0 0x40117210:0x3ffb5980 0x400911a5:0x3ffb59a0
0x400821e9: panic_abort at /home/xiang/esp/esp-idf/components/esp_system/panic.c:408

0x4008d801: esp_system_abort at /home/xiang/esp/esp-idf/components/esp_system/esp_system.c:137

0x4009494d: __assert_func at /home/xiang/esp/esp-idf/components/newlib/assert.c:85

0x400913b7: spinlock_acquire at /home/xiang/esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:122
 (inlined by) xPortEnterCriticalTimeout at /home/xiang/esp/esp-idf/components/freertos/port/xtensa/port.c:301

0x4008e135: vPortEnterCritical at /home/xiang/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) xQueueGenericSend at /home/xiang/esp/esp-idf/components/freertos/queue.c:840

0x400df190: input_key_service_event_send at /home/xiang/esp/esp-adf/components/input_key_service/input_key_service.c:59 (discriminator 2)

0x400df1ec: input_key_service_destroy at /home/xiang/esp/esp-adf/components/input_key_service/input_key_service.c:145 (discriminator 2)

0x400dae06: periph_service_destroy at /home/xiang/esp/esp-adf/components/esp_dispatcher/periph_service.c:98

0x400daceb: app_main at /home/xiang/esp/44vba/pipeline_sdcard_mp3_control/build/../main/play_sdcard_mp3_control_example.c:276

0x40117210: main_task at /home/xiang/esp/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x400911a5: vPortTaskWrapper at /home/xiang/esp/esp-idf/components/freertos/port/xtensa/port.c:142

ForgetCSX, 2023-01-06:
我调试了一下，找到问题了，https://github.com/ForgetCSX/esp-adf/commit/b68c6b6c119654daac2f1f99b4a378caaa1005c1

jason-mao, 2023-01-09:
@ForgetCSX Thanks for your report and fix.

closed 2023-01-29 by espressif-bot