espressif/esp-adf #937: Interrupt observed playing mp3 when another task comes to run

robbietree8, 2023-01-06:
## Environment

- Audio development kit:      none
- [Required] Module or chip used:  esp32s3
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v4.4.3-210-g932b6699ac
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.4.1-12-g81bac5f
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system:     macOS
- Using an IDE?: No
- Power supply:         USB

## Problem Description

When playing mp3, there will be a task reading adc every ten seconds.Observed that at the time of reading adc, mp3 will be interrupted and resume.

### Code to Reproduce This Issue

- [adc task code](https://gist.github.com/robbietree8/a600f4d41848d1c6947972318e913fc7)
- [mp3 code](https://gist.github.com/robbietree8/e0ee2b8dc833d2a9ffc50ec003bf65f2)

## Other Items If Possible

- [x] sdkconfig file (Attach the sdkconfig file from your project folder)

[sdkconfig.txt](https://github.com/espressif/esp-adf/files/10359184/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jason-mao, 2023-01-09:
@robbietree8 Would you test file from sdard or http stream?

robbietree8, 2023-01-09:
> @robbietree8 Can you test file from sdard or http stream?

I've tested http/spiffs stream example with scheduled task on official LyraT board, and it works fine, no interrupt found.
I also tested my own application code on LyraT board, it also works fine.

http stream on my custom board will be tested when the board has good wifi condition, currently, some radio frequency problem existed.

robbietree8, 2023-01-12:
@jason-mao I have tested http stream using `play_http_mp3_example`, the issue also existed.

I have modified example to add a task.

```
void demo_task()
{
    while(1) {
        ESP_LOGI(TAG, "test...");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

void app_main(void)
{
    xTaskCreate(demo_task, "demo_task", 4096, NULL, 1, NULL);
```

robbietree8, 2023-01-17:
hi @jason-mao 

I found that if I removed `ESP_LOGI...`line, interrupt issue disappeared.

closed 2023-09-05 by jason-mao