espressif/esp-adf #941: WAV解码器在24bit数据时异常，而16bit/32bit正常

Nsiso, 2023-01-10:
管道设置：
input -> HTTP_STREAM -> WAV DECODER -> I2S_STREAM -> output
音频采样率以及I2S目标采样率设置为48KHZ

当设置音频文件以及目标样本数据深度（bits_per_sample）为24bit时，播放异常，体现为收音机似的噪音，但可以隐隐约约听见一些音乐内容
当设置为16bit或32bit时，播放不存在问题，播放正常
本人保证了每次测试时的音频文件的采样率，数据深度等和I2S设置的是相同的。

DAC使用PCM5102A，**没有使用mclk引脚**
use apll和mclk倍率都尝试过不同设置，都无解
I2S流设置：
i2s_stream_cfg_t i2s_stream_cfg = {
        .type = AUDIO_STREAM_WRITER,
        .i2s_config = {
            .mode = I2S_MODE_MASTER | I2S_MODE_TX,
            .sample_rate = DEST_SAMPLE_RATE（此处为48000）,
            .bits_per_sample = DEST_SAMPLE_BITS（问题存在的变量，16/32bit正常，24bit异常）,
            .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
            .communication_format = I2S_COMM_FORMAT_STAND_I2S,
            .intr_alloc_flags = ESP_INTR_FLAG_LEVEL2 | ESP_INTR_FLAG_IRAM,
            .dma_buf_count = 8,
            .dma_buf_len = 1024,
            .use_apll = false,
            .tx_desc_auto_clear = true,
            .fixed_mclk = 0,
        },
        .i2s_port = I2S_NUM_0,
        .use_alc = true,
        .volume = 0,
        .out_rb_size = (10 * 1024),
        .task_stack = I2S_STREAM_TASK_STACK,
        .task_prio = I2S_STREAM_TASK_PRIO,
        .task_core = 0,
        .stack_in_ext = true,
        .multi_out_num = 0,
        .uninstall_drv = true,
        .need_expand = false,
        .expand_src_bits = I2S_BITS_PER_SAMPLE_24BIT,
    };

日志不存在报错，个人已排除网络问题以及电源，硬件问题。
直觉告诉我跟decoder内部的数据对齐有关，但decoder非开源无法验证

idf使用版本为4.4.3，adf使用最新版本

Nsiso, 2023-01-10:
追加硬件信息：
开发板：公司内部的EVM板，硬件已验证
模块使用 ESP32 WROVER N8R8 内置天线
DAC: PCM5102A

Nsiso, 2023-02-01:
@ahhfzhang @jason-mao Why this issue has no reply? If you need a english version issue, i can repost one.
I tried to contact espressif by email, but sadly the espressif email system shows me "Undelivered Mail Returned to Sender"
This is really frustrating me, I hope you guys can deal with this issue, because the company product is in a rush.

jason-mao, 2023-02-02:
@Nsiso 不好意思，回复晚了。您帮给一个音乐文件，我们来复现一下。

Nsiso, 2023-02-02:
> @Nsiso 不好意思，回复晚了。您帮给一个音乐文件，我们来复现一下。

Github有文件大小限制，我上传到私人网盘了
分享链接：https://disk.nsiso.com/s/alhm
密码是espressif
文件校验SHA256: 96c6ee32aa56341472c85c9e6415c283a40f8804f8d808e8a91638cf094454ce
音频信息：48000HZ 24位整形 WAV文件

TempoTian, 2023-02-08:
请试试下面的patch看是否解决你的问题
```

diff --git a/components/audio_stream/i2s_stream.c b/components/audio_stream/i2s_stream.c
index c4c03cb7..59a7075d 100644
--- a/components/audio_stream/i2s_stream.c
+++ b/components/audio_stream/i2s_stream.c
@@ -61,6 +61,8 @@ typedef struct i2s_stream {
     void                *volume_handle;
     int                 volume;
     bool                uninstall_drv;
+    int                 skip_size;
+    int                 send_size;
 } i2s_stream_t;
 #ifdef SOC_I2S_SUPPORTS_ADC_DAC
 static esp_err_t i2s_mono_fix(int bits, uint8_t *sbuff, uint32_t len)
@@ -252,7 +254,23 @@ static int _i2s_write(audio_element_handle_t self, char *buffer, int len, TickTy
                          &bytes_written,
                          ticks_to_wait);
     } else {
-        i2s_write(i2s->config.i2s_port, buffer, len, &bytes_written, ticks_to_wait);
+        int skipped = 0;
+        if (i2s->skip_size) {
+            if (len >= i2s->skip_size) {
+                len -= i2s->skip_size;
+                buffer += i2s->skip_size;
+                skipped = i2s->skip_size;
+                i2s->skip_size = 0;
+                i2s->send_size = 0;
+            } else {
+                i2s->skip_size -= len;
+                return len;
+            }
+        }
+        if (len) {
+            i2s_write(i2s->config.i2s_port, buffer, len, &bytes_written, ticks_to_wait);
+        }
+        bytes_written += skipped;
     }
 
     return bytes_written;
@@ -284,6 +302,9 @@ static int _i2s_process(audio_element_handle_t self, char *in_buffer, int in_len
         audio_element_multi_output(self, in_buffer, r_size, 0);
         w_size = audio_element_output(self, in_buffer, r_size);
         audio_element_update_byte_pos(self, w_size);
+        if (w_size > 0) {
+            i2s->send_size += w_size;
+        }
     } else {
         esp_err_t ret = i2s_stream_clear_dma_buffer(self);
         if (ret != ESP_OK) {
@@ -309,7 +330,11 @@ esp_err_t i2s_stream_set_clk(audio_element_handle_t i2s_stream, int rate, int bi
         err = ESP_FAIL;
     }
     if (state == AEL_STATE_RUNNING) {
+        int sample_size = (bits * ch) >> 3;
+        i2s->skip_size = sample_size - (i2s->send_size % sample_size);
+        ESP_LOGE(TAG, "Skip size %d", i2s->skip_size);
         audio_element_resume(i2s_stream, 0, 0);
+        i2s->send_size = 0;
     }
     return err;
 }
```

jason-mao, 2023-02-08:
@Nsiso  请使用这个patch，目前测试您提供的文件已可以。
[i2s_24bit_support.patch](https://github.com/espressif/esp-adf/files/10684947/i2s_24bit_support.patch)

Nsiso, 2023-02-09:
This issue has been fixed. waiting to merge to main branch.

closed 2023-02-09 by Nsiso

andychenli, 2023-04-24:
这个使用的是那个版本的adf，我使用i2s_24bit_support.patch这patch文件打patch提示失败。我的adf是master版本。

andychenli, 2023-04-24:
失败是提示是：
$ git apply i2s_24bit_support.patch
error: patch failed: components/audio_stream/i2s_stream.c:243
error: components/audio_stream/i2s_stream.c: patch does not apply
error: patch failed: components/audio_stream/include/i2s_stream.h:53
error: components/audio_stream/include/i2s_stream.h: patch does not apply
error: examples/player/pipeline_play_sdcard_music/main/play_sdcard_music_example.c: No such file or directory

GCTechnology, 2025-02-17:
48000 24bit的WAV复现了这个问题，转换为16bit则没有出现问题

GCTechnology, 2025-02-17:
> 48000 24bit的WAV复现了这个问题，转换为16bit则没有出现问题。 正在尝试更新分支

可惜失败了，#1106 的方案也尝试过了

TempoTian, 2025-02-17:
对应的代码已经merge到master了应该，你可以改下example的i2s 初始化的配置
```
     i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
     i2s_cfg.type = AUDIO_STREAM_WRITER;
+    i2s_cfg.i2s_config.bits_per_sample = 24;
+    i2s_cfg.need_expand = true;
+    i2s_cfg.expand_src_bits = 24;
+    i2s_cfg.buffer_len = 3072;
     i2s_stream_writer = i2s_stream_init(&i2s_cfg);
```

GCTechnology, 2025-02-17:
> 对应的代码已经merge到master了应该，你可以改下example的i2s 初始化的配置
>
> ```
>      i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
>      i2s_cfg.type = AUDIO_STREAM_WRITER;
> +    i2s_cfg.i2s_config.bits_per_sample = 24; [...]

感谢您的解答
但此方法先前已经尝试过，但是仍然没有解决噪声问题。
且IDF 5以上的版本该结构体已无 ‘i2s_config_t    i2s_config;’ 这一项，截图如下，截取自 https://github.com/espressif/esp-adf/blob/master/components/audio_stream/include/i2s_stream.h#L204

[image]

jason-mao, 2025-02-18:
@GCTechnology 你的24bit wav文件提供一下，我们来测试一下

GCTechnology, 2025-02-18:
> [@GCTechnology](https://github.com/GCTechnology) 你的24bit wav文件提供一下，我们来测试一下

[audio.zip](https://github.com/user-attachments/files/18837381/audio.zip) 截取了一个片段