espressif/esp-adf #943: Restore SIP over HTTPS Feature

sidwarkd, 2023-01-19:
**Is your feature request related to a problem? Please describe.**

The latest VoIP example changes remove the ability to do HTTPS SIP calls. `esp_rtc_config_t` does not have the option to specify a `cert_pem` like `sip_config_t`. As far as I can tell this makes HTTPS SIP integrations impossible now. The new changes are also not backward compatible as the esp_sip.h header now throws the following warning.

```txt
"This header is deprecated, please use new SIP related API in esp_rtc.h, and refer to the latest voip example in '/examples/protocols/voip' "
```

**Describe the solution you'd like**

The ability to connect to a SIP server over HTTPS like before.

ahhfzhang, 2023-01-28:
Hi @sidwarkd 

Thanks for your report, we will rollback this feature soon.

closed 2023-02-03 by espressif-bot

sidwarkd, 2023-02-06:
@ahhfzhang This is still not resolved. While the commit referenced makes changes to the closed source libraries there is still no way to pass the cert in the config struct `esp_rtc_config_t`.

ahhfzhang, 2023-02-07:
Hi @sidwarkd 

https://github.com/espressif/esp-adf-libs/blob/master/esp_media_protocols/include/esp_rtc.h#L109
Have you configured it here? This pem data will passthrough to the configuration of esp_tls. Can you provide a complete log?

Thanks.

sidwarkd, 2023-02-07:
@ahhfzhang My apologies, I did not pull latest commit of all submodules. I have the proper header now and confirmed I can connect to the server. Thank you.

ahhfzhang, 2023-02-08:
Great, glad to hear that.