espressif/esp-adf #944: Memory Leak in SIP Library

sidwarkd, 2023-01-19:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version:                 ESP32-LyraT v4.3
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4.3-316-ge86181704a
- ADF version: v2.4.1-24-ga5250e4
- Build system:    idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system:     Linux
- Using an IDE?:  VSCode
- Power supply:  USB

## Problem Description

When using the SIP client library after connecting to the SIP server the library appears to leak memory at every OPTIONS call. This can easily be reproduced with the out-of-box VoIP example. In my setup I found just maintaining a connection to the SIP server leaks about 680 bytes every 5 minutes. During SIP calls I saw leaks exceeding 1K for every call placed.

### Expected Behavior

No memory leaks.

### Actual Behavior

Leaking memory frequently.

### Steps to Reproduce

  1. Set up the [VoIP example firmware](https://github.com/espressif/esp-adf/tree/release/v2.4/examples/advanced_examples/voip) using the LyraT dev kit.
  1. In menuconfig set up the example by providing WIFI info and a valid SIP URI
  1. Add `#include "esp_heap_caps.h"` to the top of **voip_app.c**
  1. Add the following function to **voip_app.c** above `app_main`
```c
void print_memory_info(){
    size_t internal_free_mem = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    size_t spi_free_mem = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
    size_t internal_largest_free_block = heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL);
    size_t spi_largest_free_block = heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM);

    ESP_LOGI("memory", "Internal free/lfb: %d/%d SPI free/lfb: %d/%d", internal_free_mem, internal_largest_free_block, spi_free_mem, spi_largest_free_block);
}
```
  5. At the bottom of `app_main` add the following block
 ```c
while(1){
    print_memory_info();
    vTaskDelay(5000 / portTICK_PERIOD_MS);
}
```
  6. Flash and run the example
  7. After connecting to the server the board will make an audio announcement and you will see the SPI memory output which will constantly decrease indicating a leak in the library.

## Debug Logs

Here is my memory output from several minutes showing the issue. Notice the SPI free value is constantly decreasing.
At 11.67s runtime the free SPI memory was 4034095. At 176.68s it was down to 4033759 which is a loss of 336 bytes of memory in just over 165s. This loss is much worse when actual calls are made. For brevity I have not included the SIP logging in the output below but you will see the memory decreases every time the SIP client does an OPTIONS or other activity.

```
I (11671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034095/3997696
I (16671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034095/3997696
I (21671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034095/3997696
I (26671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034095/3997696
I (31671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034063/3997696
I (36671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034063/3997696
I (41671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034031/3997696
I (46671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034031/3997696
I (51671) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034031/3997696
I (56681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4034031/3997696
I (61681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033999/3997696
I (66681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033999/3997696
I (71681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033999/3997696
I (76681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033971/3997696
I (81681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033971/3997696
I (86681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033971/3997696
I (91681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033939/3997696
I (96681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033939/3997696
I (101681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033939/3997696
I (106681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033907/3997696
I (111681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033907/3997696
I (116681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033907/3997696
I (121681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033875/3997696
I (126681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033875/3997696
I (131681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033875/3997696
I (136681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033875/3997696
I (141681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033831/3997696
I (146681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033831/3997696
I (151681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033799/3997696
I (156681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033799/3997696
I (161681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033799/3997696
I (166681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033799/3997696
I (171681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033759/3997696
I (176681) memory: Internal free/lfb: 238367/110592 SPI free/lfb: 4033759/3997696
```

## Other Items If Possible

- [x] sdkconfig - using default from the example. Only modified with my WiFi credentials and SIP URI

ahhfzhang, 2023-01-28:
Hi @sidwarkd 

Thanks for your report, let us check then reply.

ahhfzhang, 2023-02-01:
Hi @sidwarkd

We can't reproduce this issue on the latest master branch, can you help to test it again？

Thanks.

sidwarkd, 2023-02-01:
@ahhfzhang Because of #943 we cannot use latest master branch. Is it possible to fix on the 2.4 release? Can you reproduce on the 2.4 release?

ahhfzhang, 2023-02-01:
We have fixed this issue and will commit to master branch soon.

ahhfzhang, 2023-02-08:
@sidwarkd  SIP tls feature already rollback, does this issue still exist on the latest master branch？

ahhfzhang, 2023-02-14:
Hi @sidwarkd 

Audio kit version: ESP32-LyraT v4.3
Module or chip used: ESP32-WROVER-E
IDF version: v4.4.3
ADF version: v2.4.1 

VoIP memleak bugfix on adf v2.4.1 ： 

1.  free ringbuf_ref here . https://github.com/espressif/esp-adf/blob/v2.4.1/examples/advanced_examples/voip/main/voip_app.c#L314
 `#if !RECORD_HARDWARE_AEC
            rb_destroy(ringbuf_ref);
        #endif`
        
2. add algorithm_stream fix https://github.com/espressif/esp-adf/commit/3e837f0

3. update adf lib [esp_media_protocols_lib.zip](https://github.com/espressif/esp-adf/files/10731566/esp_media_protocols_lib.zip)

Hope these fixes works for you, thanks.

sidwarkd, 2023-02-14:
@ahhfzhang Thank you for the update. I will try to verify these changes in the next day or two

sidwarkd, 2023-02-16:
@ahhfzhang In our code we were not using the `ringbuf_ref` so that change didn't apply. The commit we are locked to for ADF (d82e946156) already had the algorithm_stream fix on it. So the only change I made was using the updated libesp_media_protocols.a in the `esp_media_protocols` component. I ran two tests before and after the change.

  1. Maintaining a registration with the SIP server without making a call. Calling `esp_sip_start` and letting the SIP thread run and send OPTIONS. I let this sit for 132s in both cases.
  2. Making a call. The call in both cases was about 20s long

## Test 1 (Registration with OPTIONS)
### Before Change
Internal Memory Loss: 284 bytes
SPI Memory Loss: 284 bytes

### After Change
Internal Memory Loss: 220 bytes
SPI Memory Loss: 0 bytes <---------- FIXED

## Test 2 (Call)
### Before Change
Internal Memory Loss: 236 bytes
SPI Memory Loss: 1324 bytes

### After Change
Internal Memory Loss: 128 bytes
SPI Memory Loss: 1108 bytes  <---------- Better, but still leaking significantly

I can't say with certainty that all of the internal memory losses are caused by the ADF. I will continue to investigate that in our code but the SPI Memory loss on calls is definitely still an issue although slightly better. Please let me know if you have any questions about my tests or results.

ahhfzhang, 2023-02-17:
Hi @sidwarkd 

These block will be freed after calling esp_sip_destroy.

Ps. You can refer to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html?highlight=heap%20debug#how-to-diagnose-memory-leaks) to investigate memory leak issues.

sidwarkd, 2023-02-17:
@ahhfzhang Will do more testing and report back.

sidwarkd, 2023-02-17:
@ahhfzhang More testing results. The 1k or so that is lost after a call appears to be a one time hit. Making additional calls do not seem to leak additional memory and I confirm that it cleans up after calling `esp_sip_destroy`. At this point the updated sip client binary you provided seems to make a significant difference. This issue can be closed and I will open another issue if I find more specific leaks using the method linked above.

closed 2023-02-21 by jason-mao