espressif/esp-adf #950: voip例程，按play键无法接听

zdodo55, 2023-02-14:
使用环境：
esp-adf-v2.4.1
esp-idf-v4.4.3
开发板：ESP32-LyraT V4.3
例程：[https://github.com/espressif/esp-adf/tree/v2.4.1/examples/advanced_examples/voip](https://github.com/espressif/esp-adf/tree/v2.4.1/examples/advanced_examples/voip)
服务端：Freeswitch
客户端：MicroSIP

问题描述：

1. ESP32-LyraT V4.3呼出，MicroSIP接听，可以正常接通并通话。
2. MicroSIP呼出，ESP32-LyraT V4.3收到电话呼入，振铃响声，但是按play键后，MicroSIP显示已接通，但是ESP32-LyraT V4.3还是持续显示有来电呼入，一直振铃，MicroSIP和ESP32-LyraT V4.3均没有通话声音。

log:
I (35480) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:775, MEM Total:4255519 Bytes, Inter:241331 Bytes, Dram:202319 Bytes

I (35490) ESP_AUDIO_TASK: It's a decoder
I (35500) ESP_AUDIO_TASK: 1.CUR IN:[IN_flash],CODEC:[DEC_mp3],RESAMPLE:[8000],OUT:[OUT_iis],rate:0,ch:0,pos:0
I (35510) ESP_AUDIO_TASK: 2.Handles,IN:0x3f804f14,CODEC:0x3f8050a4,FILTER:0x3f8144bc,OUT:0x3f805250
I (35520) AUDIO_PIPELINE: audio_pipeline_unlinked
W (35520) AUDIO_PIPELINE: There are no listener registered
I (35530) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (35530) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (35540) AUDIO_PIPELINE: link el->rb, el:0x3f804f14, tag:IN_flash, rb:0x3f814aa8
I (35550) AUDIO_PIPELINE: link el->rb, el:0x3f8050a4, tag:DEC_mp3, rb:0x3f81480c
I (35560) AUDIO_PIPELINE: link el->rb, el:0x3f8144bc, tag:Audio_forge, rb:0x3f814848
I (35560) ESP_AUDIO_TASK: 3. Previous starting...
I (35570) AUDIO_ELEMENT: [IN_flash-0x3f804f14] Element task created
I (35580) AUDIO_ELEMENT: [IN_flash] AEL_MSG_CMD_RESUME,state:1
I (35580) AUDIO_THREAD: The DEC_mp3 task allocate stack on external memory
I (35580) TONE_PARTITION: tone partition format 0, total 21
I (35590) AUDIO_ELEMENT: [DEC_mp3-0x3f8050a4] Element task created
I (35600) TONE_STREAM: Tone offset:00020c78, Tone length:36018, pos:8

I (35600) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (35620) MP3_DECODER: MP3 opened
I (35620) ESP_AUDIO_TASK: Blocking play until received AEL_MSG_CMD_REPORT_MUSIC_INFO
I (35630) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f804f14] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (35640) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f8050a4] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (35720) ESP_AUDIO_TASK: Received muisc info then on play
I (35720) ESP_AUDIO_TASK: On event play, status:UNKNOWN, 0
I (35720) AUDIO_THREAD: The Audio_forge task allocate stack on external memory
I (35730) AUDIO_ELEMENT: [Audio_forge-0x3f8144bc] Element task created
I (35740) AUDIO_ELEMENT: [Audio_forge] AEL_MSG_CMD_RESUME,state:1
I (35740) AUDIO_FORGE: audio_forge opened
I (35740) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (35760) AUDIO_ELEMENT: [OUT_iis-0x3f805250] Element task created
I (35760) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (3570) I2S_STREAM: AUDIO_STREAM_WRITER
I (35770) ESP_AUDIO_CTRL: Exit play procedure, ret:0
I (35780) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f805250] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (35780) AUDIO_FORGE: audio_forge reopen
I (35790) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f8144bc] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (35810) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING, 0, src:0, is_stopping:0
I (35820) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:1030, MEM Total:3873251 Bytes, Inter:210687 Bytes, Dram:171675 Bytes

I (36480) VOIP_EXAMPLE: ringing... RemotePhoneNum 1000
I (36480) ESP_AUDIO_CTRL: Enter play procedure, src:0
W (36480) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.
W (37400) TONE_STREAM: No more data,ret:0 ,info.byte_pos:36018
I (37400) AUDIO_ELEMENT: IN-[IN_flash] AEL_IO_DONE,0
I (37400) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f804f14] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (37480) VOIP_EXAMPLE: ringing... RemotePhoneNum 1000
I (37480) ESP_AUDIO_CTRL: Enter play procedure, src:0
W (37480) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.
I (38290) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_DONE,-2
I (38480) VOIP_EXAMPLE: ringing... RemotePhoneNum 1000
I (38480) ESP_AUDIO_CTRL: Enter play procedure, src:0
W (38480) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.
I (38490) MP3_DECODER: Closed
[... 467 lines cut ...]
SIP/2.0 486 Busy Here
Via: SIP/2.0/TCP 192.168.2.55;branch=z9hG4bKta5N5ac0g6y2D
Contact: <sip:1001@192.168.2.99:64152>
From: "Extension 1000" <sip:1000@192.168.2.55>;tag=F6XUN4QyeN9Br
To: <sip:1001@192.168.2.99:64152;transport=TCP>;tag=-59658848
Call-ID: 39df6c7f-22eb-123c-3f96-11167911e9b9
CSeq: 63409426 INVITE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, UPDATE, CANCEL, OPTIONS, INFO
Content-Length: 0

I (56670) SIP: [1970-01-01/00:00:27]=======================>>
I (56690) SIP: [1970-01-01/00:00:27]<<=====READ 0349 bytes==
I (56690) SIP:

ACK sip:1001@192.168.2.99:64152;transport=TCP SIP/2.0
Via: SIP/2.0/TCP 192.168.2.55;branch=z9hG4bKta5N5ac0g6y2D
Max-Forwards: 69
From: "Extension 1000" <sip:1000@192.168.2.55>;tag=F6XUN4QyeN9Br
To: <sip:1001@192.168.2.99:64152;transport=TCP>;tag=-59658848
Call-ID: 39df6c7f-22eb-123c-3f96-11167911e9b9
CSeq: 63409426 ACK
Content-Length: 0

I (56720) SIP: [1970-01-01/00:00:27]<<======================

zdodo55, 2023-02-14:
adf更新至master，问题依旧！

ahhfzhang, 2023-02-14:
Hi @zdodo55 

从你的日志来看的话，是Freeswitch没有回复ACK包导致的，请检查一下服务器日志。

ps. Freeswitch 请使用ubuntu版本，win版本会出现类似没有相应的情况出现。

Thanks.

zdodo55, 2023-02-15:
我使用了Wireshark重新抓包，发现Freeswitch是应答的，如图，详细数据：[voip.zip](https://github.com/espressif/esp-adf/files/10738498/voip.zip)
其中
192.168.2.56：Freeswitch，
192.168.2.55：MicroSIP，
192.168.2.111：ESP32-LyraT V4.3，
[image]

但不知ESP32-LyraT V4.3为何没收到，或者收到了没见打印，log如下：

```
I (48730) SIP: [1970-01-01/00:00:23]=======WRITE 0844 bytes>>
I (48730) SIP: 

OPTIONS sip:1001@192.168.2.56:5060 SIP/2.0
Via: SIP/2.0/TCP 192.168.2.111:55536;branch=z9hG4bK--247014808;rport
From: <sip:1001@192.168.2.56:5060>;tag=1377756634
To: <sip:1001@192.168.2.56:5060>
Contact: <sip:1001@192.168.2.111:55536;transport=TCP>
Max-Forwards: 70
Call-ID: 4DDFB9FB181AF38658296DBF778737909A1F50846596
CSeq: 3 OPTIONS
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="1001", realm="192.168.2.56", nonce="26afadad-3c09-4f84-8c0b-bddce1263c1e", uri="sip:192.168.2.56:5060", response="34d9d7d591b3aac74db34e03bd4e85dc", algorithm=MD5, nc=00000002, cnonce="cd25abe3c435bbeb", qop="auth"

I (48800) SIP: [1970-01-01/00:00:23]=======================>>
I (48850) SIP: [1970-01-01/00:00:23]<<=====READ 0723 bytes==
I (48850) SIP: 

SIP/2.0 200 OK
Via: SIP/2.0/TCP 192.168.2.111:55536;branch=z9hG4bK--247014808;rport=55536
From: <sip:1001@192.168.2.56:5060>;tag=1377756634
To: <sip:1001@192.168.2.56:5060>;tag=tvDgatF58Baea
Call-ID: 4DDFB9FB181AF38658296DBF778737909A1F50846596
CSeq: 3 OPTIONS
Contact: <sip:192.168.2.56>
User-Agent: FreeSWITCH-mod_sofia/1.10.9-release~a615e85afc~64bit
Accept: application/sdp
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: timer, path, replaces
Allow-Events: talk, hold, conference, presence, as-feature-event, dialog, line-seize, call-info, sla, include-session-description, presence.winfo, message-summary, refer
Content-Length: 0

I (48910) SIP: [1970-01-01/00:00:23]<<======================
I (67210) SIP: [1970-01-01/00:01:03]<<=====READ 1225 bytes==
I (67210) SIP: 

INVITE sip:1001@192.168.2.111:55536;transport=TCP SIP/2.0
Via: SIP/2.0/TCP 192.168.2.56;branch=z9hG4bKvmr824Xm0DtNc
Max-Forwards: 69
From: "Extension 1000" <sip:1000@192.168.2.56>;tag=m45t05Q4m3c3B
To: <sip:1001@192.168.2.111:55536;transport=TCP>
Call-ID: 9760b5a4-2771-123c-8b93-0569c3c38f71
CSeq: 63658183 INVITE
Contact: <sip:mod_sofia@192.168.2.56:5060;transport=tcp>
User-Agent: FreeSWITCH-mod_sofia/1.10.9-release~a615e85afc~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: timer, path, replaces
Allow-Events: talk, hold, conference, presence, as-feature-event, dialog, line-seize, call-info, sla, include-session-description, presence.winfo, message-summary, refer
Content-Type: application/sdp
Content-Disposition: session
Content-Length: 244
X-FS-Support: update_display,send_info
Remote-Party-ID: "Extension 1000" <sip:1000@192.168.2.56>;party=calling;screen=yes;privacy=off

v=0
[... 357 lines cut ...]
I (77000) ESP_AUDIO_TASK: On event play, status:UNKNOWN, 0
I (77000) AUDIO_THREAD: The Audio_forge task allocate stack on external memory
I (77010) AUDIO_ELEMENT: [Audio_forge-0x3f81445c] Element task created
I (77010) AUDIO_ELEMENT: [Audio_forge] AEL_MSG_CMD_RESUME,state:1
I (77020) AUDIO_FORGE: audio_forge opened
I (77020) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (77030) AUDIO_ELEMENT: [OUT_iis-0x3f8051f0] Element task created
 (77040) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (77050) I2S_STREAM: AUDIO_STREAM_WRITER
I (77050) ESP_AUDIO_CTRL: Exit play procedure, ret:0
I (77050) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f81445c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (77070) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f8051f0] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (77080) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING, 0, src:0, is_stopping:0
I (77090) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:990, MEM Total:3878179 Bytes, Inter:211255 Bytes, Dram:172243 Bytes

I (77760) VOIP_EXAMPLE: ringing... RemotePhoneNum 1000
I (77760) ESP_AUDIO_CTRL: Enter play procedure, src:0
W (77760) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.
W (78530) TONE_STREAM: No more data,ret:0 ,info.byte_pos:36018
I (78530) AUDIO_ELEMENT: IN-[IN_flash] AEL_IO_DONE
I (78530) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f804eb4] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (78760) VOIP_EXAMPLE: ringing... RemotePhoneNum 1000
I (78760) ESP_AUDIO_CTRL: Enter play procedure, src:0
W (78760) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.
```

ahhfzhang, 2023-02-15:
Hi @zdodo55 

ok，已经找到问题了，这个是我们发的包是TCP
[image]

再看服务器回复的包，这里变成了UDP，所以设备是收不到的
[image]

客户端和MicroSIP用一样的UDP再测试一下吧。

zdodo55, 2023-02-15:
非常感谢，都没有想过往TCP/UDP方面找问题，想当然就觉得用TCP连接，Freeswitch也一样使用TCP应答，真没想到啊，困扰了我好些日。
都使用UDP后问题解决。

Kevincoooool, 2023-02-20:
如果想使用TCP通信的话怎么办呢，有没有办法兼容一下

ahhfzhang, 2023-02-21:
Hi @Kevincoooool 

我这里使用的Freeswitch版本是1.10.6，使用TCP相互通信也是完全正常的。

closed 2023-06-21 by espressif-bot