espressif/esp-adf #952: Compatibility between ADF2.4 and IDF5

abdoutech93, 2023-02-22:


jason-mao, 2023-02-23:
@abdoutech93 We plan to support IDF5 at ADF v2.6

MartinHeywang, 2023-02-25:
Hello !

When do you plan on releasing ADF v2.6 ? I am currently using IDF 5.0.1 and am really interested in using the ADF (I just found out about it). It feels a shame to downgrade my IDF version.

What would you advise me (wait for ADF v2.6 to be released or temporarily downgrade my IDF version) ?

jason-mao, 2023-02-27:
@MartinHeywang We have almost done to support IDF v5.x,  I think it will be merge to latest master after two or three weeks.   

> What would you advise me (wait for ADF v2.6 to be released or temporarily downgrade my IDF version) ?

If your project is not dependency the v5.x features, you can use the v4.4.4.If yes, please make the others work first.

texruska, 2023-05-29:
Should close this issue now, IDF v5 is working

closed 2023-09-05 by jason-mao