espressif/esp-adf #953: p2p support esp-rtc

yogiD, 2023-02-22:
Feature  : Setup p2p video sip call . 

Board : ESP32-S3-Korvo-2 V3
https://www.espressif.com/en/news/ESP-RTC
p2p call  will not be register with SIP server and does not need sip systems . To make p2p call I needs the sip client api or source  
of  esp-rtc : media_lib_sal .

jason-mao, 2023-02-23:
@yogiD Would you provide more application scenarios for us,  then we can make plan to do it.

yogiD, 2023-02-23:
Application scenarios where sip p2p suffice for communication ( No overhead of sip infrastructure) 
1. video Doorbell phone 
2. Small community intercom

Hasaken, 2023-03-02:
Hi guys.
SIP clients have "direct media" option, which can be applied after a call is established. In that case, rtp media is routed directly between sip clients, without routing through server. If you implement such a function in your SIP client it can be a solution. Free libraries, like PJSIP, already have that feature implemented.