espressif/esp-adf #955: ESP32-LYRAT4.3 Live stream using play_living_stream_example ERROR

abdoutech93, 2023-02-23:
I'm testing Live stream using play_living_stream_example and I got this error when I used the the azure media services stream with (format=m3u8-cmaf)
 
Here is the errors log lines
```
W (9206) HTTP_STREAM: No more data,errno:0, total_bytes:32887, rlen = 0
I (9206) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (9816) HTTP_STREAM: total_bytes=32745
W (11076) HTTP_STREAM: No more data,errno:0, total_bytes:32745, rlen = 0
I (11076) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (11666) HTTP_STREAM: total_bytes=33284
W (12896) HTTP_STREAM: No more data,errno:0, total_bytes:33284, rlen = 0
I (12896) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (13506) HTTP_STREAM: total_bytes=32848
W (14766) HTTP_STREAM: No more data,errno:0, total_bytes:32848, rlen = 0
I (14766) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (15346) HTTP_STREAM: total_bytes=32853
E (15986) MPEG_READER: Not find an effective frame, line:607.
E (15986) MPEG_READER: resync error (line 399)
E (15986) AAC_DECODER: Encountered error reading stream
E (15996) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN,-1
W (15996) AUDIO_ELEMENT: [aac] audio_element_on_cmd_error,7
I (16006) AAC_DECODER: Closed by [7]
W (16016) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (16596) AUDIO_ELEMENT: OUT-[http] AEL_IO_ABORT
```

here is a screenshot 
[image]

the content of the manifest file : 
`<?xml version="1.0" encoding="UTF-8"?><SmoothStreamingMedia IsLive="TRUE" MajorVersion="2" MinorVersion="2" DVRWindowLength="1737000" Duration="0" TimeScale="1000" LookAheadFragmentCount="2"><StreamIndex Chunks="338" Type="audio" Url="QualityLevels({bitrate})/Fragments(audio={start time})" QualityLevels="1" Subtype="" Name="audio" TimeScale="1000"><QualityLevel AudioTag="255" Index="0" BitsPerSample="16" Bitrate="125000" FourCC="AACL" CodecPrivateData="120856E500" Channels="2" PacketSize="4" SamplingRate="44100" /><c t="0" d="53" n="0" /><c t="12184" d="1834" /><c d="1997" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1996" /><c d="2021" /><c d="1997" r="2" /><c d="1996" /><c d="1998" /><c d="1997" r="2" /><c d="1996" /><c d="2020" /><c d="1997" r="2" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1998" /><c d="1997" r="3" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="2020" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1997" r="3" /><c d="2020" /><c d="1998" /><c d="1997" r="4" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1997" r="4" /><c d="1998" r="2" /><c d="1996" /><c d="1998" /><c d="2020" /><c d="1997" r="5" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1997" r="5" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1998" /><c d="1996" /><c d="1997" /><c d="1998" /><c d="1997" r="3" /><c d="2020" /><c d="1998" /><c d="1997" r="6" /><c d="2020" /><c d="1998" /><c d="1997" r="3" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="2020" /><c d="1997" r="3" /><c d="1998" /><c d="1997" r="3" /><c d="2020" /><c d="1998" /><c d="1997" r="5" /><c d="1998" r="2" /><c d="2020" /><c d="1998" /><c d="1997" /><c d="1998" /><c d="1997" r="3" /><c d="1998" r="2" /><c d="2020" /><c d="1998" /><c d="1997" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="1997" r="2" /><c d="2020" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1997" /><c d="1998" /><c d="1997" /><c d="1998" /><c d="2021" /><c d="1997" r="7" /><c d="2020" /><c d="1997" /><c d="1998" /><c d="1997" /><c d="1998" r="2" /><c d="1997" r="2" /><c d="1998" /><c d="2020" /><c d="1998" r="2" /><c d="1997" /><c d="1998" /><c d="1997" r="4" /><c d="2021" /><c d="1997" /><c d="1998" r="3" /><c d="1996" /><c d="1998" r="2" /><c d="1997" /><c d="2020" /><c d="1997" /><c d="1191" /><c t="370026" d="1974" /><c d="2021" /><c d="1997" /><c d="1998" /><c d="1997" r="4" /><c d="1998" /><c d="2020" /><c d="1997" /><c d="1998" /><c d="1997" /><c d="1998" r="2" /><c d="1997" /><c d="1996" /><c d="1998" /><c d="2020" /><c d="1998" /><c d="1997" r="2" /><c d="1996" /><c d="1997" /><c d="1998" /><c d="1996" /><c d="2021" /><c d="1996" /><c d="1997" r="2" /><c d="1996" /><c d="1997" r="2" /><c d="2021" /><c d="1997" r="2" /><c d="1996" /><c d="1997" /><c d="1996" /><c d="1997" /><c d="2020" /><c d="1997" /><c d="1996" /><c d="1997" r="3" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1997" r="2" /><c d="1998" /><c d="1996" /><c d="1997" /><c d="1996" /><c d="1998" /><c d="2019" /><c d="1997" r="6" /><c d="2020" /><c d="1997" /><c d="1998" /><c d="1997" /><c d="1996" /><c d="1997" r="2" /><c d="1998" /><c d="2020" /><c d="1997" r="2" /><c d="1998" /><c d="1997" r="4" /><c d="2020" /><c d="1997" r="6" /><c d="1998" /><c d="2020" /><c d="1997" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1997" r="2" /><c d="2020" /><c d="1997" /><c d="1998" /><c d="1997" r="4" /><c d="1998" /><c d="1997" /><c d="2020" /><c d="1997" r="2" /><c d="1998" /><c d="1997" r="2" /><c d="1998" /><c d="1996" /><c d="2021" /><c d="1997" r="6" /><c d="1998" /><c d="2020" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="1997" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="2020" /><c d="1997" /><c d="1998" r="2" /><c d="1997" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="1996" /><c d="2021" /><c d="1998" /><c d="1996" /><c d="1998" /><c d="1997" r="4" /><c d="2021" /><c d="1997" r="2" /><c d="1998" /><c d="1996" /><c d="1997" r="2" /><c d="1998" /><c d="2020" /><c d="1996" /><c d="1997" r="4" /></StreamIndex></SmoothStreamingMedia>`

and this is the entire log 
`I (29) boot: ESP-IDF v4.4.4-148-g4c2afac355-dirty 2nd stage bootloader
I (29) boot: compile time 03:17:06
I (30) boot: chip revision: v3.0
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) qio_mode: Enabling default flash chip QIO
I (46) boot.esp32: SPI Speed      : 80MHz
I (51) boot.esp32: SPI Mode       : QIO
I (56) boot.esp32: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 00300000
I (99) boot: End of partition table
I (103) boot_comm: chip revision: 3, min. application chip revision: 0
I (110) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f6cch (194252) map
I (172) esp_image: segment 1: paddr=0003f6f4 vaddr=3ffb0000 size=00924h (  2340) load
I (173) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=c7680h (816768) map
I (403) esp_image: segment 3: paddr=001076a8 vaddr=3ffb0924 size=028bch ( 10428) load
I (407) esp_image: segment 4: paddr=00109f6c vaddr=40080000 size=1bb70h (113520) load
I (459) boot: Loaded app from partition at offset 0x10000
I (459) boot: Disabling RNG early entropy source...
I (470) psram: This chip is ESP32-D0WD
I (471) spiram: Found 64MBit SPI RAM device
I (471) spiram: SPI RAM mode: flash 80m sram 80m
I (476) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (483) cpu_start: Pro cpu up.
I (487) cpu_start: Starting app cpu, entry point is 0x400815bc
0x400815bc: call_start_cpu1 at D:/Programmes/Espressif/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (989) spiram: SPI SRAM memory test OK
I (997) cpu_start: Pro cpu start user code
I (997) cpu_start: cpu freq: 240000000
I (997) cpu_start: Application information:
I (1000) cpu_start: Project name:     living_stream_m3u_aac
[... 321 lines cut ...]
I (7316) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (7976) HTTP_STREAM: total_bytes=32887
I (8406) CODEC_ELEMENT_HELPER: The element is 0x3f800e74. The reserve data 2 is 0x0.
I (8406) AAC_DECODER: a new song playing
I (8406) AAC_DECODER: this audio is RAW AAC
W (9206) HTTP_STREAM: No more data,errno:0, total_bytes:32887, rlen = 0
I (9206) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (9816) HTTP_STREAM: total_bytes=32745
W (11076) HTTP_STREAM: No more data,errno:0, total_bytes:32745, rlen = 0
I (11076) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (11666) HTTP_STREAM: total_bytes=33284
W (12896) HTTP_STREAM: No more data,errno:0, total_bytes:33284, rlen = 0
I (12896) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (13506) HTTP_STREAM: total_bytes=32848
W (14766) HTTP_STREAM: No more data,errno:0, total_bytes:32848, rlen = 0
I (14766) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (15346) HTTP_STREAM: total_bytes=32853
E (15986) MPEG_READER: Not find an effective frame, line:607.
E (15986) MPEG_READER: resync error (line 399)
E (15986) AAC_DECODER: Encountered error reading stream
E (15996) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN,-1
W (15996) AUDIO_ELEMENT: [aac] audio_element_on_cmd_error,7
I (16006) AAC_DECODER: Closed by [7]
W (16016) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (16596) AUDIO_ELEMENT: OUT-[http`

jason-mao, 2023-02-27:
@abdoutech93 How is your IDF and ADF version? BTW, did you use the https://github.com/espressif/esp-adf/tree/master/idf_patches for you IDF version?

abdoutech93, 2023-03-04:
> @abdoutech93 How is your IDF and ADF version? BTW, did you use the https://github.com/espressif/esp-adf/tree/master/idf_patches for you IDF version?

I have checked and  applied them and I found some of them already applied .

But still have the log: no more data, errno:0

closed 2023-09-05 by jason-mao