espressif/esp-adf #962: voip通话一分钟自动被挂断

ericnts, 2023-03-07:
## Environment

- Audio development kit:   [ESP32-S3-Korvo-2]
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): [v3|]
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v4.4.4
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): master
- Build system:         [idf.py]
- [Required] Running log:         
    ESP-ROM:esp32s3-20210327
    OOT)
    SPIWP:0xee
    mode:DIO, clock div:1
    load:0x3fcd0108,len:0x17b8
    load:0x403b6000,len:0xe28
    load:0x403ba000,len:0x30d0
    entry 0x403b6274
    I (25) boot: ESP-IDF v4.4-329-g3c9657fa4e-dirty 2nd stage bootloader
    I (25) boot: compile time 17:04:56
    I (25) boot: chip revision: 0
    I (28) qio_mode: Enabling default flash chip QIO
    I (33) boot.esp32s3: Boot SPI Speed : 80MHz
    I (38) boot.esp32s3: SPI Mode       : QIO
    I (43) boot.esp32s3: SPI Flash Size : 4MB
    I (48) boot: Enabling RNG early entropy source...
    I (53) boot: Partition Table:
    I (57) boot: ## Label            Usage          Type ST Offset   Length
    I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
    I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
    I (79) boot:  2 factory          factory app      00 00 00010000 00200000
    I (86) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
    I (94) boot: End of partition table
    I (98) esp_image: segment 0: paddr=00010020 vaddr=3c0f0020 size=3f474h (259188) map
    I (146) esp_image: segment 1: paddr=0004f49c vaddr=3fc9afc0 size=00b7ch (  2940) load
    I (147) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=e9cech (957676) map
    I (297) esp_image: segment 3: paddr=00139d14 vaddr=3fc9bb3c size=03b2ch ( 15148) load
    I (300) esp_image: segment 4: paddr=0013d848 vaddr=40374000 size=16fb8h ( 94136) load
    I (320) esp_image: segment 5: paddr=00154808 vaddr=50000000 size=00010h (    16) load
    I (330) boot: Loaded app from partition at offset 0x10000
    I (330) boot: Disabling RNG early entropy source...
    I (341) opi psram: vendor id : 0x0d (AP)
    I (341) opi psram: dev id    : 0x02 (generation 3)
    I (342) opi psram: density   : 0x03 (64 Mbit)
    I (345) opi psram: good-die  : 0x01 (Pass)
    I (350) opi psram: Latency   : 0x01 (Fixed)
    I (355) opi psram: VCC       : 0x01 (3V)
    I (359) opi psram: SRF       : 0x01 (Fast Refresh)
    I (365) opi psram: BurstType : 0x01 (Hybrid Wrap)
    I (370) opi psram: BurstLen  : 0x01 (32 Byte)
    I (375) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
    I (381) opi psram: DriveStrength: 0x00 (1/1)
    W (386) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
    I (397) spiram: Found 64MBit SPI RAM device
    I (401) spiram: SPI RAM mode: sram 80m
    I (406) spiram: PSRAM initialized, cache is in normal (1-core) mode.
    I (413) cpu_start: Pro cpu up.
    I (416) cpu_start: Starting app cpu, entry point is 0x403755d4
    I (0) cpu_start: App cpu up.
    I (849) spiram: SPI SRAM memory test OK
    I (858) cpu_start: Pro cpu start user code
    I (858) cpu_start: cpu freq: 240000000
    I (858) cpu_start: Application information:
    I (861) cpu_start: Project name:     voip_app
    I (866) cpu_start: App version:      1
    I (871) cpu_start: Compile time:     Mar  7 2023 14:43:02
    I (877) cpu_start: ELF file SHA256:  112d02adfb8ca8d2...
    I (883) cpu_start: ESP-IDF:          v4.4-329-g3c9657fa4e-dirty
    I (889) heap_init: Initializing. RAM available for dynamic allocation:
    I (896) heap_init: At 3FCA4D90 len 0003B270 (236 KiB): D/IRAM
    I (903) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
[... 963 lines cut ...]

- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     [macOS]
- Using an IDE?: [VsCode]
- Power supply:         [USB]

## Problem Description

通过voip建立通话，每次通话一分钟的时候都会被自动挂断，我们使用了一个商用voip平台和一个freeswitch，两个平台都出现了该问题。

### Expected Behavior

通话不会被自动挂断

### Actual Behavior

每次通话一分钟左右，通话就会被断掉

### Steps to Reproduce

1. 使用esp-adf演示代码中的voip项目
2. 配置voip服务。
3. 编译运行。
4. 进行通话。

ahhfzhang, 2023-03-07:
Hi @ericnts 

你的freeswitch服务器有配置这部分吗？
https://github.com/espressif/esp-adf/blob/master/examples/protocols/voip/README_CN.md?plain=1#L50

看起来是因为通话过程中服务器做了reinvite，因为我们的客户端已经发了心跳包了，不需要服务器再去做这个操作去保持连接。

ericnts, 2023-03-09:
嗯嗯   这个配置是起作用的，另外我们最近又遇到两个新的问题（换了一个freeswitch服务，而且版本跟之前可用的不一致）：
1、电脑端给设备打电话打不通，看日志，设备并没有收到打电话的请求。
2、设备可以打电话给电脑，但是电脑端挂断电话后，设备收不到结束通话的消息，设备一直处于通话的状态。
以上两个问题是同时出现的，所以我想问一下，我们这个demo代码对freeswitch的版本有要求吗，还是说上边两个问题是其他原因导致的

ericnts, 2023-03-09:
@ahhfzhang 感谢！您说的方法确实生效了。
另外我们最近换了一个freeswitch服务，而且版本跟之前用的不一致，然后又遇到两个新的问题：
1、电脑端给设备打电话打不通，看日志，设备并没有收到打电话的请求。
2、设备可以打电话给电脑，但是电脑端挂断电话后，设备收不到结束通话的消息，设备一直处于通话的状态。
以上两个问题是同时出现的，所以我想问一下，我们这个demo代码对freeswitch的版本有要求吗，还是说上边两个问题是其他原因导致的

ahhfzhang, 2023-03-13:
Hi @ericnts 

对于服务器版本没有特别的要求，不过你说的这个问题，尝试一下都是用UDP端口跑一下看看。